RROrg/rr #1874: ESXI 8.0+24.7.1.ova 无限重启

meteorkin, 2024-07-18:
进到这一步后大概5-10s后就自动重启
[image]

配置loader时仅选择3622+7.2、不做任何addons/module/其他参数调整、不在esxi上添加任何硬盘/直通硬件也是如此

请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  1581
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.7.1.ova
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS3622xs
 * version:  7.2

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-07-18:
知道了， vmware 系列有bug，先用24.6.9吧，或则等晚上更新

wjz304, 2024-07-18:
https://github.com/RROrg/rr/issues/1835

closed 2024-07-19 by meteorkin

vrain1025, 2024-07-19:
我这边测试24.7.2没有问题，也没出现首次boot时间很长的问题。OVA直接导入，也未修改虚机配置，编译引导后，一切正常，编译的是SA6400 。这是我的主机相关信息：
[image]
和虚机配置信息：
[image]