RROrg/rr #1893: BUG: Reboot Loop since upgrade from RR 2.4.7.1 to 2.4.7.2

pkdick, 2024-07-20:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  AMD Ryzen 5 PRO 4650GE with Radeon Graphics
 * NIC:  (pid & vid) 0x5583 & 0x0781

RR version: (You can find it in the update menu.)  
 * RR:  2.4.7.2
 * addons:  expands, codepatch, powersched, cpuinfo, mountloader, acpid, sensors
 * modules:  all selected
 * lkms:  prod

DSM: 
 * model:  DS1823xs+
 * version:  DSM 7.2.1-69057 Update 5

Issue:  Reboot loop since update from 2.4.7.1 (no problem with RR 2.4.7.1 --> I had to downgrade from 2.4.7.2 to recover my system). Everything seemed to work normal after RR Upgrade but after RR boot, my system reboots everytime

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

pkdick, 2024-07-21:
Sorry, I forgot to give my DMI: B450 i AORUS Wifi

I also tried RR 2.4.7.3 with no success : during the RR Boot, after the usual information, the screen blanks and my system reboots.

Please tell me how to provide the necessary information for you to debug this issue: for now on I have to stay to RR 2.4.7.1 version

wjz304, 2024-07-21:
How the versions ≤ 24.6.9 ?

wjz304, 2024-07-21:
in 24.7.1
RR shell
```
fbset -i
```

pkdick, 2024-07-21:
Versions inferior 2.4.7.2 worked without any problem...

Thank you for your support!

--190d61bbecc74cb28f3eaf92f0
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: quoted-printable

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.=
w3.org/TR/html4/loose.dtd"><html>
<head>
<meta http-equiv=3D"Content-Type" content=3D"text/html; charset=3Dutf-8"></=
head>
<body>
<div dir=3D"auto">
<div dir=3D"auto">Versions inferior 2.4.7.2 worked without any problem...</=
div><div dir=3D"auto"><br></div><div dir=3D"auto">Thank you for your suppor=
t!</div><div dir=3D"auto"><br></div>
<div id=3D"aqm-original" style=3D"color: black;">

<div class=3D"aqm-original-body">
<div style=3D"color: black;">
<p style=3D"color: black; font-size: 10pt; font-family: sans-serif; margin:=
 8pt 0;">Le 21 juillet 2024 16:51:11 Ing ***@***.***&gt; a=
 =C3=A9crit :</p>
<blockquote type=3D"cite" class=3D"gmail_quote" style=3D"margin: 0 0 0 0.75=
ex; border-left: 1px solid #808080; padding-left: 0.75ex;">
<p></p>
<p dir=3D"auto">How the versions =E2=89=A4 24.6.9 ?</p>

<p style=3D"font-size:small;-webkit-text-size-adjust:none;color:#666;">=E2=
=80=94<br>
/rr/issues/1893#issuecomment-2241637025">view it on GitHub</a>, or <a href=
ETZNPDFZAVCNFSM6AAAAABLF2YUOWVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDENBR=
 the thread.<img src=3D"https://github.com/notifications/beacon/AKHHW4KFYGS=
CZDWF52RF6FDZNPDFZA5CNFSM6AAAAABLF2YUOWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23=
LFNZ2KUY3PNVWWK3TUL5UWJTUFTSVKC.gif" height=3D"1" width=3D"1" alt=3D""><spa=
n style=3D"color: transparent; font-size: 0; display: none; visibility: hid=
den; overflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-h=
025</span><span>@</span><span>github</span><span>.</span><span>com&gt;</spa=
n></span></p>
</blockquote>
</div>
</div>

</div><div dir=3D"auto"><br></div>
</div></body>
</html>

--190d61bbecc74cb28f3eaf92f0--

pkdick, 2024-07-21:
Hello,

How to I get access to the RR shell?

When RR 2.4.7.1 starts the screen goes blank, so I think I cannot do 
anything. Maybe is it possible to get access to RR with a telnet/ssh 
session? Please advise.

wjz304, 2024-07-21:
[image]

pkdick, 2024-07-21:
Here it is!

Thank you

--190d62e03ff74cb28f3b6418a4
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: quoted-printable

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.=
w3.org/TR/html4/loose.dtd"><html>
<head>
<meta http-equiv=3D"Content-Type" content=3D"text/html; charset=3Dutf-8"></=
head>
<body>
<div dir=3D"auto">
<div dir=3D"auto">Here it is!</div><div dir=3D"auto"><br></div><div dir=3D"=
auto">Thank you </div><div dir=3D"auto"><br></div>
<div id=3D"aqm-original" style=3D"color: black;">

<div class=3D"aqm-original-body">
<div style=3D"color: black;">
<p style=3D"color: black; font-size: 10pt; font-family: sans-serif; margin:=
 8pt 0;">Le 21 juillet 2024 16:53:20 Ing ***@***.***&gt; a=
 =C3=A9crit :</p>
<blockquote type=3D"cite" class=3D"gmail_quote" style=3D"margin: 0 0 0 0.75=
ex; border-left: 1px solid #808080; padding-left: 0.75ex;">
<p></p>
<p dir=3D"auto">in 24.7.1<br>
RR shell</p>
<pre class=3D"notranslate"><code class=3D"notranslate">fbset -i
</code></pre>

<p style=3D"font-size:small;-webkit-text-size-adjust:none;color:#666;">=E2=
=80=94<br>
/rr/issues/1893#issuecomment-2241637751">view it on GitHub</a>, or <a href=
J3ZNPDN5AVCNFSM6AAAAABLF2YUOWVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDENBR=
 the thread.<img src=3D"https://github.com/notifications/beacon/AKHHW4LIDPK=
LUWAOSBCQT2TZNPDN5A5CNFSM6AAAAABLF2YUOWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23=
LFNZ2KUY3PNVWWK3TUL5UWJTUFTSWXO.gif" height=3D"1" width=3D"1" alt=3D""><spa=
n style=3D"color: transparent; font-size: 0; display: none; visibility: hid=
den; overflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-h=
751</span><span>@</span><span>github</span><span>.</span><span>com&gt;</spa=
n></span></p>

</blockquote>
</div>
</div>

</div><div dir=3D"auto"><br></div>
</div></body>
</html>

--190d62e03ff74cb28f3b6418a4--

--190d62e040074cb28f3bf8a60d
Content-Disposition: attachment; filename="IMG20240721184302.jpg"; modification-date="Sun, 21 Jul 2024 18:43:42 +0200"
Content-Type: image/jpeg; name="IMG20240721184302.jpg"
Content-Transfer-Encoding: base64

/9j/4AAQSkZJRgABAQAAAQABAAD/2wBDAAIBAQEBAQIBAQECAgICAgQDAgICAgUEBAMEBgUGBgYF
BgYGBwkIBgcJBwYGCAsICQoKCgoKBggLDAsKDAkKCgr/2wBDAQICAgICAgUDAwUKBwYHCgoKCgoK
CgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgr/wAARCAYACAADASIA
AhEBAxEB/8QAHwAAAQUBAQEBAQEAAAAAAAAAAAECAwQFBgcICQoL/8QAtRAAAgEDAwIEAwUFBAQA
AAF9AQIDAAQRBRIhMUEGE1FhByJxFDKBkaEII0KxwRVS0fAkM2JyggkKFhcYGRolJicoKSo0NTY3
ODk6Q0RFRkdISUpTVFVWV1hZWmNkZWZnaGlqc3R1dnd4eXqDhIWGh4iJipKTlJWWl5iZmqKjpKWm
p6ipqrKztLW2t7i5usLDxMXGx8jJytLT1NXW19jZ2uHi4+Tl5ufo6erx8vP09fb3+Pn6/8QAHwEA
AwEBAQEBAQEBAQAAAAAAAAECAwQFBgcICQoL/8QAtREAAgECBAQDBAcFBAQAAQJ3AAECAxEEBSEx
BhJBUQdhcRMiMoEIFEKRobHBCSMzUvAVYnLRChYkNOEl8RcYGRomJygpKjU2Nzg5OkNERUZHSElK
U1RVVldYWVpjZGVmZ2hpanN0dXZ3eHl6goOEhYaHiImKkpOUlZaXmJmaoqOkpaanqKmqsrO0tba3
uLm6wsPExcbHyMnK0tPU1dbX2Nna4uPk5ebn6Onq8vP09fb3+Pn6/9oADAMBAAIRAxEAPwDcXJXg
[... 781 lines cut ...]
rN6FJ2KGZBEIgpwy8qOB+FPjAOMjLBTzj7p9ae1nJETCYGTI/jXn3HrT4YPNB8gKGIKbs8Zxzz+l
QWrMijDk7kAPOME4z68d6kQKPutgkYBAz7d/r+lO+zDyxslEYAII/njNPEaLHteRRkbRk84oBXY1
G27WLA/3sjjNPjmngdVtm2kdwOpx70KFLKckf7JX+VP8vGXERB3dT24qXcdmCzyLcq5lbI5VepPq
K+9fhV/wXd+PPw2+HWjeBF+GOj30eh2EVmkouXRnjRQoYgA84HOK+CorYs8Tu5Kg54r0X9mr9nvx
9+0j8U7P4cfD/TpJru4b5mTIWNe7MR0X19azvoc+Jp0K1O1VXSPurSf+DiT4gW0bPrfwCtpgFIBt
dT2ncRx95eecV8QftQftU/F/9q34jT/EH4nai7fMRY6eo/dW0Z6KvPHbJA5NfoLL/wAG/Ef2O3lk
+KK+aYVMkb24wrY5wc5618jf8FEP+CfWtfsNa1pljeeKYtRt9atWmtZI0KlQpAYMTnnPStYQmo3X
U87Cf2VLEfubcx8ztMVkYrGpyCNu3HGOaatw0K4RccZDHkdP8/lX0d+wr/wTq+If7cdhq974Ovob
KLRSnm3MxBVnfJVVwPbk15d+0z+z54v/AGZPipqPwo8ZCI32nOFcxNuRcjK4OBxihRbV+x68a9F1
PZqXvdjgWnlaLax2sGyG/rUEk6qMso3BACw5z9fWh0c/MXbqCARjikZN7MR8q87ckZ+tM1uxqucF
s85ALZ4PFSIw8xY5JAVZjgsMYamGFiRkZ3Dg4xj6VHKzdAOMjkmgC01/ciRyL24TcMELMf8AGpW1
e+axaA6rebHXZJEbhirL7jOD/jVVpBLGw3LwM/M3GOOPrULrcToyuNw+6dvHHtTuxt3NnSPGnivw
zhdA8V6nYRbMuLO7eNSeOTtI6Gsm9mMskjPO8skjtJLJkncTyckkk/4mkkYQqqkMTjqO3bFQpuRC
EJAwcsO/PWlchvW4qM65BkYlhgNuwR7ioggyFR0bg5wfb9KfCjoN7PxtOVDdc0nkKwOz5QRjBXvj
P40Be7EQGZeAOCQ+3p1/z+tI5VAsnDgMNoAwMDilYysgLYz3GMYyOlKpiYMoyGXgjHUdev8A9agG
z2H9j/8AbZ+LH7FHjS+8Y/CQWcsurWQtr+3vosrLGDuGMdCDX03L/wAHAf7VKaTc2F74A0CZp7WS
ITFpBtyuNwAPPWvz/tlkknErlvmGFA7+vPtUj79zGFizhi+CeM00kclfB4XET5qkE2amt+M9W8S+
JZfF+qSyJe3Fy88kyfu/nZi3BXpyevXNfoL+zh/wX58afCv4T6X8Pfij4Bn8RXemW6wprH2sK88I
GF8zIwzAcZ74r84XnmYqs0gHOQo6D1xSSXEsbpHE/IORluG6/pVSpwnLmluXiMJQxdJQqq6R+teh
f8HHXwwEwudT+BWqptICm3miYnI7c18N/wDBQv8Ab58Tft2fFBvE8mhrpej6fAbfTLbdmXZnO6Qj
jPGMA4r528+KCFJRHweD8vI/z/WkG9gF4Ve4x1ParjGNNvl6mWHyzBYWp7SnGzDEzuJPLATPIU8n
2pqFwgVm6jAA53c+opVjuFZnt1zkjHAIx3p24B8J0IxgnHNJ7nY9z6j/AOCVX7cjfsW/tAf2/r9p
5/h3XoY9P1/s8UPmErMPUoxOR3FfrD40/bI/4JufG94pfF/xO8LajIijYbi5QkA9Mg8g1/P6zBf3
U0Yxv/jxjPof8aWOYBmKqik/MQGPNY+zipuS0ueVjcrpY2qqjk011R+/fxc/4KJfsWfsxfAie/8A
A/inR9Q

pkdick, 2024-07-22:
[image]

wjz304, 2024-07-22:
cat /proc/cmdline
fbset -i

pkdick, 2024-07-22:
[image]

wjz304, 2024-07-22:
test 24.7.3 and get them result

pkdick, 2024-07-22:
Please find the data that I get with RR 2.4.7.3: when I try to boot, it gets into a reboot loop...

[image]

pkdick, 2024-07-22:
Sorry I uploaded  the wrong photo
[image]

wjz304, 2024-07-22:
try boot，

wjz304, 2024-07-22:
is it block or poweroff?

wjz304, 2024-07-22:
try "Switch direct boot: true"

pkdick, 2024-07-22:
I guess that the screen gets first blank and then the PC reboots.

I will try to switch direct boot: true and will come back to you then to tell you the result

pkdick, 2024-07-22:
Hello,

It seems that the boot worked, but now I was asked to reinstall DSM and it fails at 57 %...

wjz304, 2024-07-22:
Did you rewrite the bootloader disk?

pkdick, 2024-07-22:
Yes

pkdick, 2024-07-22:
I tried again, and it apparently worked (no clue why ...). Do Ineed to keep the "direct boot" option ?

wjz304, 2024-07-22:
the upgrade RR does not need to be rewritten. Even if it is rewritten, it will normally appear "restored"
57% of the time, you can open another web page, https://<ip>:5000/webman/get_logs.cgi

> 你好
> 
> 似乎启动有效，但现在我被要求重新安装 DSM，它在 57 % 时失败......

wjz304, 2024-07-22:
> I tried again, and it apparently worked (no clue why ...). Do Ineed to keep the "direct boot" option ?

Keep it for now. Currently, there is no solution for Intel and AMD graphics cards.

pkdick, 2024-07-22:
Thank you very much for your support !

TheChief5055, 2024-07-24:
> Versions inferior 2.4.7.2 worked without any problem...

Same for me. Asrock J3455-ITX fw.1.80, DS918+ latest DSM7, RR version 24.7.1 works without any problem, with RR versions 24.7.2 and 24.7.3  DSM can't boot (no LAN).

Binhatgit, 2024-07-24:
Me too. Asrock J3455-ITX fw.1.80, DS920+ boot loop.
[16:15:20][   34.320433] ahci 0000:03:00.0: SSS flag set, parallel bus scan disabled
[16:15:20][   34.327459] ahci 0000:03:00.0: AHCI 0001.0200 32 slots 2 ports 6 Gbps 0x3 impl SATA mode
[16:15:20][   34.335992] ahci 0000:03:00.0: flags: 64bit ncq sntf stag led clo pmp pio slum part ccc 
[16:15:20][   34.344439] ahci 0000:03:00.0: Asmedia 1061 port=0, reg_addr=0xca4, reg_data=0xab
[16:15:20][   34.352273] ahci 0000:03:00.0: Asmedia 1061 port=1, reg_addr=0xda4, reg_data=0xab
[16:15:20][   34.360071] ahci 0000:03:00.0: Asmedia 1061 port=0, reg_addr=0xca5, reg_data=0x26
[16:15:20][   34.367855] ahci 0000:03:00.0: Asmedia 1061 port=1, reg_addr=0xda5, reg_data=0x26
[16:15:20][   34.375715] ahci 0000:03:00.0: Asmedia 1061 port=0, reg_addr=0xca6, reg_data=0x25
[16:15:20][   34.383553] ahci 0000:03:00.0: Asmedia 1061 port=1, reg_addr=0xda6, reg_data=0xd8
[16:15:20][   34.591479] Cannot find slot node of this ata_port.
[16:15:20][   34.596671] Cannot find slot node of this ata_port.
[16:15:20][   34.603631] scsi host2: ahci
[16:15:20][   34.609026] scsi host3: ahci
[16:15:20][   34.612208] ata3 : SATA max UDMA/133 abar m512@0x91100000 port 0x91100100 irq 126
[16:15:20][   34.620143] ata4 : SATA max UDMA/133 abar m512@0x91100000 port 0x91100180 irq 126
[16:15:20][   34.628030] ahci: probe of 0001:01:00.0 failed with error -22
[16:15:20][   34.634171] i8042: PNP: No PS/2 controller found. Probing ports directly.
[16:15:20][   34.641536] DMAR: DRHD: handling fault status reg 2
[16:15:20][   34.646679] DMAR: INTR-REMAP: Request device [[fa:1f.0] fault index d0b1
[16:15:20][   34.646679] INTR-REMAP:[fault reason 37] Blocked a compatibility format interrupt request
[16:15:20][   34.662468] DMAR: DRHD: handling fault status reg 2
[16:15:20][   34.667625] DMAR: INTR-REMAP: Request device [[fa:1f.0] fault index d0b1
[16:15:20][   34.667625] INTR-REMAP:[fault reason 37] Blocked a compatibility format interrupt request
[16:15:20][   34.684845] DMAR: DRHD: handling fault status reg 2

wjz304, 2024-07-25:
https://github.com/RROrg/rr-build/actions/runs/10087633704
test and report

TheChief5055, 2024-07-25:
> https://github.com/RROrg/rr-build/actions/runs/10087633704 test and report

Nah, DSM does not boot. Reverted to 24.7.1.

wjz304, 2024-07-25:
https://github.com/RROrg/rr-build/actions/runs/10092163415
test and report

TheChief5055, 2024-07-25:
Does not boot.

Model:    DS918+(apollolake)
Version:  7.2(69057u4)
Kernel:   official
LKM:      prod
DMI:      To Be Filled By O.E.M. To Be Filled By O.E.M./J3355B-ITX, BIOS P1.90 05/21/2018
CPU:      Intel(R) Celeron(R) CPU J3355 @ 2.00GHz
MEM:      7864 MB
Cmdline:
withefi **earlyprintk kvm.ignore_msrs]**="1" **CMDLINE[kvm.report_ignored_msrs** syno_hw_version=DS918+ console=ttyS0,115200n8 consoleblank=600 netif_num=2 pid=0x6387 earlycon=uart8250,io,0x3f8,115200n8 mac2=001132XXXXXX mac1=001132XXXXXX sn=1910PDNXXXXXX modprobe.blacklist=evbug,cdc_ether rootwait SMBusHddDynamicPower=1 vid=0x058f elevator=elevator loglevel=15 HddHotplug=1 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 intel_iommu=igfx_off syno_hdd_detect=0 panic=5 vender_format_version=2 syno_hdd_powerup_seq=0 log_buf_len=32M root=/dev/md0

Why cmdline is so strange?

wjz304, 2024-07-25:
> Does not boot.
>
> Model: DS918+(apollolake) Version: 7.2(69057u4) Kernel: official LKM: prod DMI: To Be Filled By O.E.M. To Be Filled By O.E.M./J3355B-ITX, BIOS P1.90 05/21/2018 CPU: Intel(R) [...]

There is a typo, but it doesn't matter.

wjz304, 2024-07-25:
> Does not boot.

I don't know, I can't guess why.

TheChief5055, 2024-07-25:
It works like 24.7.3 — only in direct boot mode.

wjz304, 2024-07-25:
> It works like 24.7.3 — only in direct boot mode.

are you r8125 ?

laomao9000, 2024-07-26:
> > It works like 24.7.3 — only in direct boot mode.
> 
> are you r8125 ?
@TheChief5055 's mainboard  is Asrock J3455-ITX fw.1.80，the nic is Realtek RTL8111GR，use  diver same as  r8168.

TheChief5055, 2024-07-26:
> > It works like 24.7.3 — only in direct boot mode.
> 
> are you r8125 ?

I'm testing it both on J3455-ITX and J3355B-ITX, same RTL8111GR NIC.

wjz304, 2024-07-26:
disable rss of all nic
https://github.com/RROrg/rr-build/actions/runs/10105619513
test and report

wjz304, 2024-08-09:
test https://github.com/RROrg/rr-build/actions/runs/10319977732

wjz304, 2024-10-18:
test v24.10.1

wjz304, 2024-10-18:
https://github.com/RROrg/rr/issues/2387

wjz304, 2024-10-18:
https://github.com/RROrg/rr/issues/3623

wjz304, 2024-10-18:
https://github.com/RROrg/rr/issues/2226

wjz304, 2024-10-18:
https://github.com/RROrg/rr/issues/2384

wjz304, 2024-11-21:
https://github.com/RROrg/rr/releases/tag/24.11.11

closed 2024-11-21 by wjz304