RROrg/rr #1952: BUG : 10G NIC only connect 1G.

azchoi, 2024-07-25:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Native
 * CPU:  Intel Xeon E-2236
 * NIC:  (pid & vid) Intel X520 (10G) PCI-E

RR version: (You can find it in the update menu.)  
 * RR:  24.7.3
 * addons:  All addon without NVME, codec patch & wireless
 * modules:  select loaded modules
 * lkms:  

DSM: 
 * model:  DS3622xs+
 * version:  7.2.1-U5
Issue:

1. Intel X520 (10G) PCI-E NIC only works 1G after 24.7.3.
2. 24.7.2 and less version works 10G.

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-07-26:
https://github.com/RROrg/rr-build/actions/runs/10105619513
test and report

azchoi, 2024-07-26:
> https://github.com/RROrg/rr-build/actions/runs/10105619513 test and report

1. Same symptoms after update.
2. RR config mode is connected to 10G, but in DSM, it is connected to 1G only.

azchoi, 2024-07-26:
> https://github.com/RROrg/rr-build/actions/runs/10105619513 test and report

I tried to delete sortnetif addon and un-select intel 1g driver nic module (only select ixgbe).
and then issue have been cleared.

I think this issue is related to "sortnetif" addon.

wjz304, 2024-07-26:
https://github.com/RROrg/rr-build/actions/runs/10108579675
test with sortnetif

azchoi, 2024-07-26:
> https://github.com/RROrg/rr-build/actions/runs/10108579675 test with sortnetif

Sorry for worng information.

I've tested serveral times, it seems that it's not related to sortnetif addon. I've disabled "sortnetif" and then it worked 10G. But I tested it once again, it shows 1G connections again.

But RR config mode is always connect to 10G, but after DSM, it was reconnected to 1G.

So I've force set to 10G on my router, it was connected to 10G everytimes. I think that this issue is related to Auto Negotiation after DSM booting.

I always thanks for your support.

wjz304, 2024-07-27:
The driver file has not changed recently, and sortnetif has not changed recently
sortnetif is just ethX down, rename, up
It should be related to auto-negotiation, which can be set through ethtool, but I don't know why it changed

wjz304, 2024-07-27:
24.7.4 Added a process to uninstall all network card drivers before RR switches to DSM to reduce memory impact

azchoi, 2024-07-27:
> 24.7.4 Added a process to uninstall all network card drivers before RR switches to DSM to reduce memory impact

Well noted! Thank you!

azchoi, 2024-07-27:
> 24.7.4 Added a process to uninstall all network card drivers before RR switches to DSM to reduce memory impact

I set my switch (Unifi enterprise 10G switch) to auto-negotiation mode again and
I tried to update to 24.7.4 (pre) with sortnetif addon and igb+ixgbe modules, my intel 10G PCI-E NIC connect to 10G succesfully.

[image]

Thank you so much!!! Have a nice weekend!!!

closed 2024-08-03 by wjz304