RROrg/rr #1970: RAID1总是掉盘

Erignik, 2024-07-28:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  i5 12500T
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.6.9
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2

Issue:  RAID1总是掉盘，换了一个新的硬盘还是如此
[image]

logs:  如需要日志，我这边去采集

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-08-03:
请检查磁盘温度 或者提供 /var/log/*

wjz304, 2024-10-24:
无进一步反馈，该issue关闭

closed 2024-10-24 by wjz304