RROrg/rr #2144: Not so much a bug - using the loader I somehow changed the language settings from En-US to a language I don't comprehend.  I can't revert the loader presentation back to En-US

Rhubarb1, 2024-08-03:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  24.7.3
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-08-05:
`rm /mnt/p1/.locale`  back to en-us

wjz304, 2024-08-05:
This problem has been considered, but there is no perfect solution.

wjz304, 2024-08-08:
24.8.1

closed 2024-08-08 by wjz304