RROrg/rr #2149: BUG

tcg123, 2024-08-04:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  见下图
 * CPU:  见下图
 * NIC:  (pid & vid)
[image]

RR version: (You can find it in the update menu.)  
 * RR:  [24.8.0](https://github.com/RROrg/rr/releases/tag/24.8.0)
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  DSM_SA6400_69057

Issue:  
pve8.2.2下创建虚拟机，在boot loader后，在控制台窗口一直提示如下的error信息（安装完群晖依然存在），不知道是不是虚拟机设置的问题，还是24.8.0这个版本引导的问题，群晖功能使用正常.

logs:  
`[ 2763.439223] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2763.445592] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2777.830393] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2777.832186] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2788.015119] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2788.021385] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2802.235224] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2802.237019] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2812.592065] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2812.598408] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2827.007575] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.
[ 2827.009292] i915 0000:06:10.0: [drm] *ERROR* VBT claims to have both internal and external displays on PHY A.  Configuring for internal.`
[image]

[
(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    ](https://github.com/RROrg/rr/issues)
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-08-05:
Check the BIOS setup, check the display output configuration.
or
try add "i915.enable_psr=0" to cmdline

tcg123, 2024-08-05:
> Check the BIOS setup, check the display output configuration. or try add "i915.enable_psr=0" to cmdline

@wjz304 感谢回复，add "i915.enable_psr=0" to cmdline，加完这个 rebuild loader后还是有error，如下截图所示，
[image]
错误一样，似乎不自动换行了。

moetayuko, 2024-08-10:
没有已知的负面影响，建议忽略。强迫症的话可以按照 https://pve.proxmox.com/wiki/PCI_Passthrough#The_'romfile'_option 导出 vbios 并设置 romfile，应该可以解决这个报错（不过我没试过

closed 2024-08-10 by moetayuko