RROrg/rr #2180: VirGL-GPU无法使用

wxxsfxyzm, 2024-08-05:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
[image]

 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:   24.8.0
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2

Issue:  在pve中安装的群晖虚拟机，因为看到安装Linux虚拟机的时候可以使用virgl-gpu使用宿主机gpu提供运算能力
[image]
然而在群晖中这个选项似乎没有起到作用，ssh进去ls /dev/dri也是没有设备，这边怀疑是因为没有驱动？所以发一个issue来询问一下，如果有希望能搞定，我可以帮忙测试

logs:  如有需要我可以去收集

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-08-05:
目前无法支持

wxxsfxyzm, 2024-08-05:
好的，感谢作者百忙之中的回复！

closed 2024-08-05 by wjz304

233cy, 2024-12-17:
有大佬能解释一下，这个VirGL GPU吗，我目前安装群晖没有/dev/dri，但是安装飞牛用VirGL GPU是有/dev/dri驱动的，但是飞牛OS并不会用这个驱动进行解码是为啥。一直没搞懂这个VirGL GPU。

wxxsfxyzm, 2024-12-18:
> 有大佬能解释一下，这个VirGL GPU吗，我目前安装群晖没有/dev/dri，但是安装飞牛用VirGL GPU是有/dev/dri驱动的，但是飞牛OS并不会用这个驱动进行解码是为啥。一直没搞懂这个VirGL GPU。

virgl使用宿主的LibEGL和libGL，在主机上调用gpu进行渲染
具体细节可以查看链接https://docs.mesa3d.org/drivers/virgl.html
他不是gpu直通，他只能支持openGL来进行3D加速，编解码器不在他的支持范围内，Linux的编解码加速走的是VAAPI
至于你为什么飞牛有很简单，因为飞牛是基于debian的，主流的发行版内核都会包含对virgl的驱动支持，例如archlinux，debian等，而一些冷门的发行版则没有相关支持，比如群晖，unraid所使用的slackware等

wxxsfxyzm, 2025-03-07:
> 有大佬能解释一下，这个VirGL GPU吗，我目前安装群晖没有/dev/dri，但是安装飞牛用VirGL GPU是有/dev/dri驱动的，但是飞牛OS并不会用这个驱动进行解码是为啥。一直没搞懂这个VirGL GPU。

因为qemu默认不会传递vaapi的解码器，网上有教程，但是需要patch qemu

sakurawill, 2025-12-26:
> > 有大佬能解释一下，这个VirGL GPU吗，我目前安装群晖没有/dev/dri，但是安装飞牛用VirGL GPU是有/dev/dri驱动的，但是飞牛OS并不会用这个驱动进行解码是为啥。一直没搞懂这个VirGL GPU。
> 
> 因为qemu默认不会传递vaapi的解码器，网上有教程，但是需要patch qemu
@wxxsfxyzm 请问下大大，有教程的链接吗？想在飞牛上去使用virgl