RROrg/rr #2189: [BUG] 编译DVA3221，扫不到任何设备

WenisOK, 2024-08-06:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  QEMU Standrad PC (Q35 + ICH9, 2009), BIOS rel-1.16.3-0-ga6ed6b701f0a-prebuilt.qemu.org 04/01/2014
 * CPU:  Intel(R) Xeon(R) CPU E5-2686 v4 @ 2.30GHz
 * NIC:  virtio_net

RR version: (You can find it in the update menu.)  
 * RR:  24.8.0
 * addons:  default
 * modules:  default
 * lkms:  default

DSM: 
 * model:  DVA3221
 * version:  7.2(69057)

Issue:  
编译DVA3221后引导，扫不到任何设备

logs:  
Serial TTy
...
Please press Enter to activate this console. Stopping ttyd
OK
Stopping dufs
OK
Stopping thttpd: OK
Stopping HPA's tftpd: done
Stopping sshd: OK
Stopping ntpd: OK
Stopping ntpd: FAIL
Stopping dhcpcd...
OK
Stopping network: OK
Stopping vmtoolsd: FAIL
Stopping QEMU Guest Agent
FAIL
Stopping klogd: OK
Stopping acpid: OK
Stopping syslogd: OK
Seeding 256 bits without crediting
Saving 256 bits of creditable seed for next boot
umount: /: not mounted.

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-08-09:
test https://github.com/RROrg/rr-build/actions/runs/10319977732

WenisOK, 2024-08-09:
cheked out

closed 2024-08-09 by WenisOK