RROrg/rr #2225: [Bug] RTL8111F网卡无法 DHCP，J1900花屏。

mmj-cn, 2024-08-07:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  花屏无法查看
 * CPU:  J1900
 * NIC:  花屏无法查看

RR version: (You can find it in the update menu.)  
 * RR:  24.8.0
 * addons:  花屏无法查看，未配置任何 addons，除了强制安装的。
 * modules:  花屏无法查看
 * lkms:  花屏无法查看

DSM: 
 * model:  918+
 * version:  7.21 69057

Issue:  j1900 rtl8111f 网卡，rr引导 24.8.0，貌似无法 dhcp，路由器显示无设备网线连接。
j1900 花屏，看不到控制台，只能看出来占据字符位置的白色方块和下划线。
把u盘拆到自己电脑上设置静态 ip，装回去重启后，路由器可以显示ip，但依然无法访问 ip:7681 完成引导配置。
继续拆到自己电脑上配置完成，编译一次后，插上 j1900 主机，一开始访问配置的静态 ip，可以进入 7681 控制面板，重启后等待引导完成，屏幕显示内容不变化 5 分钟后访问，路由器显示有 ip（新的 ip），但 http5000，https5001 端口均无法访问。

logs: 花屏无法查看

wjz304, 2024-08-09:
test https://github.com/RROrg/rr-build/actions/runs/10319977732

q331723516, 2024-08-16:
你的RTL8111测试正常了吗，我的最新版rr也是dhcp获取不到

wjz304, 2024-09-11:
无人反馈，本issue关闭

closed 2024-09-11 by wjz304