RROrg/rr #2227: BUG

evenwangcn, 2024-08-07:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  N100
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.8.1
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA600
 * version:  

Issue:  用nvme固态硬盘做引导，DSM和config loader都启动不了，提示Loader disk neither USB or SATA/SCSI/NUME/MMC/ IDE/VMBUS/XEN DoM，还原到24.7.4之后正常启动。

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-08-07:
RR shell
`lsblk -dpno KNAME,TRAN,SUBSYSTEMS`

TimKraemer, 2024-08-07:
[image]

what is the problem here, how to solve it? I basically only updated rr, it worked before

evenwangcn, 2024-08-07:
The same issue

wjz304, 2024-08-08:
> ![image](https://private-user-images.githubusercontent.com/5167505/355854190-7054af71-0227-442f-94f0-12d215796b93.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

```
echo ${LOADER_DISK}
udevadm info --query property --name ${LOADER_DISK}
```

wjz304, 2024-08-08:
What is your environment? What type of disk is it?

wjz304, 2024-08-08:
https://github.com/RROrg/rr-build/actions/runs/10295526787

evenwangcn, 2024-08-08:
ESXi中虚拟机来引导，硬盘直通

evenwangcn, 2024-08-08:
> https://github.com/RROrg/rr-build/actions/runs/10295526787 test

> https://github.com/RROrg/rr-build/actions/runs/10295526787 test

After testing, everything is working fine now. The issue has been resolved.

TimKraemer, 2024-08-08:
> ```
> echo ${LOADER_DISK}
> udevadm info --query property --name ${LOADER_DISK}
> ```

[image]

it's running in proxmox
[image]

TimKraemer, 2024-08-08:
> > [RROrg/rr-build/actions/runs/10295526787](https://github.com/RROrg/rr-build/actions/runs/10295526787) test
> 
> After testing, everything is working fine now. The issue has been resolved.

what exactly did you do to fix it? How can I apply this new image without loosing my settings?

wjz304, 2024-08-08:
> > ```
> > echo ${LOADER_DISK}
> > udevadm info --query property --name ${LOADER_DISK}
> > ```
> 
> ![图像](https://private-user-images.githubusercontent.com/5167505/356170901-d8b3b29c-572f-4da4-ad9d-a31e0ff32328.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMxMTEyNDUsIm5iZiI6MTcyMzExMDk0NSwicGF0aCI6Ii81MTY3NTA1LzM1NjE3MDkwMS1kOGIzYjI5Yy01NzJmLTRkYTQtYWQ5ZC1hMzFlMGZmMzIzMjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MDhUMDk1NTQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWFmY2E3YzM4MzM5MzljNzA0NTRlNWNhMmVmMmZhMzFlZmU0MTA5OTYyNjMxYzk1OGY5YjhjNDMwNTZiYTg4MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.mSRelp2GDZB5w46NXqv1MC2dkIuGacB0CYkW8qGEL_E)
> 
> 它在 Proxmox 中运行 ![图像](https://private-user-images.githubusercontent.com/5167505/356171108-372b9e2c-8693-42dd-9b25-47768acf391f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMxMTEyNDUsIm5iZiI6MTcyMzExMDk0NSwicGF0aCI6Ii81MTY3NTA1LzM1NjE3MTEwOC0zNzJiOWUyYy04NjkzLTQyZGQtOWIyNS00Nzc2OGFjZjM5MWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MDhUMDk1NTQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MmIwNDRmNmRmN2EzMDc4N2M5Yzg2NDU0ODBkZjUwNTkyZjA1NmI1NDA1ZjYwOTkzY2MwYTgxMWVmZjc5YWMyNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.JQIZgw8ZNOmuxQL4tvdJHrzhgwEPrLA6ZOG_5b6sYfg)

`lsblk -dpno KNAME,TRAN,,SUBSYSTEMS`

wjz304, 2024-08-08:
```
sed -i 's/vmbus/vmbus virtio sas/' /opt/rr/init.sh
menu.sh
```

TimKraemer, 2024-08-08:
> ```
> sed -i 's/vmbus/vmbus virtio sas/' /opt/rr/init.sh
> menu.sh
> ```

this worked for me 👍🏻

wjz304, 2024-08-24:
fixed

closed 2024-08-24 by wjz304