RROrg/rr #2248: BUG

wahabe, 2024-08-07:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  i7-8700K
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.7.4 / 24.8.1
 * addons:  default
 * modules:  default
 * lkms:  default

DSM: SA6400 /DS920 /SA6400
 * model:  
 * version:  

Issue:  
Boot interface cannot be found
The problem started from RR 24.7.4 to 24.8.1
Just go to settings BIOS To save and exit 
Everything works normally
I tested three different computers
logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wahabe, 2024-08-08:
I will test the version rr-202408081137

wjz304, 2024-08-08:
No, except for the boot type determination, it is no different from 24.8.1.

wahabe, 2024-08-08:
This is yesterday's report. Delete the file after downloading.

[dbgutils logs.tar.gz](https://github.com/user-attachments/files/16548365/dbgutils.logs.tar.gz)

[dmesg log.txt](https://github.com/user-attachments/files/16548368/dmesg.log.txt)

wahabe, 2024-08-08:
The error is confusing.
Just go to settings BIOS To save and exit
Everything works normally

wahabe, 2024-08-08:
Am I the only one facing this problem?

There is no problem with the version rr-24.7.3

I will test on another device.

wjz304, 2024-08-08:
sa6400 custom kernel ?

wahabe, 2024-08-08:
no

wjz304, 2024-08-08:
I'll research it.

wahabe, 2024-08-08:
Uploading ASUS 2024-08-08 175631.mp4…

wahabe, 2024-08-08:
https://github.com/user-attachments/assets/27389e5f-a874-4d27-a8f9-dc0e50079106

wahabe, 2024-08-08:
https://github.com/user-attachments/assets/acae6aa0-c359-4df0-8455-2ba262b8fa71

wahabe, 2024-08-08:
The problem started from RR 24.7.4 to 24.8.1

wahabe, 2024-08-08:
I have another question, is it possible to modify the actual processor temperature reading?

wahabe, 2024-08-08:
[image]
[image]

wjz304, 2024-08-08:
> I have another question, is it possible to modify the actual processor temperature reading?

https://github.com/GroverLau/syno_cpuinfo

wahabe, 2024-08-08:
Thank you, this is what I am looking for.

I am currently using Grafana and Prometheus Dashboard.

wahabe, 2024-08-08:
https://mariushosting.com/monitor-your-synology-with-grafana-and-prometheus-dashboard/

wjz304, 2024-08-09:
test  https://github.com/RROrg/rr-build/actions/runs/10319977732

wahabe, 2024-08-09:
ok

wahabe, 2024-08-09:
https://github.com/user-attachments/assets/902253c8-dc06-4b1e-9f4a-4e2d1b5edf5a

very fast

wahabe, 2024-08-09:
Do you need to send a report?

wjz304, 2024-08-09:
> Do you need to send a report?

No need, the OK log is meaningless

wahabe, 2024-08-09:
If you need any test, I am ready

wjz304, 2024-08-09:
There is no other test for the time being, so let's release this version first.

wjz304, 2024-08-09:
https://github.com/RROrg/rr/releases/tag/24.8.2

wahabe, 2024-08-09:
Thank you, I really appreciate your effort. Great work
I am currently using a processor.
AMD Ryzen 9 3900X 12-Core  24-Threads + SA6400
I want to move to DS1823xs+
Does the processor support?

wjz304, 2024-08-09:
> Thank you, I really appreciate your effort. Great work I am currently using a processor. AMD Ryzen 9 3900X 12-Core 24-Threads + SA6400 I want to move to DS1823xs+ Does the processor support?

I'm just an ordinary person, I don't have a lot of hardware to test, 
Can only wait for user feedback

wahabe, 2024-08-09:
I am also a normal person, a retired marine, injured

wahabe, 2024-08-09:
I will try it tomorrow

wahabe, 2024-08-09:
Thank you my friend

closed 2024-08-14 by wjz304