RROrg/rr #2283: BUG 从24.7.2及以上版本华擎 j3455 就无法加载到群辉中降级回24.7.1可以进入

zcr268, 2024-08-10:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:   j3455
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.7.2
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS918+
 * version:  7.2.1

Issue:  
    从24.7.2及以上版本华擎 j3455 就无法加载到群辉中降级回24.7.1可以进入 
logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-08-14:
无进一步反馈，无法调查，该issues关闭

closed 2024-08-14 by wjz304

wjz304, 2024-10-24:
https://github.com/RROrg/rr/issues/1893 跟踪