RROrg/rr #2287: Shutdown(Power off) issues

matapod, 2024-08-11:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  To be filled by O.E.M. To be filled by O.E.M./Aptio CRB, BIOS 5.6.5 11/24/2018
 * CPU:  J1900
 * NIC:  82583V

RR version: (You can find it in the update menu.)  
 * RR:  24.8.2
 * addons:  default
 * modules:  default
 * lkms:  prod

DSM: 
 * model:  DS3622xs+
 * version:  7.2

Issue:  When tried to power off, it just rebooted.

logs:  
root@DiskStation:/var/log# cat synopoweroff.log
2024-08-11T12:04:59+09:00 DiskStation synoscgi_SYNO.Core.System_2_shutdown[412]: system_poweroff_pre_hook.c:22 synopoweroff: Begin SYNOPowerOffPreHook action.
2024-08-11T12:05:00+09:00 DiskStation synoscgi_SYNO.Core.System_2_shutdown[412]: system_poweroff_pre_hook.c:27 synopoweroff: Finish SYNOPowerOffPreHook action.
2024-08-11T12:05:00+09:00 DiskStation synoscgi_SYNO.Core.System_2_shutdown[412]: system_sys_init.c:133 synopoweroff: System is going to [poweroff]

wjz304, 2024-08-14:
`dmesg | grep "\[ cut here \]" -A 30`

matapod, 2024-08-15:
> `dmesg | grep "\[ cut here \]" -A 30`

Should I remove something in command???

root@DiskStation:~# dmesg | grep "\[ cut here \]" -A 30

  ->> nothing appeard.

wjz304, 2024-08-15:
Search panic related

wjz304, 2024-08-15:
Some kernel panics prevent safe shutdown

matapod, 2024-08-15:
> Some kernel panics prevent safe shutdown

without slash(\)   dmesg | grep "[ cut here ]" -A 30

[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.302+ (root@build8) (gcc version 12.2.0 (GCC) ) #69057 SMP Fri Jan 12 17:02:26 CST 2024
[    0.000000] Command line: withefi earlyprintk syno_hw_version=DS3622xs+ console=ttyS0,115200n8 consoleblank=600 netif_num=2 pid=0x5591 earlycon=uart8250,io,0x3f8,115200n8 mac2=001132599e65 mac1=001132599e64 sn=2150SQRESF4NH modprobe.blacklist=evbug,cdc_ether pcie_aspm=off rootwait SMBusHddDynamicPower=1 vid=0x0781 elevator=elevator loglevel=15 HddHotplug=1 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 syno_hdd_detect=0 panic=5 vender_format_version=2 syno_hdd_powerup_seq=0 log_buf_len=32M root=/dev/md0
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] x86/fpu: Legacy x87 FPU detected.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000003efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000003f000-0x000000000003ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000040000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000001f000000-0x000000001f0fffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000001f100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000200fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020100000-0x00000000b93fbfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b93fc000-0x00000000b942bfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b942c000-0x00000000b943bfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000b943c000-0x00000000b9939fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000b993a000-0x00000000b9bbbfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b9bbc000-0x00000000b9c0cfff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000b9c0d000-0x00000000b9c0dfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b9c0e000-0x00000000b9c4ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b9c50000-0x00000000b9dbffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b9dc0000-0x00000000b9ff9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b9ffa000-0x00000000b9ffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] earlycon: Early serial console at I/O port 0x3f8 (options '115200n8')
[    0.000000] bootconsole [uart0] enabled
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x00038450-0x0003847f] usable ==> usable
[    0.000000] e820: update [mem 0x000383e0-0x0003844f] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x00000000000383df] usable
[    0.000000] reserve setup_data: [mem 0x00000000000383e0-0x000000000003847f] usable
[    0.000000] reserve setup_data: [mem 0x0000000000038480-0x000000000003efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000003f000-0x000000000003ffff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000000040000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000001effffff] usable
[    0.000000] reserve setup_data: [mem 0x000000001f000000-0x000000001f0fffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000001f100000-0x000000001fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000020000000-0x00000000200fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000020100000-0x00000000b93fbfff] usable
[    0.000000] reserve setup_data: [mem 0x00000000b93fc000-0x00000000b942bfff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b942c000-0x00000000b943bfff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000b943c000-0x00000000b9939fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000b993a000-0x00000000b9bbbfff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b9bbc000-0x00000000b9c0cfff] type 20
[    0.000000] reserve setup_data: [mem 0x00000000b9c0d000-0x00000000b9c0dfff] usable
[    0.000000] reserve setup_data: [mem 0x00000000b9c0e000-0x00000000b9c4ffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b9c50000-0x00000000b9dbffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000b9dc0000-0x00000000b9ff9fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b9ffa000-0x00000000b9ffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ffb00000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] efi: EFI v2.40 by American Megatrends
[... cut ...]
[   97.422948] audit: type=1325 audit(1723690785.767:302): table=filter family=2 entries=51
[   97.431053] audit: type=1325 audit(1723690785.775:303): table=filter family=10 entries=37
[   97.439426] audit: type=1325 audit(1723690785.783:304): table=filter family=2 entries=46
[   97.447742] audit: type=1325 audit(1723690785.791:305): table=filter family=10 entries=26
[   99.453112] docker0: port 1(docker3a7b80a) entered forwarding state
[   99.581294] docker0: port 2(docker66eaf2b) entered forwarding state

wjz304, 2024-08-15:
Check the bios incoming call status to see if it is powered on by incoming call

matapod, 2024-08-15:
> Check the bios incoming call status to see if it is powered on by incoming call

It works fine with older version

closed 2024-08-17 by matapod