RROrg/rr #2302: BUG

martin0359, 2024-08-13:
martin0359@MMSNAS:~$ dmesg | grep i915
[ 4.107774] i915 0000:00:02.0: [drm] VT-d active for gfx access
[ 4.116278] i915 0000:00:02.0: [drm] Transparent Hugepage support is recommended for optimal performance on this platform!
[ 4.118469] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[ 4.122214] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[ 4.143729] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.29.2
[ 4.144834] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[ 4.149702] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[ 4.150985] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[ 4.151790] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[ 4.152945] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[ 4.153699] i915 0000:00:02.0: [drm] ERROR GT0: GUC: mmio request 0x4100: failure 201/0
[ 4.155770] i915 0000:00:02.0: [drm] ERROR GT0: Failed to retrieve hwconfig table: -ENOENT
[ 4.251930] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[ 4.253073] i915 display info: display version: 13
[ 4.253719] i915 display info: cursor_needs_physical: no
[ 4.254442] i915 display info: has_cdclk_crawl: yes
[ 4.255076] i915 display info: has_cdclk_squash: no
[ 4.255734] i915 display info: has_ddi: yes
[ 4.256303] i915 display info: has_dp_mst: yes
[ 4.256883] i915 display info: has_dsb: yes
[ 4.257456] i915 display info: has_fpga_dbg: yes
[ 4.258057] i915 display info: has_gmch: no
[ 4.258632] i915 display info: has_hotplug: yes
[ 4.259240] i915 display info: has_hti: no
[ 4.259784] i915 display info: has_ipc: yes
[ 4.260358] i915 display info: has_overlay: no
[ 4.260942] i915 display info: has_psr: yes
[ 4.261521] i915 display info: has_psr_hw_tracking: no
[ 4.262207] i915 display info: overlay_needs_physical: no
[ 4.262916] i915 display info: supports_tv: no
[ 4.263529] i915 display info: has_hdcp: yes
[ 4.264092] i915 display info: has_dmc: yes
[ 4.264662] i915 display info: has_dsc: yes

型号: SA6400(epyc7002)
版本: 7.2(69057)
内核: official
LKM: prod
DMI: Default string Default string/NY12H, BIOS NY12H.102 02/28/2024
CPU: Genuine Intel(R) 0000
MEM: 31848 MB
Cmdline:
withefi earlyprintk syno_hw_version=SA6400 console=ttyS0,115200n8 consoleblank=600 netif_num=2 pid=0x5571 earlycon=uart8250,io,0x3f8,115200n8 mac2=9009D02A95D4 mac1=9009D02A95D3 sn=227BBUMR8HWA07 modprobe.blacklist=evbug,cdc_ether pcie_aspm=off rootwait vid=0x0781 loglevel=15 HddHotplug=1 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 panic=5 vender_format_version=2 log_buf_len=32M root=/dev/md0 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8

CPU是1270p板u，核显没问题，因为另一台win系统的小主机也是这个板u，jellyfin原来可以用，后面突然不能转码了。今天装了up5，套件版jellyfin又好了，但是av1编码不支持，容器版的jellyfin，一只是h264编码，开启hevc和av1编码无用。另外看到i915的最新驱动更新了，小白想问问，做引导的时候，怎么编译这个驱动进去。

wjz304, 2024-08-14:
你先编个 hello world，证明 我说了你能看懂？

martin0359, 2024-08-15:
> 你先编个 hello world，证明 我说了你能看懂？

#include <iostream>

int main() {
    std::cout << "Hello, World!" << std::endl;
    return 0;
}
你说吧，哥

martin0359, 2024-08-15:
[ 4.153699] i915 0000:00:02.0: [drm] ERROR GT0: GUC: mmio request 0x4100: failure 201/0
[ 4.155770] i915 0000:00:02.0: [drm] ERROR GT0: Failed to retrieve hwconfig table: -ENOENT 现在能转码了，但这条错误一直存在

wjz304, 2024-08-15:
一. “看到i915的最新驱动更新了” 不知道指哪个？
1. 如果是官方驱动，先移植兼容 5.10.55 内核， 可以参考 https://github.com/moetayuko/intel-gpu-i915-backports 的修改
2. 如果不是，不知道...

二. 搭建 SYNO 的编译环境，
方法1. 基于官方 toolchain，参考 https://github.com/moetayuko/intel-gpu-i915-backports/blob/mainline-tracking-v6.5/.github/workflows/build.yml
方法2. 基于官方 pkgscripts， 参考 https://github.com/SynologyOpenSource/pkgscripts-ng.git
方法3. 基于第三方 dockers，参考 https://github.com/SynoCommunity/spksrc.git

三. 编译。略（同 编译 hello world）

四, 将 ko 放入 系统 ， insmod <path>/i915

martin0359, 2024-08-15:
> 一. “看到i915的最新驱动更新了” 不知道指哪个？
>
> 1. 如果是官方驱动，先移植兼容 5.10.55 内核， 可以参考 https://github.com/moetayuko/intel-gpu-i915-backports 的修改
>
> 2. 如果不是，不知道...
>
>
>
> 二. 搭建 SYNO 的编译环境，
>
> 方法1. 基于官方 toolchain，参考 [...]

谢谢大佬，我看的是这个项目里有一个sa6400的驱动，编译好像做完了，有module和firmware文件：https://github.com/moetayuko/intel-gpu-i915-backports

wjz304, 2024-08-15:
我就知道 写了 也没多大意义

wjz304, 2024-08-16:
https://github.com/moetayuko/intel-gpu-i915-backports 上的修改，是我PR的，没有功能上的修改，不需要更新。

closed 2024-08-16 by wjz304

martin0359, 2024-08-19:
> https://github.com/moetayuko/intel-gpu-i915-backports 上的修改，是我PR的，没有功能上的修改，不需要更新。

好的，硬更了一次，失灵了，又给恢复了，谢谢提醒！

sacredx72, 2025-03-08:
@martin0359 
Hello, were you able to figure out and fix this problem?

> [ 4.153699] i915 0000:00:02.0: [drm] ERROR GT0: GUC: mmio request 0x4100: failure 201/0
> [ 4.155770] i915 0000:00:02.0: [drm] ERROR GT0: Failed to retrieve hwconfig table: -ENOENT 现在能转码了，但这条错误一直存在