RROrg/rr #2364: BUG: Install RR to EMMC  cause AME activation failed.

zengdabo, 2024-08-16:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  To be filled by O.E.M. To be filled by O.E.M./To be filled by O.E.M., BIOS 1.1 09/02/2016
 * CPU:  Intel(R) Celeron(R) CPU N3160 @ 1.60GHz
 * NIC:  (pid & vid)
eth0(igb): Access http://192.168.10.220:5000 to connect the DSM via web.
eth1(igb): NOT CONNECTED

RR version: (You can find it in the update menu.)  
 * RR:  24.8.2
 * addons:  
   acpid: ''
  mountloader: ''
  powersched: ''
  reboottoloader: ''
  cpuinfo: ''
  hdddb: ''
  sensors: ""
 * modules:  
     8139cp: ""
  9p: ""
  9pnet: ""
  9pnet_virtio: ""
  aacraid: ""
  acpi-cpufreq: ""
  acpi_call: ""
  adm1021: ""
  adm1031: ""
  adm9240: ""
  adt7470: ""
  adt7475: ""
  aic94xx: ""
  alx: ""
  amd_nb: ""
  apex: ""
  aqc111: ""
  asix: ""
  ata_generic: ""
  ata_piix: ""
  atkbd: ""
  atl1c: ""
  atl1e: ""
  atlantic: ""
  auxiliary: ""
  ax88179_178a: ""
  b44: ""
  backlight: ""
  bcm-phy-lib: ""
  be2net: ""
  bitblit: ""
  bnx2: ""
  bnx2x: ""
  bnx2x_mod: ""
  bnxt_en: ""
  broadcom: ""
  btcoexist: ""
  button: ""
  cdc_ether: ""
  cdc_ncm: ""
  cdrom: ""
  cfbcopyarea: ""
  cfbfillrect: ""
  cfbimgblt: ""
  cfg80211: ""
  ch: ""
  ch341: ""
  cls_fw: ""
  cls_u32: ""
  cn: ""
  cp210x: ""
  cpufreq_conservative: ""
  cpufreq_governor: ""
  cpufreq_ondemand: ""
  cpufreq_performance: ""
  cpufreq_powersave: ""
  cpufreq_stats: ""
  cpufreq_userspace: ""
  crc-ccitt: ""
  crc-itu-t: ""
  cxgb: ""
  cxgb3: ""
  cxgb4: ""
  cxgb4vf: ""
  dm-log: ""
  dm-mirror: ""
  dm-mod: ""
  dm-multipath: ""
  dm-queue-length: ""
  dm-region-hash: ""
  dm-round-robin: ""
  dm-service-time: ""
  dm9601: ""
  dme1737: ""
  drm: ""
  DRM: ""
  drm_kms_helper: ""
  drm_panel_orientation_quirks: ""
  dvb-core: ""
  dvb-usb-dib0700: ""
  dvb-usb-dvbsky: ""
  dvb-usb: ""
  dvb_usb_v2: ""
  e1000: ""
  e1000e: ""
  eeprom_93cx6: ""
  efifb: ""
  efivarfs: ""
  ehci-hcd: ""
  ehci-pci: ""
  evbug: ""
  evdev: ""
  exfat: ""
  f71882fg: ""
  f75375s: ""
  fb: ""
  fb_defio: ""
  fb_sys_fops: ""
  fbcon: ""
  fbcon_ccw: ""
  fbcon_cw: ""
  fbcon_rotate: ""
  fbcon_ud: ""
  fbdev: ""
  font: ""
  ftdi_sio: ""
  gasket: ""
  hpilo: ""
  hpsa: ""
  hvc_console: ""
  hwmon-vid: ""
  i2c-algo-bit: ""
  i40e: ""
  i8042: ""
  i915: ""
  iavf: ""
  ib_addr: ""
  ib_cm: ""
  ib_core: ""
  ib_mad: ""
  ib_sa: ""
  ib_umad: ""
  ib_uverbs: ""
  ice: ""
  ide-core: ""
  ide-gd_mod: ""
  ide-pci-generic: ""
  igb: ""
  igbvf: ""
  igc: ""
  intel_auxiliary: ""
  iosf_mbi: ""
  ipv6: ""
  irqbypass: ""
  iscsi_tcp: ""
  it87: ""
  iw_cm: ""
  iwldvm: ""
  iwlmvm: ""
  iwlwifi: ""
  ixgb: ""
  ixgbe: ""
  ixgbevf: ""
  jc42: ""
  jme: ""
  k10temp: ""
  kvm-amd: ""
  kvm-intel: ""
  kvm: ""
  lib80211: ""
  libiscsi: ""
  libiscsi_tcp: ""
  libphy: ""
  libsas: ""
  llc: ""
  lm75: ""
  lm78: ""
  lm90: ""
  lm95245: ""
  m88ds3103: ""
  mac80211: ""
  mdio: ""
  megaraid: ""
  megaraid_mbox: ""
  megaraid_mm: ""
  megaraid_sas: ""
  mii: ""
  mlx4_core: ""
  mlx4_en: ""
  mlx4_ib: ""
  mlx5_core: ""
  mlx5_ib: ""
  mlxsw_core: ""
  mlxsw_pci: ""
  mmc_block: ""
  mmc_core: ""
  mpt3sas: ""
  mptbase: ""
  mptctl: ""
  mptsas: ""
  mptscsih: ""
  mptspi: ""
  mt7601u: ""
  mtk-sd: ""
  mvsas: ""
  mwifiex: ""
  mwifiex_sdio: ""
  mwifiex_usb: ""
  nct6683: ""
  nct6775: ""
  netxen_nic: ""
  ntfs: ""
  p8022: ""
  pata_acpi: ""
  pata_legacy: ""
  phy-mt65xx-usb3: ""
  phy: ""
  piix: ""
  pl2303: ""
  pmbus: ""
  pmbus_core: ""
  prism2_usb: ""
  processor: ""
  psnap: ""
  qcaspi: ""
  qed: ""
  qede: ""
  qla3xxx: ""
  qlcnic: ""
  qlge: ""
  r8125: ""
  r8125_v9: ""
  r8126: ""
  r8152: ""
  r8168: ""
  r8169: ""
  r8169_lk: ""
  raid_class: ""
  rc-core: ""
  realtek: ""
  regmap-i2c: ""
  rndis_host: ""
  rtc-cmos: ""
  rtl8187: ""
  rtl8188ee: ""
  rtl818x_pci: ""
  rtl8192c-common: ""
  rtl8192ce: ""
  rtl8192cu: ""
  rtl8192de: ""
  rtl8192ee: ""
  rtl8192se: ""
  rtl8723-common: ""
  rtl8723ae: ""
  rtl8723be: ""
  rtl8821ae: ""
  rtl8xxxu: ""
  rtl_pci: ""
  rtl_usb: ""
  rtlwifi: ""
  rtsx_pci: ""
  rtsx_pci_sdmmc: ""
  rtsx_usb: ""
  rtsx_usb_sdmmc: ""
  sch_htb: ""
  sch_netem: ""
  sch_sfq: ""
  scsi_transport_sas: ""
  scsi_transport_spi: ""
  sdhci-acpi: ""
  sdhci-pci-data: ""
  sdhci-pci: ""
  sdhci-pltfm: ""
  sdhci: ""
  sdio_uart: ""
  sfc: ""
  sg: ""
  si2157: ""
  si2168: ""
  skge: ""
  sky2: ""
  softcursor: ""
  sp2: ""
  sr_mod: ""
  ssb: ""
  staging: ""
  stp: ""
  synobios: ""
  syscopyarea: ""
  sysfillrect: ""
  sysimgblt: ""
  tehuti: ""
  tg3: ""
  thermal: ""
  tps65217: ""
  ts2020: ""
  uhci-hcd: ""
  usbnet: ""
  usbserial: ""
  ushc: ""
  vesafb: ""
  vfio: ""
  vga16fb: ""
  vgastate: ""
  via-rhine: ""
  via-sdmmc: ""
  via-velocity: ""
  video: ""
  virtio: ""
  virtio_blk: ""
  virtio_console: ""
  virtio_input: ""
  virtio_mmio: ""
  virtio_net: ""
  virtio_pci: ""
  virtio_ring: ""
  virtio_scsi: ""
  vmw_pvscsi: ""
  vmw_vmci: ""
  vmxnet3: ""
  vub300: ""
  w83781d: ""
  w83793: ""
  wch: ""
  wch_pre: ""
  wireguard: ""
 * lkms:  

DSM: 
 * model:  DS918+(apollolake)
 * version:  7.2(69057u4)

Issue:  
    I install RR into EMMC,but cannot active advanced media extensions(AME)，I can active AME using USB drive in the same machine. I also found RR manger can not read RR configuration. I guest it is because pid/vid of EMMC setting.

logs:  
    dbgutils did not save logs to /mnt/p1/logs :(

wjz304, 2024-08-16:
AME 的激活过程中会以特殊的方式挂载引导盘，emmc 和 nvme 都会导致ame程序出现挂在失败，

dbgutils 主要为抓取启动日志，非启动问题 不需要使用 dbgutils

wjz304, 2024-08-24:
该问题并不是高优先级问题，暂时不做处理

maitien2004, 2025-02-19:
@zengdabo ssh into your nas and do the command: ls /dev/synoboot* and give me the result. I think wrongly mounted synoboot,synoboot1, synoboot2