RROrg/rr #2384: BUG 24.8.2版本华擎 j3455 就无法加载到群辉

zcr268, 2024-08-17:
请填写以下信息.
Please fill in the following information.

Install ENV: (You can find it in the boot interface.)

DMI:
CPU: j3455
NIC: (pid & vid)
RR version: (You can find it in the update menu.)

RR: 24.7.2
addons:
modules:
lkms:
DSM:

model: DS918+
version: 7.2.1
Issue:
24.8.2版本华擎 j3455 就无法加载到群辉
logs: 日志以发送到wjz304@qq.com

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)
...

(请先看一下https://github.com/RROrg/rr/issues/173、https://github.com/RROrg/rr/issues/175、#226的内容)
(Plz review the content of https://github.com/RROrg/rr/issues/173, https://github.com/RROrg/rr/issues/175, https://github.com/RROrg/rr/issues/226 first)
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)

wjz304, 2024-08-17:
日志中 没有挂掉的日志，

zcr268, 2024-08-18:
就是启动不起来屏幕变黑后一直无法链接上局域网5000一直无法访问

wjz304, 2024-08-18:
我知道，，我的意思是 日志中没有这个状态的日志，也不用试了，抓不到

zcr268, 2024-08-18:
希望越来越好,已回退.不急.大佬有空弄.

PrestarLin, 2024-08-21:
华擎j3455  2.8.4还是不行

PrestarLin, 2024-08-21:
@wjz304 经过测试,最新版本rr(2.8.4)开启直接引导可以正常启动

Oman4, 2024-08-22:
同样经过测试,最新版本rr(2.8.4)开启直接引导可以正常启动

zcr268, 2024-08-25:
确实实测UEFI 硬盘直接启动可以

littleghost2016, 2024-08-29:
实测星际蜗牛华擎j3455+16G内存，rr24.8.6版本【高级设置】-【直接启动】改为true，可正常安装7.2.2

haiyangzhijia, 2024-08-30:
> 实测星际蜗牛牛华擎j3455+16G内存，rr24.8.6版本【高级设置】-【直接启动】改为true，可正常安装7.2.2

j3455最大不就支持8G吗？你咋可以加到16G,我也是j3455

littleghost2016, 2024-08-31:
> > 实测星际蜗牛牛华擎j3455+16G内存，rr24.8.6版本【高级设置】-【直接启动】改为true，可正常安装7.2.2
> 
> j3455最大不就支持8G吗？你咋可以加到16G,我也是j3455

华擎官网写的最大支持16G，我自己也确实是用的两条8G。[华擎科技> J3455-ITX](https://www.asrock.com/mb/Intel/J3455-ITX/index.cn.asp)
[image]

haiyangzhijia, 2024-08-31:
> > > 实测星际蜗牛牛牛华擎j3455+16G内存，rr24.8.6版本【高级设置】-【直接启动】改为true，可正常安装7.2.2
> > 
> > 
> > j3455最大不支持8G吗？你咋可以加到16G，我也是j3455
> 
> 华擎官网写的最大支持16G，我自己也确实是用的愚蠢8G。[华擎科技> J3455-ITX](https://www.asrock.com/mb/Intel/J3455-ITX/index.cn.asp) ![图片](https://private-user-images.githubusercontent.com/22980536/363401534-f21db9f1-f78b-4985-b404-ca406525175c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUxMjQ1NzEsIm5iZiI6MTcyNTEyNDI3MSwicGF0aCI6Ii8yMjk4MDUzNi8zNjM0MDE1MzQtZjIxZGI5ZjEtZjc4Yi00OTg1LWI0MDQtY2E0MDY1MjUxNzVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODMxVDE3MTExMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNiNDYzMzFmMTJkOTM5NDc5MDI5YTc5NWRmN2U0Mjk0NWU2NGFmNjhlODYxODYzYTdiMTQyNTgzMGM3ZmE2NDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.kBxvaBj6cXyAunyztVBc_kIpgKL7C7azsAvzVSr2ItQ)

哈哈，非常感谢，我是看的英特尔的介绍，写的最大内存8G，刚又回看下，英特尔应该是按照单通道最大8G写的
[image]

haiyangzhijia, 2024-08-31:
> > > 实测星际蜗牛牛华擎j3455+16G内存，rr24.8.6版本【高级设置】-【直接启动】改为true，可正常安装7.2.2
> > 
> > 
> > j3455最大不就支持8G吗？你咋可以加到16G,我也是j3455
> 
> 华擎官网写的最大支持16G，我自己也确实是用的两条8G。[华擎科技> J3455-ITX](https://www.asrock.com/mb/Intel/J3455-ITX/index.cn.asp) ![图片](https://private-user-images.githubusercontent.com/22980536/363401534-f21db9f1-f78b-4985-b404-ca406525175c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUxMjQ5MjgsIm5iZiI6MTcyNTEyNDYyOCwicGF0aCI6Ii8yMjk4MDUzNi8zNjM0MDE1MzQtZjIxZGI5ZjEtZjc4Yi00OTg1LWI0MDQtY2E0MDY1MjUxNzVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODMxVDE3MTcwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU4ZGYyMTdmOWIzZDM4NGIwY2I0OGU5ZjAyZTA3ZjViZGJjNTYwYmNlYzk5NjE1ZDg3OWFiODk4NzVkZTQ3MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.se0Vf2ypkldPPflK4JPv2zi6bnShIKwgIqRn4VuZcTc)

哈哈，非常感谢，我是看的英特尔的介绍，写的最大内存8G，刚又回看下，英特尔应该是按照单通道最大8G写的
[image]

littleghost2016, 2024-09-01:
> > > > 实测星际蜗牛牛华擎j3455+16G内存，rr24.8.6版本【高级设置】-【直接启动】改为true，可正常安装7.2.2
> > > 
> > > 
> > > j3455最大不就支持8G吗？你咋可以加到16G,我也是j3455
> > 
> > 
> > 华擎官网写的最大支持16G，我自己也确实是用的两条8G。[华擎科技> J3455-ITX](https://www.asrock.com/mb/Intel/J3455-ITX/index.cn.asp) ![图片](https://private-user-images.githubusercontent.com/22980536/363401534-f21db9f1-f78b-4985-b404-ca406525175c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUxMjQ5MjgsIm5iZiI6MTcyNTEyNDYyOCwicGF0aCI6Ii8yMjk4MDUzNi8zNjM0MDE1MzQtZjIxZGI5ZjEtZjc4Yi00OTg1LWI0MDQtY2E0MDY1MjUxNzVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODMxVDE3MTcwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU4ZGYyMTdmOWIzZDM4NGIwY2I0OGU5ZjAyZTA3ZjViZGJjNTYwYmNlYzk5NjE1ZDg3OWFiODk4NzVkZTQ3MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.se0Vf2ypkldPPflK4JPv2zi6bnShIKwgIqRn4VuZcTc)
> 
> 哈哈，非常感谢，我是看的英特尔的介绍，写的最大内存8G，刚又回看下，英特尔应该是按照单通道最大8G写的 <img alt="截屏2024-09-01 01 13 07" width="1018" src="https://private-user-images.githubusercontent.com/23657541/363428880-33ebcad6-09f8-4b46-a649-adc8662bc76f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUxNjI0MjksIm5iZiI6MTcyNTE2MjEyOSwicGF0aCI6Ii8yMzY1NzU0MS8zNjM0Mjg4ODAtMzNlYmNhZDYtMDlmOC00YjQ2LWE2NDktYWRjODY2MmJjNzZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTAxVDAzNDIwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRmYWQ0MzM4MWE2MGI5NGNkYzFiNDNlNzYzYTNmNDdkNWNhOGVmNmNlZWM4NWI5NzM1NGUxNTBhYWNiNDhmZjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.MYPV2pZXJelhWpfbTGpMSqhHzP5HulwcPgcU5bHkxl0">

我刚开始查资料也是被英特尔的说法干扰了，不用客气哈哈

wjz304, 2024-10-24:
https://github.com/RROrg/rr/issues/1893 跟踪

closed 2024-10-24 by wjz304