RROrg/rr #2387: BUG 从24.8.1版本--24.8.3版本传统的和非转统主板就无法加载到群辉也就是失联（编译正常）。

Yanajnp, 2024-08-17:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    

...

wjz304, 2024-08-17:
那怎么办？

curtain12138, 2024-08-18:
同样遇到该问题，回退旧版本正常

qiangxinglin, 2024-08-20:
同样问题, 24.8.0升24.8.3失联  
@curtain12138 顺带问一下, 怎么回退版本, 重新刷一遍U盘?

curtain12138, 2024-08-20:
> 相同问题, 24.8.0升24.8.3失联 @curtain12138顺带问一下，怎么回退版本，重新刷一遍U盘？

是的，重写u盘，重新编译就行

PrestarLin, 2024-08-20:
什么叫做传统的和非传统的主板

wjz304, 2024-10-18:
test 24.10.1

wjz304, 2024-10-24:
https://github.com/RROrg/rr/issues/1893 跟踪

closed 2024-10-24 by wjz304