RROrg/rr #2388: Show RR Loader release version directly after the large "RR" displaying on systems that have booted the DSM kernel.

Rhubarb1, 2024-08-17:
On systems operational after several days using the RR Loader, you can't tell from the IPMI screen showing on systems that have booted into the loader, what version of the loader they are operating from.
The display of the version number: (eg: v:24.8.3.....booted (date booted) immediately after the large 'RR" symbol in centre of the screen of a running system, allows the operator the immediately determine what version of the loader is operational and when it was last booted.
Just a suggestion. Many thanks for your excellent and valued contribution!

wjz304, 2024-08-18:
https://github.com/RROrg/rr/issues/1791
[image]
You mean the display effect of "Switch direct boot: true"?

wjz304, 2024-08-18:
[image]
Meet the needs?

wjz304, 2024-08-24:
v24.8.4

closed 2024-08-24 by wjz304