RROrg/rr #2472: 功能请求：添加可触摸控制副屏显示功能[海康威视R1]

Kaouning, 2024-08-20:
设备：海康威视R1
cpu：n100
内存：16g
双2.5g网口
2个m.2接口
4个sata硬盘位
。。。。。。
主要是这个是大厂生产的，做工很棒，还是大厂100系列成品nas中最便宜的
最最最关键是它有rgb灯效、玻璃侧盖、还有副屏，外观简直了，唯一缺点原厂系统太垃圾了，希望大佬出个插件还是啥的让副屏能用起来

zlx84, 2024-08-21:
我也有这款nas，希望大佬考虑下这个网友建议。实在不行把之前那个启动完关闭屏幕的功能可以加回来。做成可选的也行，感谢

zhu000530, 2024-08-21:
我也是这款购买的R1 16G 就馋海康R1的身子了！

wjz304, 2024-08-24:
没环境，随缘吧

tycallen, 2024-09-03:
蹲一个

wjz304, 2024-11-07:
https://rrorg.cn/archives/wei-ming-ming-wen-zhang

naoki66, 2024-11-08:
小黄鱼上看到有人定制了，可以显示CPU\RAM\网速等等

naoki66, 2024-12-16:
> https://rrorg.cn/archives/wei-ming-ming-wen-zhang

反馈一个bug，用这个方法，root执行root@DSM:~#  modprobe -r i915 &&  modprobe i915
modprobe: ERROR: could not remove 'drm_mipi_dsi': Device or resource busy
可以关闭lcd。但是会导致硬解失效，emby找不到硬件
[image]