RROrg/rr #2607: Updating to 7.2.2 and getting "unable to perform this operation (21)"

eragon4k, 2024-08-26:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.8.4
 * addons:  n/a
 * modules:  n/a
 * lkms:  n/a

DSM: 
 * model:  3622
 * version:  DSM 7.2.1-69057 Update 5

Issue:  Updating to 7.2.2 and getting "unable to perform this operation (21)" I'm not sure if this has to do with the loader, but  just reporting it.

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
... 
[image]

eragon4k, 2024-08-26:
Little update,
RR: 24.8.2 and RR: 24.8.1 were able to update DSM 7.2.2. I didn't try with RR: 24.8.3.

eragon4k, 2024-08-27:
I did a little more test with friend's Dell desktops, and regardless of RR version, i'm still getting "unable to perform this operation (21).

lxh0508, 2024-08-27:
Install ENV: (You can find it in the boot interface.)

DMI: ASRock J3710
CPU: J3710
NIC: (pid & vid)
RR version: (You can find it in the update menu.)

RR: 24.8.4
addons: n/a
modules: n/a
lkms: n/a
DSM:

model: 918+
version: DSM 7.2.1-69057 Update 5
Issue: Updating to 7.2.2 and getting "unable to perform this operation (21)" I'm not sure if this has to do with the loader, but just reporting it.

Install ENV: (You can find it in the boot interface.)

DMI: ASRock B460M
CPU: I5-10500
NIC: (pid & vid)
RR version: (You can find it in the update menu.)

RR: 24.8.4
addons: n/a
modules: n/a
lkms: n/a
DSM:

model: 3622
version: DSM 7.2.1-69057 Update 5
Update to DSM7.2.2 OK

wjz304, 2024-08-27:
[image]

joewen85, 2024-08-27:
me too
model: dva3221
version: DSM 7.2.1-69057 Update 5

upgrade 7.2.2  "unable to perform this operation (21)"

maybe apollolake framework

eragon4k, 2024-08-27:
> ![image](https://private-user-images.githubusercontent.com/5615843/361753926-7c4312db-f385-4950-a1e8-8fd9f433dd52.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NzkwNjYsIm5iZiI6MTcyNDc3ODc2NiwicGF0aCI6Ii81NjE1ODQzLzM2MTc1MzkyNi03YzQzMTJkYi1mMzg1LTQ5NTAtYTFlOC04ZmQ5ZjQzM2RkNTIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MjdUMTcxMjQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGFmYTcwYTVlZmQxNWVhMTU0NTA0NWI3ZTZiY2M3MTM1ODE0YTdjOGJlNDNhMzVjYmIwNWY3MDNhODAwMTRkNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.cmykY7mdD7CvXhjbFDO-8gTqFTyeootSWqwi7s7egMc)

Though ds3622 is not apollolake and still had issue.

PeterSuh-Q3, 2024-08-28:
Have you tried using SATA bootloader instead of USB on DS3622xs+?
In my case, when I installed with TCRP, the same problem occurred with USB,
but it proceeded normally with SATA.

eragon4k, 2024-08-28:
> Have you tried using SATA bootloader instead of USB on DS3622xs+? In my case, when I installed with TCRP, the same problem occurred with USB, but it proceeded normally with SATA.

안년하세요.
그렇게 해보지는 안아습니다.
포럼에서 만이 도움 밥고있습니다. 
한국말 잘 못적어서 글은 안저고 보기만 합니다.

wjz304, 2024-08-28:
> Have you tried using SATA bootloader instead of USB on DS3622xs+? In my case, when I installed with TCRP, the same problem occurred with USB, but it proceeded normally with SATA.

[image]
[image]
i test, it's ok

eragon4k, 2024-08-28:
> ![image](https://private-user-images.githubusercontent.com/5615843/362171126-02f2fa05-76d3-4dde-bfa4-6bb9317ffa26.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

I tested two more devices, the DS3622, and all with  same results "unable to perform this operation (21)"

This time I tried with rr24.51, 24.6.7, 24.6.8 and 24.7.4.

wjz304, 2024-08-28:
Let's study it. It seems that there is some judgment.

PeterSuh-Q3, 2024-08-28:
@wjz304 

It's still just a hypothesis, but
in USB-type bare metals such as Apollo Lake, Broadwell, and Broadwell NK,
if the synoboot bootloader exists as a normal node, it becomes a problem,
and on the contrary, if there is no normal node, the Sata bootloader that forcibly creates a virtual node by the Addon seems to work normally.

In the case of Apollo Lake, I guess this exceptional situation occurs because the Sata bootloader exists as a normal node.

I will create a synoboot virtual node created by the Sata-type bootloader and conduct a test.

PeterSuh-Q3, 2024-08-28:
NonDT, ApolloLake, Broadwell, BroadwellNK, Denverton (under verification) etc. seem to have the same problem with USB bootloader.

PeterSuh-Q3, 2024-08-28:
> > Have you tried using SATA bootloader instead of USB on DS3622xs+? In my case, when I installed with TCRP, the same problem occurred with USB, but it proceeded normally with SATA.
> 
> 안년하세요. 그렇게 해보지는 안아습니다. 포럼에서 만이 도움 받고있습니다. 한국말 잘 못적어서 글은 안적고 보기만 합니다.

안녕하세요.
한국의 서버포럼을 의미하는것 같네요.
도움이 되셨다니 저도 기쁘네요.

이번 현상은 rr(arpl) 과 저의 mshell(tcrp) 이 동일하게 나타나는 현상이라서 7.2.2 버전의 DSM 에 어떤 매커니즘의 변화가 있는지 심도 있게 살펴봐야 할것 같습니다.
아직은 Update 0 버전에 해당하기 때문에 ramdisk 패치가 발동해야 하는 케이스는 아닙니다.
Update 1 에서 최초 ramdisk 패치가 필요한 시점이 도래할 때 이슈가 없을지 모르겠습니다.

wjz304님도 원인을 열심히 찾고 계시고 저도 계속 찾는중입니다.

wjz304, 2024-08-28:
```
Jan  1 00:07:11  updater: updater.c:5991 Start of the updater...
Jan  1 00:07:11  updater: updater.c:3287 orgBuildNumber = 72803.0, newBuildNumber = 72803.0
Jan  1 00:07:11  updater: util/updater_util.cpp:90 fail to read company in /tmpRoot//etc.defaults/synoinfo.conf
Jan  1 00:07:11  updater: util/check_patch_support_hw_revision.cpp:63 This patch supports HW revision of this machine
Jan  1 00:07:12  updater: child exit normally with [0]
Jan  1 00:07:12  updater: updater.c:6297 ==== Start flash update ====
Jan  1 00:07:12  updater: updater.c:6301 This is X86 platform
Jan  1 00:07:12  kernel: [ 3250.037863] ext2: synoboot2 mounted, process=updater
Jan  1 00:07:12  kernel: [ 3250.049839] synoboot2 unmounted, process=updater
Jan  1 00:07:12  kernel: [ 3250.061967] ext2: synoboot2 mounted, process=updater
Jan  1 00:07:12  kernel: [ 3250.065824] synoboot2 unmounted, process=updater
Jan  1 00:07:12  kernel: [ 3250.079529] ext2: synoboot2 mounted, process=updater
Jan  1 00:07:12  kernel: [ 3250.083163] synoboot2 unmounted, process=updater
Jan  1 00:07:12  kernel: [ 3250.097185] ext2: synoboot2 mounted, process=updater
Jan  1 00:07:12  kernel: [ 3250.101625] synoboot2 unmounted, process=updater
Jan  1 00:07:12  updater: updater.c:6324 The SynoBoot partitions exist.
Jan  1 00:07:12  updater: updater.c:4149 SYNORedBootUpdCheckAndApply(4149): Skip bootloader update, no uboot_do_upd.sh exists
Jan  1 00:07:12  kernel: [ 3250.111408] ext2: synoboot2 mounted, process=updater
Jan  1 00:07:12  updater: updater.c:613 checksum generated by [/bin/grep -i zImage checksum.syno >> /tmp/checksum.syno.tmp]
Jan  1 00:07:12  updater: updater.c:642 file [zImage] is being copied to [/tmp/bootmnt]
Jan  1 00:07:12  updater: updater.c:650 file [/tmp/bootmnt/zImage] process done.
Jan  1 00:07:12  updater: updater.c:613 checksum generated by [/bin/grep -i rd.gz checksum.syno >> /tmp/checksum.syno.tmp]
Jan  1 00:07:12  updater: updater.c:642 file [rd.gz] is being copied to [/tmp/bootmnt]
Jan  1 00:07:12  updater: updater.c:650 file [/tmp/bootmnt/rd.gz] process done.
Jan  1 00:07:12  updater: updater.c:613 checksum generated by [/bin/grep -i grub_cksum.syno checksum.syno >> /tmp/checksum.syno.tmp]
Jan  1 00:07:12  updater: updater.c:642 file [grub_cksum.syno] is being copied to [/tmp/bootmnt]
Jan  1 00:07:12  updater: updater.c:650 file [/tmp/bootmnt/grub_cksum.syno] process done.
Jan  1 00:07:12  updater: updater.c:633 file [/tmp/bootmnt/hda1.tgz] is being removed
Jan  1 00:07:12  updater: updater.c:650 file [/tmp/bootmnt/hda1.tgz] process done.
Jan  1 00:07:12  updater: updater.c:633 file [/tmp/bootmnt/updater] is being removed
Jan  1 00:07:12  updater: updater.c:650 file [/tmp/bootmnt/updater] process done.
Jan  1 00:07:12  updater: updater.c:633 file [/tmp/bootmnt/VERSION] is being removed
Jan  1 00:07:12  updater: updater.c:650 file [/tmp/bootmnt/VERSION] process done.
Jan  1 00:07:12  updater: updater.c:660 checksum file updated by [/bin/cp -f /tmp/checksum.syno.tmp /tmp/bootmnt/checksum.syno]
Jan  1 00:07:12  kernel: [ 3250.553433] synoboot2 unmounted, process=updater
Jan  1 00:07:17  kernel: [ 3255.566702] ext2: synoboot2 mounted, process=updater
Jan  1 00:07:17  kernel: [ 3255.631685] synoboot2 unmounted, process=updater
Jan  1 00:07:17  updater: updater.c:1187 Failed to mount factory partition
Jan  1 00:07:17  updater: updater.c:5665 failed to update factory partition, retry 0
Jan  1 00:07:17  updater: updater.c:1187 Failed to mount factory partition
Jan  1 00:07:17  updater: updater.c:5665 failed to update factory partition, retry 1
Jan  1 00:07:17  updater: updater.c:1187 Failed to mount factory partition
Jan  1 00:07:17  updater: updater.c:5665 failed to update factory partition, retry 2
Jan  1 00:07:17  updater: updater.c:1187 Failed to mount factory partition
Jan  1 00:07:17  updater: updater.c:5665 failed to update factory partition, retry 3
Jan  1 00:07:17  updater: updater.c:1187 Failed to mount factory partition
Jan  1 00:07:17  updater: updater.c:5665 failed to update factory partition, retry 4
Jan  1 00:07:17  updater: updater.c:6511 fail to upgrade factory partition
Jan  1 00:07:17  kernel: [ 3255.649573] ext2: synoboot2 mounted, process=updater
Jan  1 00:07:17  updater: updater.c:746 Restoring [zImage]
Jan  1 00:07:17  updater: updater.c:757 Copying file [/tmpData/synob@ckup/zImage] as [/tmp/bootmnt/zImage], result[0]
Jan  1 00:07:17  updater: updater.c:746 Restoring [rd.gz]
Jan  1 00:07:17  updater: updater.c:757 Copying file [/tmpData/synob@ckup/rd.gz] as [/tmp/bootmnt/rd.gz], result[0]
Jan  1 00:07:17  updater: updater.c:743 Skip restoring [vendor]
Jan  1 00:07:17  updater: updater.c:746 Restoring [grub_cksum.syno]
Jan  1 00:07:17  updater: updater.c:757 Copying file [/tmpData/synob@ckup/grub_cksum.syno] as [/tmp/bootmnt/grub_cksum.syno], result[0]
Jan  1 00:07:17  updater: updater.c:746 Restoring [hda1.tgz]
Jan  1 00:07:17  updater: updater.c:760 Removing file [/tmp/bootmnt/hda1.tgz], result[-1]
Jan  1 00:07:17  updater: updater.c:763 Restore [hda1.tgz] fail, remove it from synoboot
Jan  1 00:07:17  updater: updater.c:746 Restoring [updater]
Jan  1 00:07:17  updater: updater.c:760 Removing file [/tmp/bootmnt/updater], result[-1]
Jan  1 00:07:17  updater: updater.c:763 Restore [updater] fail, remove it from synoboot
Jan  1 00:07:17  updater: updater.c:746 Restoring [VERSION]
Jan  1 00:07:17  updater: updater.c:760 Removing file [/tmp/bootmnt/VERSION], result[-1]
[... cut ...]
Jan  1 00:07:17  install.cgi: ErrFHOSTCleanPatchDirFile: Remove /tmpData/upd@te.pat...
Jan  1 00:07:17  install.cgi: ErrFHOSTDoUpgrade(2037): child process failed, retv=-21
Jan  1 00:07:17  install.cgi: ninstaller.c:2059(ErrFHOSTDoUpgrade) err=[-1]
Jan  1 00:07:17  install.cgi: ninstaller.c:2063(ErrFHOSTDoUpgrade) retv=[-21]
Jan  1 00:07:17  install.cgi: install.c:502 Upgrade for reinstll by the download patch fail.
Jan  1 00:07:17  install.cgi: install.c:785 Upgrade by download patch fail.
Jan  1 00:00:00  install.cgi: ninstaller.c:169 umount partition /tmpData
```

wjz304, 2024-08-28:
> updater.c:6911 Failed to accomplish the update! (errno = 21)

Need to analyze what the "updater" is doing,

nillebor, 2024-08-28:
Update on different devices works without problems. Tested on Gigabyte z97 with Intel i5 and HP Elitedesk Mini with I5 with 24.8.4 loader.

I test this again in a VM with a DS3622 and report.

PeterSuh-Q3, 2024-08-28:
`Aug 28 12:56:30  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on Virtual SATA Hard Drive 00000001 72803.0
Aug 28 12:56:30  install.cgi: install.c:571 Unknown param localinstallreq=false
Aug 28 12:56:30  install.cgi: install.c:571 Unknown param _dc=1724849791633
Aug 28 12:56:30  install.cgi: ninstaller.c:699 Dev: sda, DiskPath: /dev/sda, Partition version: 9
Aug 28 12:56:30  install.cgi: ninstaller.c:140 Mount partion /dev/md0 /tmpRoot
Aug 28 12:56:30  kernel: [ 6385.492770] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Aug 28 12:56:30  kernel: [ 6385.512328] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Aug 28 12:56:30  install.cgi: ninstaller.c:169 umount partition /tmpRoot
Aug 28 12:56:30  install.cgi: ninstaller.c:140 Mount partion /dev/md0 /tmpRoot
Aug 28 12:56:30  kernel: [ 6385.651350] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Aug 28 12:56:30  kernel: [ 6385.662537] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Aug 28 12:56:30  install.cgi: ninstaller.c:169 umount partition /tmpRoot
Aug 28 12:56:30  install.cgi: ninstaller.c:140 Mount partion /dev/md0 /tmpRoot
Aug 28 12:56:30  kernel: [ 6385.683533] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Aug 28 12:56:30  kernel: [ 6385.694486] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Aug 28 12:56:30  install.cgi: ninstaller.c:511 fail to get admin port from /tmpRoot/etc/synoinfo.conf... try another way
Aug 28 12:56:30  install.cgi: util_junior.c:359 Models are not the same. [synology_apollolake_918+, synology_epyc7002_sa6400]
Aug 28 12:56:30  install.cgi: ninstaller.c:881 Model not match [/etc.defaults/synoinfo.conf, /tmpRoot//etc.defaults/synoinfo.conf]
Aug 28 12:56:30  install.cgi: ninstaller.c:1307 SYSTEM_NORMAL, [Recoverable=0]
Aug 28 12:56:30  install.cgi: ninstaller.c:169 umount partition /tmpRoot
Aug 28 12:56:30  install.cgi: ninstaller.c:1222(FillUpgradeVolumeInfo): gszUpgradeVolDev = /dev/md0 
Aug 28 12:56:30  install.cgi: ninstaller.c:1223(FillUpgradeVolumeInfo): gszUpgradeVolMnt = /tmpData 
Aug 28 12:56:30  install.cgi: ninstaller.c:1331 gblSupportRaid: 1, gSysStatus: 1, gblCreateDataVol: 0, gblSystemRecoverable: 0
Aug 28 12:56:30  install.cgi: synoupgrade.c:785 system status = 1
Aug 28 12:56:30  install.cgi: ninstaller.c:140 Mount partion /dev/md0 /tmpData
Aug 28 12:56:30  kernel: [ 6385.757831] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Aug 28 12:56:30  kernel: [ 6385.776887] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Aug 28 12:56:30  install.cgi: ninstaller.c:1390 FreeSize = 6773374976 bytes
Aug 28 12:56:47  install.cgi: generic_hash.c:80 Generic hash: /tmpData/@autoupdate/upload.pat: 0f2bc409380f900d02e9b0f4f6db0a83116b2fae81f6a40db0be477383031f1d
Aug 28 12:56:47  install.cgi: synoupgrade.c:1366 Extracting VERSION from Old Patch Format [/tmpData/@autoupdate/upload.pat][/tmpData/@autoupdate]
Aug 28 12:56:47  install.cgi: synoupgrade.c:1372 Extracting VERSION from SynoArchive
Aug 28 12:56:47  install.cgi: lib/synoarchive.c:440 synoinstall: synoarchive: [/tmpData/@autoupdate/upload.pat][OPEN] Type: System
Aug 28 12:56:47  install.cgi: lib/synoarchive.c:84 synoinstall: synoarchive: [/tmpData/@autoupdate/upload.pat][INIT] [/tmpData/@autoupdate]
Aug 28 12:56:47  install.cgi: lib/synoarchive.c:92 synoinstall: synoarchive: [/tmpData/@autoupdate/upload.pat][OPEN] Start
Aug 28 12:56:48  install.cgi: lib/synoarchive.c:102 synoinstall: synoarchive: [/tmpData/@autoupdate/upload.pat][OPEN] End: [Success][3688618971][0.324s]
Aug 28 12:56:48  install.cgi: lib/synoarchive.c:288 synoinstall: synoarchive: [/tmpData/@autoupdate/upload.pat][EXTRACT_ONE][VERSION] Start
Aug 28 12:56:48  install.cgi: lib/synoarchive.c:305 synoinstall: synoarchive: [/tmpData/@autoupdate/upload.pat][EXTRACT_ONE][VERSION] End: [Success][3688618971][0.000s]
Aug 28 12:56:48  install.cgi: lib/synoarchive.c:229 synoinstall: synoarchive: [/tmpData/@autoupdate/upload.pat][FREE]
Aug 28 12:56:48  install.cgi: Remove /tmpData/upd@te...cmd=[/bin/rm -rf /tmpData/upd@te > /dev/null 2>&1]
Aug 28 12:56:48  install.cgi: Create /tmpData/upd@te...cmd=[/bin/mkdir -p /tmpData/upd@te > /dev/null 2>&1]
Aug 28 12:56:48  install.cgi: ninstaller.c:1443 Extracting SynoArchive [/tmpData/upd@te.pat][/tmpData/upd@te]
Aug 28 12:56:48  install.cgi: lib/synoarchive.c:440 synoinstall: synoarchive: [/tmpData/upd@te.pat][OPEN] Type: System
Aug 28 12:56:48  install.cgi: lib/synoarchive.c:84 synoinstall: synoarchive: [/tmpData/upd@te.pat][INIT] [/tmpData/upd@te]
Aug 28 12:56:48  install.cgi: lib/synoarchive.c:92 synoinstall: synoarchive: [/tmpData/upd@te.pat][OPEN] Start
Aug 28 12:56:48  install.cgi: lib/synoarchive.c:102 synoinstall: synoarchive: [/tmpData/upd@te.pat][OPEN] End: [Success][3688618971][0.317s]
Aug 28 12:56:48  install.cgi: lib/synoarchive.c:286 synoinstall: synoarchive: [/tmpData/upd@te.pat][EXTRACT_ALL] Start
Aug 28 12:56:49  install.cgi: lib/synoarchive.c:299 synoinstall: synoarchive: [/tmpData/upd@te.pat][EXTRACT_ALL] End: [Success][3688618971][0.689s]
Aug 28 12:56:49  install.cgi: lib/synoarchive.c:229 synoinstall: synoarchive: [/tmpData/upd@te.pat][FREE]
Aug 28 12:56:49  install.cgi: RemoveUpgradeFile: Remove /tmpData/upd@te.pat...
Aug 28 12:56:49  install.cgi: Verify checksum of [/tmpData/upd@te]...
[... cut ...]
Aug 28 12:56:56  install.cgi: ErrFHOSTDoUpgrade(2037): child process failed, retv=-21
Aug 28 12:56:56  install.cgi: ninstaller.c:2059(ErrFHOSTDoUpgrade) err=[-1]
Aug 28 12:56:56  install.cgi: ninstaller.c:2063(ErrFHOSTDoUpgrade) retv=[-21]
Aug 28 12:56:56  install.cgi: install.c:458 Upgrade by the manual patch fail.
Aug 28 12:56:56  install.cgi: install.c:780 Upgrade by the uploaded patch /tmpData/@autoupdate/upload.pat fail.
Jan  1 00:00:00  install.cgi: ninstaller.c:169 umount partition /tmpData`

PeterSuh-Q3, 2024-08-28:
This log is the contents of the junior section when installing DSM on DS918+. The same 21 error code is confirmed in the middle.

PeterSuh-Q3, 2024-08-28:
```
Aug 28 13:24:15  updater: updater.c:5991 Start of the updater...
Aug 28 13:24:15  updater: updater.c:3287 orgBuildNumber = 72803.0, newBuildNumber = 72803.0
Aug 28 13:24:15  updater: updater.c:3287 orgBuildNumber = 72803.0, newBuildNumber = 72803.0
Aug 28 13:24:15  updater: util/check_patch_support_hw_revision.cpp:63 This patch supports HW revision of this machine
Aug 28 13:24:16  updater: child exit normally with [0]
Aug 28 13:24:16  updater: updater.c:6297 ==== Start flash update ====
Aug 28 13:24:16  updater: updater.c:6301 This is X86 platform
Aug 28 13:24:16  kernel: [   95.113611] ext2: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.115195] ext4: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.116457] vfat: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.122995] FAT-fs (synoboot2): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Aug 28 13:24:16  kernel: [   95.124675] synoboot2 unmounted, process=updater
Aug 28 13:24:16  kernel: [   95.131125] ext2: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.132072] ext4: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.133019] vfat: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.139110] FAT-fs (synoboot2): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Aug 28 13:24:16  kernel: [   95.140799] synoboot2 unmounted, process=updater
Aug 28 13:24:16  kernel: [   95.146168] ext2: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.147113] ext4: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.148061] vfat: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.154293] FAT-fs (synoboot2): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Aug 28 13:24:16  kernel: [   95.155969] synoboot2 unmounted, process=updater
Aug 28 13:24:16  kernel: [   95.162273] ext2: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.163336] ext4: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.164262] vfat: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.170239] FAT-fs (synoboot2): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Aug 28 13:24:16  kernel: [   95.171888] synoboot2 unmounted, process=updater
Aug 28 13:24:16  updater: updater.c:6324 The SynoBoot partitions exist.
Aug 28 13:24:16  updater: updater.c:4149 SYNORedBootUpdCheckAndApply(4149): Skip bootloader update, no uboot_do_upd.sh exists
Aug 28 13:24:16  kernel: [   95.181155] ext2: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.182103] ext4: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.183055] vfat: synoboot2 mounted, process=updater
Aug 28 13:24:16  kernel: [   95.189331] FAT-fs (synoboot2): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Aug 28 13:24:16  updater: updater.c:613 checksum generated by [/bin/grep -i zImage checksum.syno >> /tmp/checksum.syno.tmp]
Aug 28 13:24:16  updater: updater.c:642 file [zImage] is being copied to [/tmp/bootmnt]
Aug 28 13:24:16  updater: updater.c:650 file [/tmp/bootmnt/zImage] process done.
Aug 28 13:24:16  updater: updater.c:613 checksum generated by [/bin/grep -i rd.gz checksum.syno >> /tmp/checksum.syno.tmp]
Aug 28 13:24:16  updater: updater.c:642 file [rd.gz] is being copied to [/tmp/bootmnt]
Aug 28 13:24:16  updater: updater.c:650 file [/tmp/bootmnt/rd.gz] process done.
Aug 28 13:24:16  updater: updater.c:613 checksum generated by [/bin/grep -i grub_cksum.syno checksum.syno >> /tmp/checksum.syno.tmp]
Aug 28 13:24:16  updater: updater.c:642 file [grub_cksum.syno] is being copied to [/tmp/bootmnt]
Aug 28 13:24:16  updater: updater.c:650 file [/tmp/bootmnt/grub_cksum.syno] process done.
Aug 28 13:24:16  updater: updater.c:633 file [/tmp/bootmnt/hda1.tgz] is being removed
Aug 28 13:24:16  updater: updater.c:650 file [/tmp/bootmnt/hda1.tgz] process done.
Aug 28 13:24:16  updater: updater.c:633 file [/tmp/bootmnt/updater] is being removed
Aug 28 13:24:16  updater: updater.c:650 file [/tmp/bootmnt/updater] process done.
Aug 28 13:24:16  updater: updater.c:633 file [/tmp/bootmnt/VERSION] is being removed
Aug 28 13:24:16  updater: updater.c:650 file [/tmp/bootmnt/VERSION] process done.
Aug 28 13:24:16  updater: updater.c:660 checksum file updated by [/bin/cp -f /tmp/checksum.syno.tmp /tmp/bootmnt/checksum.syno]
Aug 28 13:24:16  kernel: [   95.380034] synoboot2 unmounted, process=updater
Aug 28 13:24:16  scemd: oob/oob_reg_read.c:46 Read OOB reg 316 failed, ret=-1
Aug 28 13:24:19  scemd: oob/oob_reg_read.c:46 Read OOB reg 316 failed, ret=-1
Aug 28 13:24:21  kernel: [  100.392387] ext2: synoboot2 mounted, process=updater
Aug 28 13:24:21  kernel: [  100.393870] ext4: synoboot2 mounted, process=updater
Aug 28 13:24:21  kernel: [  100.395185] vfat: synoboot2 mounted, process=updater
Aug 28 13:24:21  kernel: [  100.402428] FAT-fs (synoboot2): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Aug 28 13:24:21  kernel: [  100.440869] synoboot2 unmounted, process=updater
Aug 28 13:24:21  kernel: [  100.448956] ext2: synoboot1 mounted, process=updater
Aug 28 13:24:21  kernel: [  100.449896] ext4: synoboot1 mounted, process=updater
Aug 28 13:24:21  kernel: [  100.450871] vfat: synoboot1 mounted, process=updater
[... cut ...]
Aug 28 13:24:27  scemd: oob/oob_reg_read.c:46 Read OOB reg 316 failed, ret=-1
Aug 28 13:24:27  root: Failed to unregister quickconnect [1][]
Aug 28 13:24:27  root: Failed to unregister quickconnect
Aug 28 13:24:29  scemd: oob/oob_reg_read.c:46 Read OOB reg 316 failed, ret=-1
Aug 28 13:24:30  scemd: oob/oob_reg_read.c:46 Read OOB reg 316 failed, ret=-1
Aug 28 13:24:31  scemd: oob/oob_reg_read.c:46 Read OOB reg 316 failed, ret=-1
```

This log is the contents of the junior level when installing DSM on VM DS3622xs+. It is a successful result.

PeterSuh-Q3, 2024-08-28:
₩₩₩
Aug 28 09:38:41  updater: updater.c:1904 ./H2OFFT-Lx64-0815 bios.ROM -BIOS -ALL -N failed! 166
Aug 28 09:38:41  updater: updater.c:6536 fail to upgrade BIOS
₩₩₩

Could this part be the key?

wjz304, 2024-08-28:
> ```
> Aug 28 09:38:41  updater: updater.c:1904 ./H2OFFT-Lx64-0815 bios.ROM -BIOS -ALL -N failed! 166
> Aug 28 09:38:41  updater: updater.c:6536 fail to upgrade BIOS
> ```
> 
> Could this part be the key?

only 918 (apollolake) has this

wjz304, 2024-08-28:
> ```
> Aug 28 09:38:41  updater: updater.c:1904 ./H2OFFT-Lx64-0815 bios.ROM -BIOS -ALL -N failed! 166
> Aug 28 09:38:41  updater: updater.c:6536 fail to upgrade BIOS
> ```
>
> Could this part be the key? [...]

However, it is possible that his failure will lead to a chain reaction.

wjz304, 2024-08-28:
> ```
> Aug 28 09:38:41  updater: updater.c:1904 ./H2OFFT-Lx64-0815 bios.ROM -BIOS -ALL -N failed! 166
> Aug 28 09:38:41  updater: updater.c:6536 fail to upgrade BIOS
> ```
>
> Could this part be the key? [...]

[image]
You are right, installed successfully after lkm  adding the H2OFFT-Lx64-0815 shield

nillebor, 2024-08-28:
> I test this again in a VM with a DS3622 and report.

i have set up a new DS3622 in the VM with DSM 7.1.1. The online update to DSM 7.2.1 worked without any problems. The manual update to DSM 7.2.2 also worked without the above error message. As a loader, I have tested an old and the current version. Both worked without errors.

Can or should I test something else?

wjz304, 2024-08-28:
try 24.8.6, 

if new installll ，plz get http:<ip>:500/webman/get_logs.cgi
if update in DSM, plz get /var/log/messages

eragon4k, 2024-08-28:
> try 24.8.6,
> 
> if new install, plz get `http://<ip>:5000/webman/get_logs.cgi` if update in DSM, plz get `/var/log/messages`

Two of the four devices (DS3622) that I tested passed, while the other two failed.

I'll try with 24.8.6 later today

lxh0508, 2024-08-29:
This issue has been resolved in version 24.8.6. Thank you very much! @wjz304 @PeterSuh-Q3

PeterSuh-Q3, 2024-08-29:
[image]

The issue of not being able to install natively on NON-DT models such as DS918+ / DS3622xs+ / DS3617xs is probably due to Synology's incompatible disk firmware database check failure.
The firmware of the virtual disk created by VMWARE FUSION seems to be skipped by the compatibility firmware check.

I have two ADATA SP600 32GB SSDs that I use for bare metal testing.
It seems that the firmware check of these SSDs failed.

Originally, @007revad's script was configured to run in the late event,
so it does not trigger during the DSM installation phase.

I think we need to check if there is a change related to this in this DSM 7.2.2.

```
DiskStation> tail -f /var/log/messages
Aug 29 02:22:06  synodiskport: external/external_disk_port_check.c:158 expected sdx or /dev/sdx, not bus/usb/, not match any port type
Aug 29 02:22:06  synodiskport: external/external_disk_port_check.c:158 expected sdx or /dev/sdx, not , not match any port type
Aug 29 02:22:11  root: Failed to register alias [1]
Aug 29 02:22:11  root: Failed to register quickconnect alias []
Aug 29 02:22:11  root: Failed to run https installation
Aug 29 02:22:11  root: Failed to run JuniorExpansionPack [1]
Aug 29 02:22:11  root: Failed to run JuniorExpansionPack
Aug 29 02:22:11  synoagentregisterd: synoagentregisterd.c:149 Generating register key...
Aug 29 02:22:11  synoagentregisterd: synoagentregisterd.c:163 Generated register key [010D325A.F710.4E01.86D9.75D842FA051D].
Aug 29 02:50:46  login[9253]: root login on 'pts/0'
Aug 29 02:51:50  synodiskupdatehclport: misc/misc_curl_to_buffer_or_file.c:115 Failed to curl url https://diskupdate.synology.com/getSSDsInfo/v7?model=ds918%2B&sn=1980PDN015562&ds_version=72803.0, ret=60
Aug 29 02:51:50  synodiskupdatehclport: synodiskupdatehclport.c:112 Failed to download file from server: https://diskupdate.synology.com/getSSDsInfo/v7?model=ds918%2B&sn=1980PDN015562&ds_version=72803.0
Aug 29 02:51:50  synodiskupdatehclport: synodiskupdatehclport.c:318 Fail to get hcl info from online server
Aug 29 02:52:13  synoupgrade: autocleanup/mount.c:19 run '/usr/bin/mount /dev/md0 /_disk'
Aug 29 02:52:13  kernel: [ 1862.368878] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Aug 29 02:52:13  synoupgrade: autocleanup/mount.c:27 mount's exit status is zero
Aug 29 02:52:13  kernel: [ 1862.383061] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Aug 29 02:52:13  synoupgrade: get_disk_dsm_version.c:59 try reading disk dsm's version at file '/_disk//etc.defaults/VERSION' of key 'buildnumber'
Aug 29 02:52:13  synoupgrade: autocleanup/mount.c:44 run '/usr/bin/umount -l /_disk'
Aug 29 02:52:14  synoupgrade: autocleanup/mount.c:54 umount's exit status is zero
Aug 29 02:52:14  synoupgrade: autocleanup/directory.c:38 remove directory /_disk, result is 0
Aug 29 02:52:14  raidtool: root_block_get.c:42 Failed to get op info for [/dev/sde][0x0900 root_block_op_info_get.c:85]
Aug 29 02:52:14  raidtool: root_block_get.c:127 Failed to read root block from [/dev/sde][0x0900 root_block_op_info_get.c:85]
Aug 29 02:52:14  raidtool: raidtool.c:234 Failed to get root block from /dev/sde
Aug 29 02:52:14  synossdcache: space_disk_block_get.c:48 Failed to open path: /dev/sde, errno=No such file or directory
Aug 29 02:52:14  synossdcache: synossdcache.c:58 Failed to get synoblock on [/dev/sde]
Aug 29 02:52:14  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on SP600 2.9 72803.0
Aug 29 02:52:14  synodiskport: disk/disk_rule_junior.c:245 Failed to get target on SP600 2.3 72803.0
Aug 29 02:52:14  synodiskport: DoSmartFirmAndSerialRead(218) open device /dev/sde fail
Aug 29 02:52:14  synodiskport: disk/disk_rule_junior.c:399 Fail to read firmware from device /dev/sde
Aug 29 02:52:14  synodiskport: disk/disk_rule_junior.c:445 Failed to get basic information for sde
Aug 29 03:15:00  login[9495]: root login on 'pts/1'
```

PeterSuh-Q3, 2024-08-29:
```
{
  "smart": {
    "ver": 166,
    "url": "https://diskdownload.synology.com/ssd.drivedb.db",
    "checksum": "aeacec8c371459a5e3c261d5d6adde60"
  },
  "temperature": {
    "ver": 196,
    "url": "https://diskdownload.synology.com/temperature.drivedb.db",
    "checksum": "215c9ade560313459cc8b5e65bf76b7e"
  },
  "opensource": {
    "ver": 115,
    "url": "https://diskdownload.synology.com/opensource.drivedb.db",
    "checksum": "be64023ee11e6564d8386f4960a74ece"
  },
  "drive_attribute": {
    "ver": 15,
    "url": "https://diskdownload.synology.com/drive_attribute/drive_attribute.db",
    "checksum": "04c3e8af07cafb5753890211950c5c7b"
  },
  "diskaction": {
    "ver": 276,
    "rel": 20240524,
    "url": "https://diskdownload.synology.com/diskaction/diskaction.xml",
    "checksum": "b39370d2af2b671becb490cd0b3772f4"
  },
  "HCL_v6_ds918+": {
    "ver": 4036,
    "rel": 20231128,
    "url": "https://diskdownload.synology.com/HCLv6/nas/ds918+/host",
    "checksum": "ddaf9ea4c1333e9c0275a589bfaf7077"
  },
  "MemHCL_v6_ds918+": {
    "ver": 4002,
    "rel": 20240430,
    "url": "https://diskdownload.synology.com/HCLv6/nas/ds918+/mem_host",
    "checksum": "c9cd00ea045d8192f275200b3b792e6a"
  },
  "general_settings_ds918+": {
    "ver": 8002,
    "url": "https://diskdownload.synology.com/products/ds918+/general_settings.db",
    "checksum": "3f7ff20271bd3735116c9b8c671d0691",
    "rel": 20230706
  },
  "HCL_v7_ds918+": {
    "ver": 8048,
    "url": "https://diskdownload.synology.com/HCLv7/ds918+/host",
    "checksum": "97514474fba5e1eb5538e1637b38fff4",
    "rel": 20240820
  },
  "HCL_v7_ds918+_m2d18": {
    "ver": 8002,
    "url": "https://diskdownload.synology.com/HCLv7/ds918+/m2d18",
    "checksum": "3d82413176355dd0de2546a48b64bd8c",
    "rel": 20210702
  },
  "HCL_v7_ds918+_m2d17": {
    "ver": 8002,
    "url": "https://diskdownload.synology.com/HCLv7/ds918+/m2d17",
    "checksum": "3d82413176355dd0de2546a48b64bd8c",
    "rel": 20210702
  },
  "HCL_v7_ds918+_e10m20-t1": {
    "ver": 8002,
    "url": "https://diskdownload.synology.com/HCLv7/ds918+/e10m20-t1",
    "checksum": "3d82413176355dd0de2546a48b64bd8c",
    "rel": 20210702
  },
  "HCL_v7_ds918+_m2d20": {
    "ver": 8002,
    "url": "https://diskdownload.synology.com/HCLv7/ds918+/m2d20",
    "checksum": "3d82413176355dd0de2546a48b64bd8c",
    "rel": 20210702
  }
}
```

PeterSuh-Q3, 2024-08-29:
```
Aug 29 03:30:43  synodiskupdatehclport: misc/misc_curl_to_buffer_or_file.c:115 Failed to curl url https://diskupdate.synology.com/getSSDsInfo/v7?model=ds918%2B&sn=1860PDN003097&ds_version=72803.0, ret=60
Aug 29 03:30:43  synodiskupdatehclport: synodiskupdatehclport.c:112 Failed to download file from server: https://diskupdate.synology.com/getSSDsInfo/v7?model=ds918%2B&sn=1860PDN003097&ds_version=72803.0
Aug 29 03:30:43  synodiskupdatehclport: synodiskupdatehclport.c:318 Fail to get hcl info from online server
```

According to this log, some kind of check is performed in real time with the input Synology serial number, and it seems to fail if it is not a genuine serial number. This leads to disk firmware issues, and if the disk is not officially supported by the Synology model, it seems to fail.

eragon4k, 2024-08-29:
The other two systems successfully went through an update with 24.8.6.

PeterSuh-Q3, 2024-08-29:
I need to check more, but it seems like there is a pattern.

https://diskdownload.synology.com/ssd.drivedb.db

When I tested it at home yesterday, I installed DSM by only connecting the SSD, and it showed the same message as above,
and when I replaced the SSD with an HDD, it proceeded normally.

It seems that checking the genuine serial number is not related.

eragon4k, 2024-08-29:
> the virtual disk created by VMWARE FUSION seems to be skipped by the compatibility

One of the unsuccessful devices used a genuine serial number and failed, but the update went through today with 24.8.6.

chinesepy, 2024-08-29:
同样的问题，我提供/var/log/messages里这部分日志供分析
`2024-08-29T16:30:48+08:00 N100 updater[27484]: updater.c:5991 Start of the updater...
2024-08-29T16:30:48+08:00 N100 updater[27486]: ccc.cpp:499 CCC host upgrade preparing...
2024-08-29T16:30:48+08:00 N100 updater[27486]: ccc.cpp:511 New version DSM is compatible with current cluster
2024-08-29T16:30:48+08:00 N100 updater[27486]: ccc.cpp:434 Start to shutdown guests before upgrading
2024-08-29T16:30:48+08:00 N100 update-entry[27159]: launch.cpp:363 ==== Finish updater pre-hook ====
2024-08-29T16:30:48+08:00 N100 update-entry[27159]: launch.cpp:175 Executing [/var/tmp/swap-raid-patch/upd@te/updater -v /var/tmp/swap-raid-patch]
2024-08-29T16:30:48+08:00 N100 updater[27493]: updater.c:5991 Start of the updater...
2024-08-29T16:30:48+08:00 N100 updater[27493]: updater.c:3287 orgBuildNumber = 69057.5, newBuildNumber = 72803.0
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:6297 ==== Start flash update ====
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:6301 This is X86 platform
2024-08-29T16:30:49+08:00 N100 kernel: [  608.348162] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
2024-08-29T16:30:49+08:00 N100 kernel: [  608.352529] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
2024-08-29T16:30:49+08:00 N100 kernel: [  608.355953] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
2024-08-29T16:30:49+08:00 N100 kernel: [  608.359318] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:6324 The SynoBoot partitions exist.
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:4149 SYNORedBootUpdCheckAndApply(4149): Skip bootloader update, no uboot_do_upd.sh exists
2024-08-29T16:30:49+08:00 N100 kernel: [  608.362778] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:613 checksum generated by [/bin/grep -i zImage checksum.syno >> /tmp/checksum.syno.tmp]
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:642 file [zImage] is being copied to [/tmp/bootmnt]
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/zImage] process done.
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:613 checksum generated by [/bin/grep -i rd.gz checksum.syno >> /tmp/checksum.syno.tmp]
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:642 file [rd.gz] is being copied to [/tmp/bootmnt]
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/rd.gz] process done.
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:613 checksum generated by [/bin/grep -i grub_cksum.syno checksum.syno >> /tmp/checksum.syno.tmp]
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:642 file [grub_cksum.syno] is being copied to [/tmp/bootmnt]
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/grub_cksum.syno] process done.
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:633 file [/tmp/bootmnt/hda1.tgz] is being removed
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/hda1.tgz] process done.
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:633 file [/tmp/bootmnt/updater] is being removed
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/updater] process done.
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:633 file [/tmp/bootmnt/VERSION] is being removed
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/VERSION] process done.
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:613 checksum generated by [/bin/grep -i model.dtb checksum.syno >> /tmp/checksum.syno.tmp]
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:642 file [model.dtb] is being copied to [/tmp/bootmnt]
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/model.dtb] process done.
2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:660 checksum file updated by [/bin/cp -f /tmp/checksum.syno.tmp /tmp/bootmnt/checksum.syno]
2024-08-29T16:30:55+08:00 N100 kernel: [  614.728005] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:1211 This model SA6400
 needs no factory upgrade
2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:1844 fail to mount ESP
2024-08-29T16:30:55+08:00 N100 kernel: [  614.777663] Module [phy_alloc] is removed. 
2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:6536 fail to upgrade BIOS
2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:746 Restoring [zImage]
2024-08-29T16:30:55+08:00 N100 kernel: [  614.787519] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:757 Copying file [/var/tmp/swap-raid-patch/synob@ckup/zImage] as [/tmp/bootmnt/zImage], result[0]
2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:746 Restoring [rd.gz]
2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:757 Copying file [/var/tmp/swap-raid-patch/synob@ckup/rd.gz] as [/tmp/bootmnt/rd.gz], result[0]
2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:743 Skip restoring [vendor]
[... cut ...]
2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:6911 Failed to accomplish the update! (errno = 21)
2024-08-29T16:30:55+08:00 N100 update-entry[27159]: launch.cpp:179 failed to execute updater cmd=[/var/tmp/swap-raid-patch/upd@te/updater -v /var/tmp/swap-raid-patch], r=21
2024-08-29T16:30:55+08:00 N100 update-entry[27159]: launch.cpp:539 synoinstall: System update stage 'launch-updater' failed
2024-08-29T16:30:55+08:00 N100 update-entry[27159]: launch.cpp:376 ==== Start updater post-hook ====
2024-08-29T16:30:55+08:00 N100 updater[27606]: updater.c:5991 Start of the updater...`

wjz304, 2024-08-29:
> ```
> Aug 29 03:30:43  synodiskupdatehclport: misc/misc_curl_to_buffer_or_file.c:115 Failed to curl url [...]
[image]

try `echo "success" > /tmp/update_hcl_via_sa_file_result`

PeterSuh-Q3, 2024-08-29:
> try `echo "success" > /tmp/update_hcl_via_sa_file_result`

It doesn't work. The same error keeps happening.

[image]

PeterSuh-Q3, 2024-08-30:
There was one mistake while generating 7.2.2 config.json for mshell.

The ramdisk-005-disable-disabled-ports.patch was missing.

[image]

The contents of DisabledPortDisks were deleted as above, so it was patched and now Non-DT models can be installed on bare metal.

I think the issue can be closed.

Thanks.

PeterSuh-Q3, 2024-08-30:
> ESP

DS3622xs+ 에서의 모든 이슈가 해결된것 입니까?
아니면 아직도 RR 에서 에러코드 21이 발생합니까?

PeterSuh-Q3, 2024-08-30:
> > the virtual disk created by VMWARE FUSION seems to be skipped by the compatibility
> 
> One of the unsuccessful devices used a genuine serial number and failed, but the update went through today with 24.8.6.

DS3622xs+ 에서의 모든 이슈가 해결된것 입니까?
아니면 아직도 RR 에서 에러코드 21이 발생합니까?

eragon4k, 2024-08-30:
> > > the virtual disk created by VMWARE FUSION seems to be skipped by the compatibility
> > 
> > 
> > One of the unsuccessful devices used a genuine serial number and failed, but the update went through today with 24.8.6.
> 
> DS3622xs+ 에서의 모든 이슈가 해결된것 입니까? 아니면 아직도 RR 에서 에러코드 21이 발생합니까?

RR:24.8.6 으로 업데이트후 문제가있던 ds3622xs  device 2개 모두다 문제업이 7.2.2 오 업데이트 할수있엇습니다.

wjz304, 2024-08-30:
```
Aug 30 18:36:03 scemd: net/net_wol_enable.c:117 Cannot get current wake-on-lan settings
Aug 30 18:36:03 scemd: net/net_wol_enable.c:256 Failed to set wake-on-lan eth1 d
Aug 30 18:36:03 scemd: scemd.c:288 failed to disable WOL. Interface [eth1]
Aug 30 18:36:03 findhostd: ninstaller.c:689 Dev: sda, DiskPath: /dev/sda, Partition version: 9
Aug 30 18:36:03 findhostd: ninstaller.c:129 Mount partion /dev/md0 /tmpRoot
Aug 30 18:36:03 kernel: [   34.281094] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Aug 30 18:36:03 kernel: [   34.375239] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Aug 30 18:36:03 findhostd: ninstaller.c:158 umount partition /tmpRoot
Aug 30 18:36:03 avahi-daemon[6237]: /etc/avahi/services/http.service: parse failure: service group incomplete.
Aug 30 18:36:03 avahi-daemon[6237]: Failed to load service group file /etc/avahi/services/http.service, ignoring.
Aug 30 18:36:04 kernel: [   34.478989] usbcore: registered new interface driver usb-storage
Aug 30 18:36:04 kernel: [   34.500306] usbcore: registered new interface driver usbhid
Aug 30 18:36:04 kernel: [   34.505907] usbhid: USB HID core driver
Aug 30 18:36:04 synohwctrl: invalid microp command 16388
Aug 30 18:36:04 findhostd: ninstaller.c:129 Mount partion /dev/md0 /tmpRoot
Aug 30 18:36:04 kernel: [   34.583018] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Aug 30 18:36:04 findhostd: ninstaller.c:158 umount partition /tmpRoot
Aug 30 18:36:04 kernel: [   34.612171] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Aug 30 18:36:04 findhostd: ninstaller.c:129 Mount partion /dev/md0 /tmpRoot
Aug 30 18:36:04 kernel: [   34.651339] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
Aug 30 18:36:04 kernel: [   34.684699] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
Aug 30 18:36:04 findhostd: ninstaller.c:436 failed to get upnpmodelname from [/tmpRoot/etc.defaults/synoinfo.conf].
Aug 30 18:36:04 findhostd: ninstaller.c:443 failed to get buildnumber from [/tmpRoot/etc.defaults/VERSION].
Aug 30 18:36:04 findhostd: ninstaller.c:450 failed to get productversion from [/tmpRoot/etc.defaults/VERSION].
Aug 30 18:36:04 findhostd: ninstaller.c:452 failed to get majorversion from [/tmpRoot/etc.defaults/VERSION].
Aug 30 18:36:04 findhostd: ninstaller.c:455 failed to get minorversion from [/tmpRoot/etc.defaults/VERSION].
Aug 30 18:36:04 findhostd: lib/system/system_platform_get.c:35 failed to get unique from /tmpRoot/etc.defaults/synoinfo.conf errno=[0x0900 file_get_key_value.c:28]
Aug 30 18:36:04 findhostd: ninstaller.c:467 failed to get platform from [/tmpRoot/etc.defaults/synoinfo.conf].
Aug 30 18:36:04 findhostd: lib/system/system_max_align_get.c:86 [/tmpRoot/.system_info/pgsql_alignment] doesn't exist, check model name for alignment
Aug 30 18:36:04 findhostd: lib/system/system_endian_get.c:61 [/tmpRoot/.system_info/endian] doesn't exist, check model name for endian
Aug 30 18:36:04 findhostd: lib/system/system_bit_get.c:57 [/tmpRoot/.system_info/bits] doesn't exist, check model name for bits
Aug 30 18:36:04 findhostd: ninstaller.c:500 fail to get admin port from /tmpRoot/etc/synoinfo.conf... try another way
Aug 30 18:36:04 findhostd: ninstaller.c:380 /tmpRoot/usr/syno/etc/www/DSM.json not exist
Aug 30 18:36:04 findhostd: ninstaller.c:503 fail to get admin port from /tmpRoot/usr/syno/etc/www/DSM.json
Aug 30 18:36:04 findhostd: ninstaller.c:531 failed to get nvr key from [/tmpRoot/etc.defaults/synoinfo.conf].
Aug 30 18:36:04 findhostd: root_block_get.c:42 Failed to get op info for [/dev/md][0x0900 root_block_op_info_get.c:78]
Aug 30 18:36:04 findhostd: root_block_get.c:123 Failed to read root block from [/dev/md][0x0900 root_block_op_info_get.c:78]
Aug 30 18:36:04 findhostd: util_junior.c:348 fail to read [unique] from [/tmpRoot//etc.defaults/synoinfo.conf]
Aug 30 18:36:04 findhostd: ninstaller.c:870 Model not match [/etc.defaults/synoinfo.conf, /tmpRoot//etc.defaults/synoinfo.conf]
Aug 30 18:36:04 findhostd: ninstaller.c:1227 SYSTEM_NORMAL, [Recoverable=0]
Aug 30 18:36:04 findhostd: ninstaller.c:158 umount partition /tmpRoot
Aug 30 18:36:04 findhostd: ninstaller.c:1142(FillUpgradeVolumeInfo): gszUpgradeVolDev = /dev/md0 
Aug 30 18:36:04 findhostd: ninstaller.c:1143(FillUpgradeVolumeInfo): gszUpgradeVolMnt = /tmpData 
Aug 30 18:36:04 findhostd: ninstaller.c:1245 gblSupportRaid: 1, gSysStatus: 1, gblCreateDataVol: 0, gblSystemRecoverable: 0
Aug 30 18:36:04 inetd[6378]: telnet/tcp: bind: Address already in use
Aug 30 18:36:04 syno_extract_junior_expansion_pack: lib/synoarchive.c:596 synoinstall: synoarchive: [/tmp/JEP.jep][OPEN] Type: JuniorExpansionPack
Aug 30 18:36:04 syno_extract_junior_expansion_pack: lib/synoarchive.c:84 synoinstall: synoarchive: [/tmp/JEP.jep][INIT] [/tmp/JEP.extract]
Aug 30 18:36:04 syno_extract_junior_expansion_pack: lib/synoarchive.c:92 synoinstall: synoarchive: [/tmp/JEP.jep][OPEN] Start
Aug 30 18:36:04 syno_extract_junior_expansion_pack: lib/synoarchive.c:106 synoinstall: synoarchive: [/tmp/JEP.jep][OPEN] End: [Success][3688618971][0.035s]
Aug 30 18:36:04 syno_extract_junior_expansion_pack: lib/synoarchive.c:286 synoinstall: synoarchive: [/tmp/JEP.jep][EXTRACT_ALL] Start
Aug 30 18:36:04 syno_extract_junior_expansion_pack: lib/synoarchive.c:303 synoinstall: synoarchive: [/tmp/JEP.jep][EXTRACT_ALL] End: [Success][3688618971][0.041s]
[... cut ...]
Aug 30 18:38:33 install.cgi: ErrFHOSTCleanPatchDirFile: Remove /tmpData/upd@te.pat...
Aug 30 18:38:33 install.cgi: ErrFHOSTDoUpgrade(1845): child process failed, retv=-21
Aug 30 18:38:33 install.cgi: ninstaller.c:1862(ErrFHOSTDoUpgrade) err=[-1]
Aug 30 18:38:33 install.cgi: ninstaller.c:1866(ErrFHOSTDoUpgrade) retv=[-21]
Aug 30 18:38:33 install.cgi: install.c:462 Upgrade for reinstll by the download patch fail.
Aug 30 18:38:33 install.cgi: install.c:692 Upgrade by download patch fail.
Jan  1 00:00:00 install.cgi: ninstaller.c:158 umount partition /tmpData

```

wjz304, 2024-08-30:
Aug 30 18:38:29 updater: boot/boot_lock.c(228): failed to mount boot device /dev/synoboot2 /tmp/bootmnt (errno:2)
Aug 30 18:38:29 updater: updater.c:6167 Failed to mount boot partition

PeterSuh-Q3, 2024-08-31:
> Aug 30 18:38:29 updater: boot/boot_lock.c(228): failed to mount boot device /dev/synoboot2 /tmp/bootmnt (errno:2) Aug 30 18:38:29 updater: updater.c:6167 Failed to mount boot partition

Is this the case that was caused by using the genuine SN for SA6400, as mentioned yesterday? But why is the revision 42962.0? Is it a JUN characteristic that should be used from 42962 when building 7.2.2?

wjz304, 2024-08-31:
I haven't had time to watch it yet. I've been busy these days.

PeterSuh-Q3, 2024-08-31:
> I haven't had time to watch it yet. I've been busy these days.

Get enough rest and let's take it slow. Have a nice weekend.^^

wjz304, 2024-08-31:
> > Aug 30 18:38:29 updater: boot/boot_lock.c(228): failed to mount boot device /dev/synoboot2 /tmp/bootmnt (errno:2) Aug 30 18:38:29 updater: updater.c:6167 Failed to mount boot partition
> 
> Is this the case that was caused by using the genuine SN for SA6400, as mentioned yesterday? But why is the BuildNumber 42962.0? Is it a JUN characteristic that should be used from 42962 when building 7.2.2?

`Aug 30 18:38:23 install.cgi: generic_hash.c:80 Generic hash: /tmpData//@autoupdate/DSM_DS918%2B_42962.pat: 27f81b2248eac790f191aac1558f573a039c7d4313df0eba659487f6e84ca23b`
This should be the log of installing 42962

wjz304, 2024-08-31:
This log was posted by someone in the TG group. I only discovered it later. I posted it here when I saw error 21.

> ```
> Aug 30 18:36:03 scemd: net/net_wol_enable.c:117 Cannot get current wake-on-lan settings
> Aug 30 18:36:03 scemd: net/net_wol_enable.c:256 Failed to set wake-on-lan eth1 d
> Aug 30 18:36:03 scemd: [...]

PeterSuh-Q3, 2024-09-05:
I have confirmed that this H2OFFT-Lx64-0815 patch (blocking redpill-lkm firmware updates) actually causes KP on KVM-based Proxmox or QEMU. 
I believe that only Apollolake needs the H2OFFT-Lx64-0815 patch. 
Originally, this firmware update blocking lkm module was said to crash the virtual CPU when running on KVM. 
I think I was lucky that the Sata-type loader worked without any problems on KVM so far.

closed 2024-09-07 by PeterSuh-Q3

chinesepy, 2024-09-16:
> 同样的问题，我提供/var/log/messages里这部分日志提供分析
> 
> ```
> 2024-08-29T16:30:48+08:00 N100 updater[27484]: updater.c:5991 Start of the updater...
> 2024-08-29T16:30:48+08:00 N100 updater[27486]: ccc.cpp:499 CCC host upgrade preparing...
> 2024-08-29T16:30:48+08:00 N100 updater[27486]: ccc.cpp:511 New version DSM is compatible with current cluster
> 2024-08-29T16:30:48+08:00 N100 updater[27486]: ccc.cpp:434 Start to shutdown guests before upgrading
> 2024-08-29T16:30:48+08:00 N100 update-entry[27159]: launch.cpp:363 ==== Finish updater pre-hook ====
> 2024-08-29T16:30:48+08:00 N100 update-entry[27159]: launch.cpp:175 Executing [/var/tmp/swap-raid-patch/upd@te/updater -v /var/tmp/swap-raid-patch]
> 2024-08-29T16:30:48+08:00 N100 updater[27493]: updater.c:5991 Start of the updater...
> 2024-08-29T16:30:48+08:00 N100 updater[27493]: updater.c:3287 orgBuildNumber = 69057.5, newBuildNumber = 72803.0
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:6297 ==== Start flash update ====
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:6301 This is X86 platform
> 2024-08-29T16:30:49+08:00 N100 kernel: [  608.348162] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
> 2024-08-29T16:30:49+08:00 N100 kernel: [  608.352529] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
> 2024-08-29T16:30:49+08:00 N100 kernel: [  608.355953] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
> 2024-08-29T16:30:49+08:00 N100 kernel: [  608.359318] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:6324 The SynoBoot partitions exist.
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:4149 SYNORedBootUpdCheckAndApply(4149): Skip bootloader update, no uboot_do_upd.sh exists
> 2024-08-29T16:30:49+08:00 N100 kernel: [  608.362778] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:613 checksum generated by [/bin/grep -i zImage checksum.syno >> /tmp/checksum.syno.tmp]
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:642 file [zImage] is being copied to [/tmp/bootmnt]
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/zImage] process done.
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:613 checksum generated by [/bin/grep -i rd.gz checksum.syno >> /tmp/checksum.syno.tmp]
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:642 file [rd.gz] is being copied to [/tmp/bootmnt]
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/rd.gz] process done.
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:613 checksum generated by [/bin/grep -i grub_cksum.syno checksum.syno >> /tmp/checksum.syno.tmp]
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:642 file [grub_cksum.syno] is being copied to [/tmp/bootmnt]
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/grub_cksum.syno] process done.
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:633 file [/tmp/bootmnt/hda1.tgz] is being removed
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/hda1.tgz] process done.
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:633 file [/tmp/bootmnt/updater] is being removed
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/updater] process done.
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:633 file [/tmp/bootmnt/VERSION] is being removed
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/VERSION] process done.
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:613 checksum generated by [/bin/grep -i model.dtb checksum.syno >> /tmp/checksum.syno.tmp]
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:642 file [model.dtb] is being copied to [/tmp/bootmnt]
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:650 file [/tmp/bootmnt/model.dtb] process done.
> 2024-08-29T16:30:49+08:00 N100 updater[27493]: updater.c:660 checksum file updated by [/bin/cp -f /tmp/checksum.syno.tmp /tmp/bootmnt/checksum.syno]
> 2024-08-29T16:30:55+08:00 N100 kernel: [  614.728005] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
> 2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:1211 This model SA6400
>  needs no factory upgrade
> 2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:1844 fail to mount ESP
> 2024-08-29T16:30:55+08:00 N100 kernel: [  614.777663] Module [phy_alloc] is removed. 
> 2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:6536 fail to upgrade BIOS
> 2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:746 Restoring [zImage]
> 2024-08-29T16:30:55+08:00 N100 kernel: [  614.787519] ext2 filesystem being mounted at /tmp/bootmnt supports timestamps until 2038 (0x7fffffff)
> 2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:757 Copying file [/var/tmp/swap-raid-patch/synob@ckup/zImage] as [/tmp/bootmnt/zImage], result[0]
> 2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:746 Restoring [rd.gz]
> 2024-08-29T16:30:55+08:00 N100 updater[27493]: updater.c:757 Copying file [/var/tmp/swap-raid-patch/synob@ckup/rd.gz] as [/tmp/bootmnt/rd.gz], result[0]
[... cut ...]
> 2024-08-29T16:30:55+08:00 N100 update-entry[27159]: launch.cpp:179 failed to execute updater cmd=[/var/tmp/swap-raid-patch/upd@te/updater -v /var/tmp/swap-raid-patch], r=21
> 2024-08-29T16:30:55+08:00 N100 update-entry[27159]: launch.cpp:539 synoinstall: System update stage 'launch-updater' failed
> 2024-08-29T16:30:55+08:00 N100 update-entry[27159]: launch.cpp:376 ==== Start updater post-hook ====
> 2024-08-29T16:30:55+08:00 N100 updater[27606]: updater.c:5991 Start of the updater...
> ```

中秋节放假有时间自己试了一下，最后试出来是把固态硬盘的存储池删掉后拆掉固态硬盘就可以更新了
不过移动文件和重新安装应用真是超级麻烦，后续不打算用固态硬盘作为存储空间了...