RROrg/rr #2646: BUG realtek R8168 not working

ksd3971, 2024-08-28:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  native
 * CPU:  i7 12800h
 * NIC:  10ec:8125 (rev 05)/10ec:8168 (rev 15)

RR version: (You can find it in the update menu.)  
 * RR:  24.8.4
 * addons:  mountloader/cpuinfo/powersched/reboottoloader/addincards/acpid/dbgutils/storagepannel/sensor
 * modules: coretemp/ipv6/r8125/r8168
 * lkms:  prod

DSM: 
 * model:  sa6400 native
 * version:  7.2.2 72803

Issue:  realtek r8168 ethernet card not working

The R8168 network card driver did not work well in versions 24.8.0 and 24.8.4. So, I included the R8168 driver from version 24.4.8 into 24.8.0, and it worked fine. However, this method no longer works in 24.8.4

logs:  

[logs.tar.gz](https://github.com/user-attachments/files/16784080/logs.tar.gz)
[serial.log](https://github.com/user-attachments/files/16784086/serial.log)

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

nillebor, 2024-08-28:
You can test "direct Boot" = true in Advance Settings

ksd3971, 2024-08-28:
I tried setting directboot = 'true', but it still doesn't work

> You can test "direct Boot" = true in Advance Settings

wjz304, 2024-08-28:
8125 工作正常吗？

ksd3971, 2024-08-29:
> 8125 工作正常吗？

it work's well on 24.8.x but don't works 24.7.x

wjz304, 2024-08-29:
in ssh of DSM:
```
lsmod | grep r81
dmesg | grep r8168 -A 20
ifconfig
ls /sys/class/net/*
```

ksd3971, 2024-08-29:
> in ssh of DSM:
> 
> ```
> lsmod | grep r81
> dmesg | grep r8168 -A 20
> ifconfig
> ls /sys/class/net/*
> ```

@SA:/$ lsmod|grep r81
r8168                 593920  0
r8125                 241664  0
@SA:/$ dmesg |grep r8168 -A 20
@SA:/$ ifconfig
eth0      Link encap:Ethernet  HWaddr 00:E2:69:70:58:8D
          inet addr:192.168.137.29  Bcast:192.168.137.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3025 errors:0 dropped:0 overruns:0 frame:0
          TX packets:314 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:573649 (560.2 KiB)  TX bytes:43234 (42.2 KiB)
          Interrupt:16

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:4 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:240 (240.0 B)  TX bytes:240 (240.0 B)

@SA:/$ ls /sys/class/net/*
/sys/class/net/eth0:
addr_assign_type    dev_port           name_assign_type      rtl_adv
address             dormant            napi_defer_hard_irqs  speed
addr_len            duplex             netdev_group          statistics
broadcast           flags              operstate             subsystem
carrier             gro_flush_timeout  phys_port_id          syno_eth_info
carrier_changes     ifalias            phys_port_name        testing
carrier_down_count  ifindex            phys_switch_id        tx_queue_len
carrier_up_count    iflink             power                 type
device              link_mode          proto_down            uevent
dev_id              mtu                queues

/sys/class/net/lo:
addr_assign_type    dormant            napi_defer_hard_irqs  statistics
address             duplex             netdev_group          subsystem
addr_len            flags              operstate             syno_eth_info
broadcast           gro_flush_timeout  phys_port_id          testing
carrier             ifalias            phys_port_name        tx_queue_len
carrier_changes     ifindex            phys_switch_id        type
carrier_down_count  iflink             power                 uevent
carrier_up_count    link_mode          proto_down
dev_id              mtu                queues
dev_port            name_assign_type   speed

/sys/class/net/sit0:
addr_assign_type    dormant            napi_defer_hard_irqs  statistics
address             duplex             netdev_group          subsystem
addr_len            flags              operstate             syno_eth_info
broadcast           gro_flush_timeout  phys_port_id          testing
carrier             ifalias            phys_port_name        tx_queue_len
carrier_changes     ifindex            phys_switch_id        type
carrier_down_count  iflink             power                 uevent
carrier_up_count    link_mode          proto_down
dev_id              mtu                queues
dev_port            name_assign_type   speed
@SA:/$

wjz304, 2024-08-29:
```
ls /usr/lib/modules/r8* -al
rmmod r8168
insmod /usr/lib/modules/r8168.ko
dmesg | grep r8168 -A 20
```

ksd3971, 2024-08-30:
> You can also try to load the extracted r8168.ko under the system rmmod r8168 modinfo /r8168.ko insmod /r8168.ko

I have tried everything you suggested, but it did not work.

I tried using r8168.ko version 24.5.0 this time, and it worked.

I still don't understand why this works.

[image]
Something is different, but I’m not sure what.

SynologyNAS> ifconfig
eth0      Link encap:Ethernet  HWaddr 00:E2:69:70:58:8D
          inet addr:192.168.137.204  Bcast:192.168.137.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7711 errors:0 dropped:0 overruns:0 frame:0
          TX packets:890 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:10480179 (9.9 MiB)  TX bytes:73342 (71.6 KiB)
          Interrupt:16

eth1      Link encap:Ethernet  HWaddr 00:E2:69:70:58:8C
          inet addr:169.254.99.68  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:170 Base address:0x9000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

SynologyNAS> modinfo /usr/lib/modules/r8168_2450.ko
filename:       /usr/lib/modules/r8168_2450.ko
version:        8.053.00-RSS
license:        GPL
description:    RealTek RTL-8168 Gigabit Ethernet driver (Compiled by RR for DSM)
author:         Realtek and the Linux r8168 crew <netdev@vger.kernel.org>
srcversion:     C9EE9120D3DBA243144CD19
alias:          pci:v00001186d00004300sv00001186sd00004B10bc*sc*i*
alias:          pci:v000010ECd00002600sv*sd*bc*sc*i*
alias:          pci:v000010ECd00002502sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008161sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008168sv*sd*bc*sc*i*
depends:
retpoline:      Y
name:           r8168
vermagic:       5.10.55+ SMP mod_unload
parm:           speed_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           duplex_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           autoneg_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           advertising_mode:force phy operation. Deprecated by ethtool (8). (uint)
parm:           dynamic_aspm:int
parm:           aspm:Enable ASPM. (int)
parm:           s5wol:Enable Shutdown Wake On Lan. (int)
parm:           s5_keep_curr_mac:Enable Shutdown Keep Current MAC Address. (int)
parm:           use_dac:Enable PCI DAC. Unsafe on 32 bit PCI slot. (int)
parm:           timer_count:Timer Interrupt Interval. (int)
parm:           eee_enable:Enable Energy Efficient Ethernet. (int)
parm:           hwoptimize:Enable HW optimization function. (ulong)
parm:           s0_magic_packet:Enable S0 Magic Packet. (int)
parm:           dynamic_aspm_packet_threshold:Dynamic ASPM packet threshold. (int)
parm:           disable_wol_support:Disable PM support. (int)
parm:           debug:Debug verbosity level (0=none, ..., 16=all) (int)

wjz304, 2024-08-31:
You load it yourself first, and then you need to confirm some things, but I'm busy these days and don't have much time.

wjz304, 2024-09-06:
https://github.com/RROrg/rr/issues/2769

q331723516, 2024-09-07:
> 在 DSM 的 ssh 中：
> 
> ```
> lsmod | grep r81
> dmesg | grep r8168 -A 20
> ifconfig
> ls /sys/class/net/*
> ```
[image]

q331723516, 2024-09-07:
> in ssh of DSM:
> 
> ```
> lsmod | grep r81
> dmesg | grep r8168 -A 20
> ifconfig
> ls /sys/class/net/*
> ```
dsm：
[image]
[image]
RR:
[image]
[image]

q331723516, 2024-09-07:
> #2769

尝试以上方法依然没有解决问题

wjz304, 2024-09-08:
[r8168.zip](https://github.com/user-attachments/files/16920773/r8168.zip)
test this

wjz304, 2024-09-11:
https://github.com/RROrg/rr/issues/2225

wjz304, 2024-10-24:
无人反馈，本issue关闭

closed 2024-10-24 by wjz304