RROrg/rr #2722: BUG

namezys, 2024-08-30:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  Intel(R) Core(TM) i5-9500T CPU @ 2.20GHz
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  v24.8.6
 * lkms:  prod
Direct bool

DSM: 
 * model:  DS920+
 * version:  7.2 (72803)

Issue:  

Is it possible to recognize GPU from 9th generation for hardware encoding/decoding/etc? I found that VID:PIC "8086:3e92" is not recognized even it is supported.

I'm trying to find information on how to do it but I was able to get only a script that made hex patch.
 
```
root@rr:/opt/rr# lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers [8086:3ec2] (rev 07)
00:02.0 VGA compatible controller [0300]: Intel Corporation CoffeeLake-S GT2 [UHD Graphics 630] [8086:3e92]
00:12.0 Signal processing controller [1180]: Intel Corporation Cannon Lake PCH Thermal Controller [8086:a379] (rev 10)
00:14.0 USB controller [0c03]: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller [8086:a36d] (rev 10)
00:14.2 RAM memory [0500]: Intel Corporation Cannon Lake PCH Shared SRAM [8086:a36f] (rev 10)
00:17.0 SATA controller [0106]: Intel Corporation Cannon Lake PCH SATA AHCI Controller [8086:a352] (rev 10)
00:1b.0 PCI bridge [0604]: Intel Corporation Cannon Lake PCH PCI Express Root Port #17 [8086:a340] (rev f0)
00:1b.4 PCI bridge [0604]: Intel Corporation Cannon Lake PCH PCI Express Root Port #21 [8086:a32c] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Q370 Chipset LPC/eSPI Controller [8086:a306] (rev 10)
00:1f.4 SMBus [0c05]: Intel Corporation Cannon Lake PCH SMBus Controller [8086:a323] (rev 10)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH SPI Controller [8086:a324] (rev 10)
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (7) I219-LM [8086:15bb] (rev 10)
01:00.0 SATA controller [0106]: ASMedia Technology Inc. ASM1166 Serial ATA Controller [1b21:1166] (rev 02)
02:00.0 Ethernet controller [0200]: Mellanox Technologies MT27500 Family [ConnectX-3] [15b3:1003]
```

I hope I skip some simple method to do it.

namezys, 2024-08-30:
logs. It's a test build so nothing sensitive.
[logs (1).tar.gz](https://github.com/user-attachments/files/16822803/logs.1.tar.gz)

namezys, 2024-08-31:
To be honest, I expected some information here:

[image]
[image]

wjz304, 2024-08-31:
AME and igpu have no direct relationship

wjz304, 2024-08-31:
https://github.com/007revad/Video_Station_for_DSM_722
https://imnks.com/385.html
# DSM7.2 AME版本3.1.0-3005
curl -L http://code.imnks.com/ame3patch/ame72-3005.py | python

namezys, 2024-08-31:
Thank you.

Can you clarify/correct me if I'm not correct?

* https://github.com/007revad/Video_Station_for_DSM_722 -- Install Video Station on DSM 7.2.2
  * However, it removes AME v4 and replaces by  3.1.0-3005
* https://imnks.com/385.html -- description of how to add features to AME
  * should be `ffmpeg` installed
  * add gtreamer for processing DTS/EAC3/TrueHD -- https://github.com/AlexPresso/VideoStation-FFMPEG-Patcher
  * remove checking of serial number -- https://github.com/wirgen/synocodectool-patch
  * add a lot of other codec combination -- https://github.com/darknebular/Wrapper_VideoStation
* `curl -L http://code.imnks.com/ame3patch/ame72-3005.py | python` -- Add offline license

However, my main goal is the ability to use **hardware encoder/decoder** by Synology services like photos to analyze content.
Transcoding is not necessary and destructive as I want to get untouched streams for decoding.

Of course, I would like to play around using cams and AI features but it is out of scope and unreachable as I can see.

OrpheeGT, 2024-09-03:
Just for a test, you should try DVA1622 loader.

kimsonvu, 2025-04-01:
> Thank you.
> 
> Can you clarify/correct me if I'm not correct?
> 
> * https://github.com/007revad/Video_Station_for_DSM_722 -- Install Video Station on DSM 7.2.2
>   
>   * However, it removes AME v4 and replaces by  3.1.0-3005
>   * works
> * https://imnks.com/385.html -- description of how to add features to AME
>   
>   * should be `ffmpeg` installed
>   * add GStreamer for processing DTS/EAC3/TrueHD -- https://github.com/AlexPresso/VideoStation-FFMPEG-Patcher
>     
>     * works
>   * remove checking of serial number -- https://github.com/wirgen/synocodectool-patch
>     
>     * version is not compatible
>   * add "everything" -- https://github.com/darknebular/Wrapper_VideoStation
>     
>     * Installation of the Advanced Wrapper: works (as I can it uses ffmpeg4)
>     * fix of AME doesn't work (different md5, in other words, different version)
> * `curl -L http://code.imnks.com/ame3patch/ame72-3005.py | python` -- Add offline license
> 
> However, my main goal is the ability to use **hardware encoder/decoder** by Synology services like photos to analyze content. Transcoding is not necessary and destructive as I want to get untouched streams for decoding.
> 
> Of course, I would like to play around using cams and AI features but it is out of scope and unreachable as I can see.

I have same iGPU with you Intel(R) Xeon(R) E-2124G CPU @ 3.40GHz: Intel Corporation CoffeeLake-S GT2 [UHD Graphics P630]

I test with Emby on Synology package and it detect right name igpu. I tried to install Ubuntu docker to check intel_gpu_top but i failed.

Some comment with you..