RROrg/rr #2724: Latest ova files not support esxi7.

Handsome1080P, 2024-08-31:
import got error:

`Failed to deploy OVF package. -1:-1:VALUE_ILLEGAL: No supported hardware versions among [vmx-21]; supported: [vmx-04, vmx-07, vmx-08, vmx-09, vmx-10, vmx-11, vmx-12, vmx-13, vmx-14, vmx-15, vmx-16, vmx-17, vmx-18, vmx-19].
`
Use vmware workstation changed the hardware version and imported esxi.Don't know this are some issues or not.

wjz304, 2024-08-31:
解包 ova，手动修改 ovf, 
[image]

然后导入 ovf with vmdk

Handsome1080P, 2024-09-01:
> 解包 ova，手动修改 ovf, [...]

not need do this.Just import to VM workstation then change hardware version.

closed 2024-09-01 by Handsome1080P