RROrg/rr #2764: BUG - Unable to run Virtual Machine Manger migrating from ARPL to RR

nando2301, 2024-09-02:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  To Be Filled By O.E.M. B660M-ITX/ac/B660M-ITX/ac, BIOS 1.25 12/20/2021
 * CPU:  12th Gen Intel(R) Core(TM) i5-12400
 * NIC:
00:1f.6 Class [0200]: Intel Corporation Ethernet Connection (17) I219-V [8086:1a1d] (rev 11)
01:00.0 Class [0108]: Device [1e4b:1202] (rev 01)
03:00.0 Class [0280]: Intel Corporation Dual Band Wireless-AC 3168NGW [Stone Peak] [8086:24fb] (rev 10)

RR version: (You can find it in the update menu.)  
 * RR:  RR v24.8.6
 * addons:  Default
 * modules:  cpuinfo
 * lkms:  prod

DSM: 
 * model:  DS3622xs+(broadwellnk)
 * version:  7.2(72803)

Issue:  
Cmdline:
withefi earlyprintk syno_hw_version=DS3622xs+ console=ttyS0,115200n8 consoleblank=600 netif_num=1 pid=0x1666 earlycon=uart8250,io,0x3f8,115200n8 mac1=a0---809db-- sn=2150SQR---- modprobe.blacklist=evbug,cdc_ether pcie_aspm=off rootwait SMBusHddDynamicPower=1 vid=0x0951 elevator=elevator loglevel=15 HddHotplug=1 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 syno_hdd_detect=0 panic=5 vender_format_version=2 syno_hdd_powerup_seq=0 log_buf_len=32M root=/dev/md0

I setted netif_num=1  mac1=a0---809db-- and sn=2150SQRC---- 

But my Virtual Machine Manger fails to get Network Interface, DSM detects my network card, then I putted Static IP

[image]

VMM Network fails
[image]

My VMs previously working fail to start

[image]

Please help me, this was working with @fbelavenuto's ARPL with DSM 7.1.

logs:  

Not obtained yet, please give some time.

nillebor, 2024-09-07:
What's your Network interface in VM? Virtio or e1000?
I also have my problems with virtio but e1000 works without problems.

nando2301, 2024-09-07:
Hi
I think it was VirtIO. Hcan I see that or change?, Now I don't see anything on Network, let me document that I did.

nillebor, 2024-09-07:
You can edit/change this in your VM > Network

nando2301, 2024-09-07:
Ok in this moment I don't have Network Card to add because I executed some steps to fix and now is not working.

nillebor, 2024-09-07:
What's your network settings in VM?
What kind of network card do you use?

nando2301, 2024-09-07:
I was doing these tests:

1. Boot RR on Configure loader again
2. Select all Modules from RR menu, including e1000 and e1000e

[image]

4. In `Cmdline` set `Ramdom` values.
5. In `Advanced menu` --) `Set static IP`

[image]

7. Then I executed `Initialize DSM network settings`

[image]

9. Now DSM works fine with Static IP but I don't have Network Cards on VMM.

nando2301, 2024-09-07:
> What's your network settings in VM? What kind of network card do you use?

I use the network card included in the board, is a `e100e` card `I-219V`

[image]

nillebor, 2024-09-07:
I have not entered any network data in the loader, but that will not be the cause.
How is your network setup in the VM machine itself? Where you can set the RAM, CPU, USB, BIOS, etc. You can also set the network card. Do you have a screenshot of it?

nando2301, 2024-09-07:
Sorry, NAS was booting

[image]

nando2301, 2024-09-07:
When I try to add card, assistent says this:

[image]

[image]

nando2301, 2024-09-07:
If it helps:

[image]

nillebor, 2024-09-07:
[image]

[image]

[image]

nando2301, 2024-09-07:
I have two USB Network Adapter Cards, when I connect these detects a new Network Card that works in VMM, but in restart desappear and all fails again. Do you know if `netif_num=1` has an effect here? I have this setted because it was in `ALPR`, the I setted it again here in `RR`.

[image]

nando2301, 2024-09-07:
> ![grafik](https://private-user-images.githubusercontent.com/59985723/365384404-8a16f246-b1bc-4e1b-823a-1c5cfc17003f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3Mjc2NDIsIm5iZiI6MTcyNTcyNzM0MiwicGF0aCI6Ii81OTk4NTcyMy8zNjUzODQ0MDQtOGExNmYyNDYtYjFiYy00ZTFiLTgyM2EtMWM1Y2ZjMTcwMDNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA3VDE2NDIyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThjNDY2YzhjMWMxNjUzY2RkZjBmYjUyY2QwNjJiMGJhNmY1MGViZTE0YzNkN2E0ZDZkZTY1MDNiYzZkZjBiMDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.32ybFoUxOURFaDnt3NtRECdObE2oR9WM7Bz3ZKAI5Zg)
> 
> ![grafik](https://private-user-images.githubusercontent.com/59985723/365384496-175e0c6b-c116-4343-a728-52be38f2c01b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3Mjc2NDIsIm5iZiI6MTcyNTcyNzM0MiwicGF0aCI6Ii81OTk4NTcyMy8zNjUzODQ0OTYtMTc1ZTBjNmItYzExNi00MzQzLWE3MjgtNTJiZTM4ZjJjMDFiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA3VDE2NDIyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVmMTU5OGEyYjcwYmQxZmUyZTUxNTMwZWE2ZWZlYTA1YTdhOGVkMDE2YjE4ZGFlZDhjMzBkYzJjYzM4NGQ3ZDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.N3Qvwwny6zreuinh0X_WrCMlRE83cfH5UZemetGVCg8)
> 
> ![grafik](https://private-user-images.githubusercontent.com/59985723/365384667-396ed033-a7c0-4e52-9c7c-710a57d2df26.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3Mjc2NDIsIm5iZiI6MTcyNTcyNzM0MiwicGF0aCI6Ii81OTk4NTcyMy8zNjUzODQ2NjctMzk2ZWQwMzMtYTdjMC00ZTUyLTljN2MtNzEwYTU3ZDJkZjI2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA3VDE2NDIyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU1NjgwMGIyZTU4YWYwMGIzNzYzY2NlMjdkMzg2NjcxMjExMmQyYmM5ZjNhZjNmZDFmZmUyNDExNzQ3NGVmYmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OzkUagVjzqDKA9zXbo1dEI41qa75yWADLOkL97WBHho)

In my case it returns an error, derived that clean DSM Network Settings on RR loader menu

[image]

nillebor, 2024-09-07:
[image]

I think it's your USB network adapters. the network chip is on board with me. What about the network card on your motherboard?

nando2301, 2024-09-07:
Yes is on board my card too, the USB Adaptes are only for test.

nando2301, 2024-09-07:
Reinstall VMM Package it says this:

[image]

[image]

[image]

[image]

[image]

nando2301, 2024-09-07:
The I add Staorage again:

[image]

nillebor, 2024-09-07:
The network card is not displayed under Network in VMM?

nando2301, 2024-09-07:
> Yes is on board my card too, the USB Adaptes are only for test.

No, no items detected, before clean DSM Network Setting it was there, but now anything is displayed.

[image]

nando2301, 2024-09-07:
> When I try to add card, assistent says this:
> <img alt="image" width="300" src="https://private-user-images.githubusercontent.com/32605309/365383757-035aae20-d911-420f-911f-1cee77150287.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3MjcwOTEsIm5iZiI6MTcyNTcyNjc5MSwicGF0aCI6Ii8zMjYwNTMwOS8zNjUzODM3NTctMDM1YWFlMjAtZDkxMS00MjBmLTkxMWYtMWNlZTc3MTUwMjg3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA3VDE2MzMxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVmNDE5NmRiOTE2NzdhMDQ3NTVlMDk1MzczOGVlOGFjNDEzYzQwZmI1ZGE5Yzk3YTQ3MTU0MzY4NzhlYjVmOWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0qmi6NIVpbAYTKVyPurBtH0eqi3rrhyRpsDQwiZz4qU"> <img alt="image" width="300" src="https://private-user-images.githubusercontent.com/32605309/365383765-3ca98b0d-0ed2-42d2-ba44-0aea983319fd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3MjcwOTEsIm5iZiI6MTcyNTcyNjc5MSwicGF0aCI6Ii8zMjYwNTMwOS8zNjUzODM3NjUtM2NhOThiMGQtMGVkMi00MmQyLWJhNDQtMGFlYTk4MzMxOWZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA3VDE2MzMxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY3N2RkYzQyYjRkNDJkN2I1M2UyODlmOTIxZjgzMzk3ZmZmMDQwNjc0Y2E5MzkwYTVjNjg3ZDk4MWMwYjdjYWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.95ISI9S5pIOk6ZvCBpG4qb7acyRnIAmR5jypc4-puGI">

Whe I try to add the Network Card it fails with the message previously shared here

nando2301, 2024-09-07:
This is my Panel Configuration about Network Card:

[image]

nillebor, 2024-09-07:
It's funny that there is no default VM network there. You use your onborad-Natzwerkkarte for the DSM? Have you tried another Diskstation version? I also have the latest software. We only use different DiskStation. What version of the VMM do you have? I have installed the 2.6.5-12202.

You came from another loader. Have you recorded a backup or re-set up the DS?

[image]

nando2301, 2024-09-07:
> It's funny that there is no default VM network there. You use your onborad-Natzwerkkarte for the DSM? Have you tried another Diskstation version? I also have the latest software. We only use different DiskStation. What version of the VMM do you have? I have installed the 2.6.5-12202.
> 
> You came from another loader. Have you recorded a backup or re-set up the DS?
> 
> ![grafik](https://private-user-images.githubusercontent.com/59985723/365387729-05148553-ebdd-450f-af1c-2d9fad09034b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3Mjk5ODMsIm5iZiI6MTcyNTcyOTY4MywicGF0aCI6Ii81OTk4NTcyMy8zNjUzODc3MjktMDUxNDg1NTMtZWJkZC00NTBmLWFmMWMtMmQ5ZmFkMDkwMzRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA3VDE3MjEyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU0NjVkYzdiYjBiMTllMmMxNTU1NWMzYmU3OGUzNmFhNzZmNzQ5ZDMwNjdmYTk2NDYwZGIwMGI1NzMwOGQ3M2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bY3810k1B7nVS08Lv1naTckZXp1NP2iqS1rYW_aApzE)

I do not know what is this `onborad-Natzwerkkarte`.

This is my Network info:

[image]

My version:

[image]

Then now I connected USB adapter and reboot and see this:

[image]

But my onboard network card not appears.

nillebor, 2024-09-07:
It's funny that the network card works in DSM, but not in VMM. VMM is just an app. I can then set up my test machine again to a 3622xs + and try to recreate it.

nando2301, 2024-09-07:
> It's funny that there is no default VM network there. You use your onborad-Natwork card for the DSM? Have you tried another Diskstation version? I also have the latest software. We only use different DiskStation. What version of the VMM do you have? I have installed the 2.6.5-12202.
> 
> You came from another loader. Have you recorded a backup or re-set up the DS?
> 
> ![grafik](https://private-user-images.githubusercontent.com/59985723/365387729-05148553-ebdd-450f-af1c-2d9fad09034b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3MzA0ODQsIm5iZiI6MTcyNTczMDE4NCwicGF0aCI6Ii81OTk4NTcyMy8zNjUzODc3MjktMDUxNDg1NTMtZWJkZC00NTBmLWFmMWMtMmQ5ZmFkMDkwMzRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA3VDE3Mjk0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIyNjJjMThjNWZjMTBkMTkzMDYwMDQwODhiNmNiNzM2YWVjMjVkZTA2NTQ2YjU0OTcyMzY2ZDU2Zjc3ODFmOTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OxYVGrbJ94Bomt3IsNO08y3rVmU39dm6ubu8W1bgQuo)

My DSM was `ALPR`, I have the USB, but create Other One from scratch with `RR` and do DSM Migration with pat file in the DSM assistant, pat obtained from RR Excel List.

I dot know how to backup as you say.

nillebor, 2024-09-07:
Clean Install DSM:
[image]

Install VMM:
[image]

Looks good here. I would reinstall the DS without any settings.

If the network card would not work with DSM, I would have to say the hardware. Since only the VMM is affected, the error must lie somewhere else. The VMM is probably not the problem either.

nando2301, 2024-09-07:
> Looks good here. I would reinstall the DS without any settings.
> 
> If the network card would not work with DSM, I would have to say the hardware. Since only the VMM is affected, the error must lie somewhere else. The VMM is probably not the problem either.

Probably, but you know hot to reinstall without loosing the configuration?, I have Dockers, Certificates, VMs and another things but without account the backups are disabled.

nillebor, 2024-09-07:
You should always have a backup!!!
You can easily create the backup with Hyperbackup (app). When importing, you can select what should be restored. This is also possible afterwards. Hyperbackup can place the backup on another DS, USB or other server.

But you have to store the containers extra. This can also be done via a script or via the GUI. The certificate can also be secured via GUI. It is better to apply for a new one right away, but it is best to apply with acme.sh . Of course, the VM can be backed up from.

But you may also have an old PC for testing.

nando2301, 2024-09-21:
Hi, I solved my issue.

How I solved this:

1. I recreated a new RR Loader USB with default settings.
2. I connected a USB LAN Adapter.
3. I was working before with another virtualization applications and then remember about virtual cards on Linux operating System. Then I check script on Synology `/etc/sysconfig/network-scripts` folder.
4. The eth0 network cards was deleted from there.
5. How network card was configured
```
root@Cloudy:/etc/sysconfig/network-scripts# ls -1
ifcfg-eth0
ifcfg-eth1
ifcfg-eth2
ifcfg-eth3
ifcfg-eth4
ifcfg-eth5
ifcfg-eth6
ifcfg-eth7
ifcfg-lo
ifcfg-ovs_eth1
ifcfg-tun
```
If you see I have a lot of ifcfg-ethX cards but the important thing here are `ifcfg-eth0, ifcfg-eth1` and `ifcfg-ovs_eth1`. I lose ifcfg-ovs_eth0 network script, and I was setting eth0 as Fixed IP.
7. I setted eth0 as DHCP and then compare `ifcfg-eth0` with `ifcfg-eth1`.
ifcfg-eth0:
```
DEVICE=eth0
BOOTPROTO=dhcp
ONBOOT=yes
IPV6INIT=dhcp
IPV6_ACCEPT_RA=1
```
ifcfg-eth1:
```
DEVICE=eth1
BOOTPROTO=dhcp
ONBOOT=yes
IPV6INIT=dhcp
IPV6_ACCEPT_RA=1
BRIDGE=ovs_eth1
```
Then the bridge is not configure in `eth0` network card. I added bridge to `ifcfg-eth0`.
ifcfg-eth0:
```
DEVICE=eth0
BOOTPROTO=dhcp
ONBOOT=yes
IPV6INIT=dhcp
IPV6_ACCEPT_RA=1
BRIDGE=ovs_eth0
```
9. Afted edit de file I need to create Bridge Interface `ifcfg-ovs_eth0`, I copied `ifcfg-ovs_eth1` to `ifcfg-ovs_eth0`.
ifcfg-ovs_eth0:
```
DEVICE=ovs_eth0
BOOTPROTO=dhcp
ONBOOT=yes
TYPE=OVS
PRIMARY=eth0
IPV6INIT=dhcp
IPV6_ACCEPT_RA=1
```
11. After this change I disconnected my USB LAN Adapter and connect the cable to OnBoard Network Card and my connection problem un VMM solved.
My scripts now:
```
root@Cloudy:/etc/sysconfig/network-scripts# ls -1
ifcfg-eth0
ifcfg-eth1
ifcfg-eth2
ifcfg-eth3
ifcfg-eth4
ifcfg-eth5
ifcfg-eth6
ifcfg-eth7
ifcfg-lo
ifcfg-ovs_eth0
ifcfg-tun
```
13. In VMM I needed to create again my network and restart my Synology.
[image]

[image]

Thanks a lot for your help.

closed 2024-10-25 by wjz304