RROrg/rr #2766: BUG vmware 安装黑群晖sa6400 第二天无法连接

hanfenglong315, 2024-09-02:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  N100
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.8.6
 * addons:  
 * modules:  SA6400
 * lkms:  

DSM: 
 * model:  SA-6400
 * version:  7.2

Issue:  24.8.6
[image]

N100小主机 安装vmware17.x vm里安装了黑群晖sa6400 
昨天还能稳定运行一天 今天就连接不上 重启vm显示

vm群晖 eth0 connected.
Waiting
IP
eth0(e1000): LINK LOCAL (No DHCP server detected.) 报这个错误 请问咋回事
重启物理机才好用
给了2核心
8g内存

之前的ds920，ds918 都有这种问题
vmware里的黑群晖没有挂 就是莫名其秒连接不上 
然后必须物理机器重启才行

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

hanfenglong315, 2024-09-06:
把vmware的复制物理地址 打勾就好了。同时得在虚拟网卡那加入物理机的网卡

akumajac, 2024-09-18:
我也遇到了，不过我是畅网n100实体机装的，有没有解决办法> 把vmware的复制物理地址 打勾就好了。同时得在虚拟网卡那加入物理机的网卡

wjz304, 2024-10-11:
与 RR/DSM 无关

closed 2024-10-11 by wjz304