RROrg/rr #2769: BUG,RTL8111在RR中不能获取DHCP

q331723516, 2024-09-03:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  华擎z370 itx
 * CPU:  i5 8500
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.8.6
 * addons:  默认
 * modules:  默认
 * lkms:  默认

DSM: 
 * model:  DS920+
 * version:  7.2.2

Issue
rtl8111g，在7.3之前的版本，rr中可以获取到ip，之后的版本总是不能获取dhcp，偶尔会获取到但是丢包率很高，很不稳定
rr设置及主板bios都是默认，未作任何更改
logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

Brian099, 2024-09-06:
主板名称	MSI MAG B550M Mortar (MS-7C94)
处理器名称	OctalCore AMD Ryzen 7 5700G, 4572 MHz (45.75 x 100)
网络适配器	Realtek PCIe 2.5GbE Family Controller
DSM 918+
version 7.2.1 69057

q331723516, 2024-09-07:
> in ssh of DSM:
> 
> ```
> lsmod | grep r81
> dmesg | grep r8168 -A 20
> ifconfig
> ls /sys/class/net/*
> ```
dsm：
[image]
[image]
RR:
[image]
[image]

q331723516, 2024-09-07:
我还尝试了intel的pcie网卡；发现两种情况；
1，rtl8111和板载网卡（intel）是rtl8111不能dhcp
2，intel网卡和板载网卡（intel）是板载网卡 不能dhcp

wjz304, 2024-09-08:
ethtool -k eth0

wjz304, 2024-09-08:
[r8168.zip](https://github.com/user-attachments/files/16920770/r8168.zip)
test this

q331723516, 2024-09-08:
> ethtool -k eth0

[image]

> [r8168.zip](https://github.com/user-attachments/files/16920770/r8168.zip) test this

这个如何使用是这里吗
[image]

q331723516, 2024-09-08:
我看好多人物理机器，都有在7.1版本之后都有丢失ip的情况或双网卡只有一个的情况，还有不改直接启动无法引导的情况，如需物理机测试，我都有时间

wjz304, 2024-09-08:
> 这个如何使用是这里吗

是的，上传后确认被选中，重新编一下，启动即可

q331723516, 2024-09-08:
> > 这个如何使用是这里吗
> 
> 
> 
> 是的，上传后确认被选中，重新编一下，启动即可

已经测试过，问题还是存在

wjz304, 2024-09-08:
> > > 这个如何使用是这里吗
> > 
> > 
> > 是的，上传后确认被选中，重新编一下，启动即可
> 
> 已经测试过，问题还是存在

ethtool -k eth0

q331723516, 2024-09-08:
> > > > 这个如何使用是这里吗
> > > 
> > > 
> > > 是的，上传后确认被选中，重新编一下，启动即可
> > 
> > 
> > 已经测试过，问题还是存在
> 
> ethtool -k eth0

[image]

wjz304, 2024-09-08:
额，我才注意到你上面的也是在 RR 下...
哪里有问题 在哪里取啊

wjz304, 2024-09-08:
```
dmesg | grep r8168 -A 20
ethtool -k eth0
ifconfig
```

q331723516, 2024-09-08:
> ethtool -k eth0

[image]

[image]
[image]
以上jie截图是在DSM；
DSM和RR两个系统都不能获取到IP

q331723516, 2024-09-08:
这个是RR下：
[image]
[image]
[image]

wjz304, 2024-09-08:
对换下两个网卡的网线看看

q331723516, 2024-09-08:
对换网线之后eth0可以获取ip，但是不能通讯，eth1变成了不能dhcp；

q331723516, 2024-09-08:
兑换网线之后我没有重启，我刚刚重启了一下，还是eth0不能dhcp，eth1正常，
[image]

wjz304, 2024-09-08:
dmesg | grep -i dhcp

q331723516, 2024-09-08:
> dmesg | grep -i dhcp

[image]
好像没什么反应

wjz304, 2024-09-08:
cat /tmp/messages | grep -i dhcp

q331723516, 2024-09-08:
> cat /tmp/messages | grep -i dhcp
RR系统详细界面上还是显示未检测到dhcp，但好像是可以拿到ip的，进入dsm网卡也正常工作了，
RR下重启前：
[image]
RR下重启后：
[image]

q331723516, 2024-09-08:
这是我更换了intel  82574L的pcie网卡，一样还是出现eth0，在dsm和rr里都不能获取dhcp的情况，但是这次板载网卡变成eth0，pcie网卡是eth1，下图为RR下的截图
[image]
[image]
[image]
[image]

q331723516, 2024-09-08:
> cat /tmp/messages | grep -i dhcp
RR下：
[image]
反复重启测试多次，RR下，偶尔是可以获取到dhcp，无论RR下是否获取到dhcp，改用直接启动才能正常引导进入DSM，网卡也是可以正常工作的，不知道这种情况正常吗
我也测试了24.6.7版本，无以上问题，均正常

yanghanbin, 2024-09-09:
j4125,rtl8125网卡，更新到24.8.6后，能启动到配置界面，设置完毕后启动，dsm就失联了，下次启动，还是进入配置界面，之前试过降级到未更新网卡驱动的版本是好的，物理机安装的，现场还在，因为暂时没用，就挂哪儿了

laomao9000, 2024-09-09:
测试过各个版本的驱动，都无法联机，感觉和驱动无关，和系统的设置有关，这个问题发生在24.7.2的更新。在24.7.2以后都不好用了，手工换上以前版本的驱动模块也不行。
请参考！

ZMDLBX, 2024-09-09:
不仅仅是8168网卡，intel网卡也一样，我记得24.5.1之前都正常的，后来就出现到DSM后失联，其中有一段时间的版本，当发现DSM失联后，可以采取两种方法：一是可以拔掉网线，等几秒再插上，就出现IP了，二是直接改引导为直接启动。现在24.8.0-24.8.6版本，拔网线的方法不管用了，只能在高级设置里改直接启动了。 这个DSM失联问题，我感觉与网卡驱动无关的，与引导有关，就是有一次对引导做了大的改动，什么架构做了大的调整那次。从那以后就频发DSM失联问题了。

Brian099, 2024-09-09:
24.5.1 和 24.7.1 分别更新了一次网卡驱动...

yanghanbin, 2024-09-09:
> 不仅仅是8168网卡，intel网卡也一样，我记得24.5.1之前都正常的，后来就出现到DSM后失联，其中有一段时间的版本，当发现DSM失联后，可以采取两种方法：一是可以拔掉网线，等几秒再插上，就出现IP了，二是直接改引导为直接启动。现在24.8.0-24.8.6版本，拔网线的方法不管用了，只能在高级设置里改直接启动了。 这个DSM失联问题，我感觉与网卡驱动无关的，与引导有关，就是有一次对引导做了大的改动，什么架构做了大的调整那次。从那以后就频发DSM失联问题了。

晚上回去还是降级得了，反正近期也没啥特别的更新，或者换换飞牛试试，

Brian099, 2024-09-09:
[image]

q331723516, 2024-09-09:
没办法，等大佬有时间想办法修复吧

q331723516, 2024-09-09:
今天我朋友用一个十几年前的联想主板，奔腾cpu，不改直接启动，一样可以正常启动

wjz304, 2024-09-15:
test 24.9.0

q331723516, 2024-09-16:
> test 24.9.0

这个版本8168模块正常，不过依然要改直接启动模式，👍

laomao9000, 2024-09-21:
测试24.9.1，问题依旧——8168无法使用，必须改成直接启动。请大佬恢复24.7.1之前的模式吧。

cdzyp520, 2024-09-28:
我的也是8168安装的时候就出现过这个问题，然后找了一堆办法总算安装上了，然后今天关机后再开就再也连不上了，路由里地址都不出现，折腾了好久，看到这个换成24.51就可以了，然后重启看了下24.51是加载全部模块的驱动，又把24.91的重新插入把模块的驱动 全部选择上，然后就可以了，到底是因为哪个模块驱动没人选择导致启动后不能获取IP搞不懂懒的一个个试了，就这样用吧，等大佬们找到问题修复吧。。。。

wangyz1997, 2024-10-23:
我的RTL8111H使用最新的RR 24.10.1，能获取IP，但是会不停的连接断开，最后握手成功也是百兆网卡。已经按照网上的教程取消掉了r8125的驱动，还是无法解决问题。

```
[ 2791.669097] r8168: eth0: link up
[ 2792.680987] r8168: eth0: link down
```

closed 2024-11-22 by wjz304