RROrg/rr #3104: BUG

Gurabity, 2024-09-19:
kernel: official
rd-compressed: "false"
satadom: "2"
lkm: prod
dsmlogo: "true"
directboot: "false"
prerelease: "false"
bootwait: "10"
bootipwait: "10"
kernelway: power
kernelpanic: "5"
odp: "false"
hddsort: "false"
emmcboot: "false"
platform: "apollolake"
model: "DS918+"
modelid: "DS918+"
productver: "7.2"
buildnum: "72806"
smallnum: "0"
paturl: "https://global.synologydownload.com/download/DSM/release/7.2.2/72806/DSM_DS918%2B_72806.pat"
patsum: "e78a003a9679790c19fc11974d2d0453"
sn: ""
mac1: ""
mac2: ""
layout: qwerty
keymap: ""
zimage-hash: "04b91d5667efa28398a631ef5a6c6ca6a79ed5c00ed0826c8aa4f357ab87c201"
ramdisk-hash: "81a608d50f695271ec88101f10558d39be0e647184aec950ea8a5c3e08367f1f"
cmdline: {}
synoinfo:
  supportext4: yes
  support_uasp: yes
  support_printer: yes
  support_usb_printer: yes
  support_disk_compatibility: no
  support_memory_compatibility: no
  support_led_brightness_adjustment: no
  support_leds_atmega1608: no
  support_leds_lp3943: no
  support_oob_ctl: no
  support_syno_hybrid_raid: no
  supportraidgroup: no
  enableRCPower: yes
  support_fan: no
  supportadt7490: no
  maxlanport: "8"
  netif_seq: 0 1 2 3 4 5 6 7
  buzzeroffen: "0xffff"
  HddEnableDynamicPower: no
addons:
  acpid: ""
  mountloader: ""
  powersched: ""
  reboottoloader: ""
  addincards: ""
  codecpatch: ""
  console: ""
  cpuinfo: ""
  dbgutils: ""
  expands: ""
  hdddb: ""
  nvmevolume: ""
  photosfacepatch: ""
  remotefs: ""
  setrootpw: ""
  storagepanel: ""
  synoconfbkp: ""
  updatenotify: ""
  sortnetif: ""
modules:
  8021q: ""
  coretemp: ""
  e1000e: ""
  ipv6: ""
  irqbypass: ""
  kvm: ""
  kvm_intel: ""
  r8168: ""
  raid1: ""
  x86_pkg_temp_thermal: ""
  xhci_hcd: ""
  xhci_pci: ""
  xhci_pci_renesas: ""
modblacklist: evbug,cdc_ether
vid: "0x0781"
pid: "0x557d"

Issue:  
I am using r8168 and e1000e LAN cards.
When building the loader, I selected only the relevant LAN card, selected all modules, and selected only the currently existing ones to be automatically selected.
I also tried changing the USB, downloading a new image, and rebuilding.
Since I have two LAN cards, I'll try them in turns.
My problem is that the boot completes normally and even the IP address is assigned.
Also, the IP appears normally on my Asus router.
However, I cannot find it on the find synology site, and I cannot connect to the IP shown on the boot screen.
And the ping dies within about 5 minutes.
Due to this problem, we are currently using the M-SHELL loader.
Both LAN cards can be used normally with the M-SHELL loader.
In the case of rrloader, it was possible to use it in version 6, but currently it cannot be used at all.
Is there any solution?

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

Gurabity, 2024-09-20:
We enabled direct boot by referring to the document below and confirmed that it operates normally when the function is enabled....
If you disable direct boot, the same symptom persists...
Can I activate and use this feature?
The default is disabled. What is the difference?

wjz304, 2024-10-17:
https://rrorg.cn/archives/issues
[image]

closed 2024-10-17 by wjz304