RROrg/rr #3219: 内存占用不匹配

qiezihwy, 2024-09-25:
[image]

[image]

[image]

在使用pve部署的时候，遇到一个问题，在pve的监控界面和dsm上看到的内存占用不一致，大家有遇到这个情况么

qiezihwy, 2024-09-25:
[image]

硬件信息如上

awalol, 2024-10-01:
这是因为缺少 virtio-balloon 驱动的原因
目前这个驱动好像只在SA6400的custom内核有编译

而开启balloon后系统也不会立即释放内存，需要发送 drop_caches 释放缓存

qiezihwy, 2024-10-06:
> 这是因为缺少 virtio-balloon 驱动的原因 目前这个驱动好像只在SA6400的custom内核有编译
> 
> 而开启balloon后系统也不会立即释放内存，需要发送 drop_caches 释放缓存

理解了，谢谢大佬