RROrg/rr #3245: BUG 机械硬盘全消失没有了Marvell 88SE9230

siulam, 2024-09-27:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  AMD 
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  v24.9.1
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS918+
 * version:  7.2.2-72806

Issue: 升级系统为7.2.2-72806后，机械硬盘全消失没有了，机械硬盘是连接Marvell 88SE9230  SATA Raid Controller，但发现没有该模块驱动，请问怎么解决呢？

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

robincao2025, 2024-11-07:
我安装完成后，拔掉U盘，重新启动，会出现invalid partition table,但是插上U盘，可以正常启动，这个是什么原因？

wjz304, 2024-11-07:
> 我安装完成后，拔掉U盘，重新启动，会出现invalid partition table,但是插上U盘，可以正常启动，这个是什么原因？

因为 拔掉了U盘

closed 2025-05-08 by wjz304