RROrg/rr #3623: BUG - can boot into DSM on RR 24.8.1 or higher

gogixai7, 2024-10-02:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Default string... BIOS ALN4l101 09/27/3023
 * CPU:  Intel N100
 * NIC: Intel I226  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.8.1 <-> 24.10.0
 * addons:  default for DS920 + nvmesystem
 * modules:  default for DS920
 * lkms:  prod 

DSM: 
 * model:  DS920+ or similar geminilake models
 * version:  7.2(72806)

Issue:  
I have MiniPC from CWWK with Intel N100.
Decided to switch from SA6400 to DS920+ because of problems with Windows virtual machines in VMM.
Updated RR to newest version and DSM is not loaded.
After some testing found that DSM is loaded with RR24.8.0, but newest version are not working.
At the same time SA6400 and FS6400 works w/o issues with newest RR version.

I tried to get logs using , but it seems that they are not being created

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

TAN-Z-H, 2024-10-12:
同样，我几乎测试了28.x的所有版本与29.x的前两个版本，也是无法引导，但27.x均可。另外，与你不同的是，我的是i3 8100T

wjz304, 2024-10-18:
test v24.10.1

gogixai7, 2024-10-18:
tried 24.10.1, no changes

wjz304, 2024-10-24:
https://github.com/RROrg/rr/issues/1893 跟踪

closed 2024-10-24 by wjz304