RROrg/rr #3690: UPS Addon shutdown task

noobs69, 2024-10-03:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  MAXSUN W680
 * CPU:  12500T
 * NIC:  (pid & vid) X550

RR version: (You can find it in the update menu.)  
 * RR:  newest
 * addons:  UPS addon
 * modules:  
 * lkms:  

DSM: 
 * model:  7.2.2
 * version:  

Issue:  
Hi Master of RR , i would like to request a new features for the UPS addon shutdown schedule task ? , the shutdown task will skipped because of there is ISCSI target connected , is there a way to forced it to disconnect ISCSI client before running the shutdown task.
[image]

 just closed the topic if you're busy or there is no way to fix it . I appreciate your time and attention

wjz304, 2024-10-06:
添加一个关机的任务计划可以吧

noobs69, 2024-10-06:
that's would be great sir .

wjz304, 2024-10-17:
请做一个测试，
在执行 `synopkg stop ScsiTarget` 后模拟断电的操作，常看是否会正常休眠

noobs69, 2024-10-18:
work flawlessly sir, there is 2 output after few tried. 
synopkg stop ScsiTarget
{"action":"stop","beta":false,"error":{"code":0},"finished":true,"language":"enu","last_stage":"stopped","package":"ScsiTarget","pid":16230,"scripts":[{"code":0,"message":"","type":"prestop"},{"code":0,"message":"","type":"stop"}],"stage":"stopped","status":"stop","status_code":324,"status_description":"translate from systemd status","success":true,"username":"","version":"1.0.10-0314"

synopkg stop ScsiTarget
{"action":"prepare","error":{"code":0},"stage":"prepare","success":true}

wjz304, 2024-10-18:
模拟断电的操作，正常关机了？

noobs69, 2024-10-18:
yes it does shutdown normally. very quick . San manager need to re-enabled but it's fine .

wjz304, 2024-10-18:
OK, thanks for the test, I will try to make it into a addon

noobs69, 2024-10-18:
OMG thanks you so much .

wjz304, 2024-10-18:
[image]

You can try adding the following two tasks
start san :  synopkg start ScsiTarget
stop san :  synopkg stop ScsiTarget

Check if they works?

noobs69, 2024-10-18:
the start san task working but . the Stop SAN in doesnt triggered .
[image]

wjz304, 2024-11-04:
[ups.zip](https://github.com/user-attachments/files/17614224/ups.zip)

[image]
1. update ups.addon
2. add ups with parameter "-f"
3. build and boot
4. test

wjz304, 2024-11-05:
https://github.com/RROrg/rr/releases/tag/24.11.0
test

noobs69, 2024-11-05:
[image]
Wow, you're an absolute legend! You completely aced it, and I can't thank you enough. Your expertise really shines through!

closed 2024-11-05 by noobs69

wjz304, 2024-11-06:
ups addon 有一个bug，等待24.11.1 更新的时候 更新一下