RROrg/rr #3752: BUG - 无法启动dsm，在RR 24.6.4或更高版本

ckx000, 2024-10-08:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  To Be Filled By O.E.M. To Be Filled By O.E.M./H310CM-ITX/ac, BIOS P4.30 08/16/2021
 * CPU:  Intel(R) Core(TM) i5-8500T CPU @ 2.10GHz
 * NIC: 
`root@rr:/opt/rr# lspci -nn`
`00:00.0 Host bridge [0600]: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers [8086:3ec2] (rev 07)`
`00:02.0 VGA compatible controller [0300]: Intel Corporation CoffeeLake-S GT2 [UHD Graphics 630] [8086:3e92]`
`00:14.0 USB controller [0c03]: Intel Corporation 200 Series/Z370 Chipset Family USB 3.0 xHCI Controller [8086:a2af]`
`00:14.2 Signal processing controller [1180]: Intel Corporation 200 Series PCH Thermal Subsystem [8086:a2b1]`
`00:16.0 Communication controller [0780]: Intel Corporation 200 Series PCH CSME HECI #1 [8086:a2ba]`
`00:17.0 SATA controller [0106]: Intel Corporation 200 Series PCH SATA controller [AHCI mode] [8086:a282]`
`00:1d.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Root Port #11 [8086:a29a] (rev f0)`
`00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a2ca]`
`00:1f.2 Memory controller [0580]: Intel Corporation 200 Series/Z370 Chipset Family Power Management Controller [8086:a2a1]`
`00:1f.3 Audio device [0403]: Intel Corporation 200 Series PCH HD Audio [8086:a2f0]`
`00:1f.4 SMBus [0c05]: Intel Corporation 200 Series/Z370 Chipset Family SMBus Controller [8086:a2a3]`
`00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (2) I219-V [8086:15b8]`
`01:00.0 Network controller [0280]: Intel Corporation Dual Band Wireless-AC 3168NGW [Stone Peak] [8086:24fb] (rev 10)`

 Cmdline:   只增加了skip_vender_mac_interfaces参数 和自定mac
`withefi earlyprintk syno_hw_version=DS918+ console=ttyS0,115200n8 consoleblank=600 netif_num=2 pid=0x88aa earlycon=uart8250,io,0x3f8,115200n8 mac2= mac1= sn= SMBusHddDynamicPower=1 vid=0x88aa elevator=elevator loglevel=15 HddHotplug=1 skip_vender_mac_interfaces intel_iommu=igfx_off syno_hdd_detect=0 panic=5 vender_format_version=2 syno_hdd_powerup_seq=0 log_buf_len=32M root=/dev/md0`

synoinfo：default for DS918（7.2.1）也是默认的 没动

RR version: (You can find it in the update menu.)  
 * RR:  24.6.4
 * addons:  default for DS918（7.2.1）+  dbgutils:（均为默认插件 就添加了日志插件（然而出问题的时候啥都没记录到
 * modules: default for DS918（7.2.1） 全是默认的 没动
 * lkms:  prod
其他高级设置：均为默认 未做改动，与能正常启动的24.6.3大体相同

DSM: 
 * model:  DS918
 * version:  7.2.1

Issue:  
现在的状况是在24.6.3 采用默认设置能启动dsm  但是换24.6.4以上均无法启动，显示画面会卡在rr提示画面不可操作的画面。10分钟以上的时间都无法找到dsm的ip（观察了路由器均为发现dsm的dhcp信息）在24.6.3及以下版本时，在RR画面冻结后，1分钟内就能ping同dsm的ip。。
最一开始我是用24.4.4升级到7.0.1没问题，准备到7.2.1 也没问题，能出dsm的提示更新页面。（之所以用24.4.4是看到它是最后支持6.2的版本，因为我是6.23升上来的，本来想用24.4.4升级623到624，结果怎么都不成功，反复安装重启都还提示要安装dsm，最后升级到7.0.1成功）
然后准备换用最新版的rr24.10.0，发现出状况了。就一步步排查到发现是24.6.4开始就无法启动了。

请问24.6.4相比之前具体改动了哪些 我如何做排查？**一开始以为是那个pci stubs的问题 但是我尝试了回滚pci stubs的24.6.5也无法启动dsm**。想提取日志也提取不了。
我是物理机，加载的dbgutils也获取不到启动不了dsm的日志，只能获得正常启动dsm的24.6.3的日志。

logs:  
因无法抓到24.6.4无法启动的log（RR画面冻结那里等了20多分钟，再重启后RR也提示找不到日志），只能先贴上24.6.3正常启动dsm后 使用http://:5000/webman/get_logs.cgi获取到的日志
[24.6.3.log](https://github.com/user-attachments/files/17295705/24.6.3.log)

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-10-10:
直接启动：true

closed 2024-10-10 by wjz304

ckx000, 2024-10-14:
确认直接启动能进dsm了，，不过突然有点不适应这样的启动。。我最后还是选择留在24.6.3了。。就只是把插件模块升级到24.10.0