RROrg/rr #3754: 无法关机，关机后会自动重启，用计划任务关机后依然会自动重启

mocl1220, 2024-10-09:
蜗牛星际b款，ds918+的固件，无法通过网页端关机，关机后2秒自动重启，计划任务关机后依然会自动重启，通过ssh登录用poweroff命令依然如此，用U盘引导进入pe然后关机可以正常关机，因此应该是引导的问题，且群晖控制面板中的“修复电源问题后自动重启”选项无法取消，即使取消勾选重启后又被勾上了。
[image]

ruols, 2024-10-09:
同款蜗牛B，我也遇到这个问题，不过我之前一直是正常的，但只要插上UPS的USB通讯线，就会出现这个问题，拔了就一切正常正常关机
昨天试了各种BIOS选项都搞不定，升级了最新引导也不行，只能先把UPS的通讯线拔了

ruols, 2024-10-09:
已解决，折腾了两天各种刷系统改bios设置，刚刚无意把UPS的USB线插到蜗牛最下面的USB口解决了😅，之前一直插的上面USB3和旁边USB2的都会自动重启，实在有点莫名其妙，搞不懂为什么。
楼主的问题我不会遇到，无论用不用UPS保存设置都正常，楼主是不是装了rr 强制wol的插件

ilongli, 2024-10-17:
同款问题，这个问题应该还是看硬件，我的是精粤的H97I-GAMING，RR: 24.10.0。
实际上，在RR里面关机也是一样的，过几秒后又会上电开机。
找了一下资料，大概是主板USB设置和内核冲突的问题。

**解决方案：**
1. 进BIOS，找到：USB CONFIGURATION -> XHCI MODE -> DISABLE，不一定叫这个，反正就是把XHCI的相关设置关掉。关掉这个之后USB 3.0好像会失效，没测过，我用不上。
2. 还有一个方法是改内核参数，编辑grub.cfg，找到set RR_CMDLINE那行，加上：xhci_hcd.quirks=270336。这样RR就能正常关机了，但DSM还是不能正常关机，我在“设置Cmdline”加上这个参数还是不行，不知道是不是操作有误。最终这个方法我没成功。
[image]

**参考：**
1. https://bbs.archlinux.org/viewtopic.php?id=244687
2. https://wiki.archlinux.org/title/Wake-on-LAN#Fix_by_Kernel_quirks （看5.2）

wjz304, 2024-10-18:
> 同款问题，这个问题应该还是看硬件，我的是精粤的H97I-GAMING，RR: 24.10.0。 实际上，在RR里面关机也是一样的，过几秒后又会上电开机。 找了一下资料，大概是主板USB设置和内核冲突的问题。
>
> **解决方案：**
>
> 1. 进BIOS，找到：USB CONFIGURATION -> XHCI MODE -> [...]

尝试 更换 主板电池 复位bios

mocl1220, 2024-10-18:
> 已解决，折腾了两天各种刷系统改bios设置，刚刚无意把UPS的USB线插到蜗牛最下面的USB口解决了😅，之前一直插的上面USB3和旁边USB2的都会自动重启，实在有点莫名其妙，搞不懂为什么。 楼主的问题我不会遇到，无论用不用UPS保存设置都正常，楼主是不是装了rr 强制wol的插件

## 感谢解答！！尝试了以上方法，但是问题依旧，没有安装wol插件

wulalaxing, 2024-10-29:
我也出现上述问题了，楼上的解决办法尝试了，一样无法解决。

wjz304, 2024-12-12:
bios问题，不再跟进..