RROrg/rr #3760: BUG SA6400 TANK定制版 VMM启动虚拟机一直显示准备中

pumpkin-nbc, 2024-10-10:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU: i5-8600K
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.9.1
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  7.2.2
 * version:  

Issue:  
TANK定制版的24.9.1引导 全部数据清空重装之后，启动win11虚拟机 显示一直在准备中 
logs:  

日志找不到 麻烦提示一下

wjz304, 2024-10-29:
https://github.com/RROrg/rr/releases/tag/24.10.3

pumpkin-nbc, 2024-10-29:
> https://github.com/RROrg/rr/releases/tag/24.10.3

ok 店家那边也反馈过这个问题了

Liuruicheng520, 2025-01-04:
> > https://github.com/RROrg/rr/releases/tag/24.10.3
> 
> ok 店家那边也反馈过这个问题了

你最后解决这个虚拟的问题了吗？

pumpkin-nbc, 2025-01-04:
> > > https://github.com/RROrg/rr/releases/tag/24.10.3
> > 
> > 
> > ok 店家那边也反馈过这个问题了
> 
> 你最后解决这个虚拟的问题了吗？

用了切换了老版本并且用了ds918 老版本的sa6400也是失败的

Liuruicheng520, 2025-01-04:
> > > > https://github.com/RROrg/rr/releases/tag/24.10.3
> > > 
> > > 
> > > ok 店家那边也反馈过这个问题了
> > 
> > 
> > 你最后解决这个虚拟的问题了吗？
> 
> 用了切换了老版本并且用了ds918 老版本的sa6400也是失败的

你现在用的是降级的 918？  具体是什么型号 什么版本引导 我目前i9 9900t 也是虚拟机 一直准备中 能否截图 告知一下 或者输入 告诉我也行~~~这半个月不停开机 关机 折腾的要命

pumpkin-nbc, 2025-01-04:
> > > > > https://github.com/RROrg/rr/releases/tag/24.10.3
> > > > 
> > > > 
> > > > ok 店家那边也反馈过这个问题了
> > > 
> > > 
> > > 你最后解决这个虚拟的问题了吗？
> > 
> > 
> > 用了切换了老版本并且用了ds918 老版本的sa6400也是失败的
> 
> 你现在用的是降级的 918？ 具体是什么型号 什么版本引导 我目前i9 9900t 也是虚拟机 一直准备中 能否截图 告知一下 或者输入 告诉我也行~~~这半个月不停开机 关机 折腾的要命

我现在用的tank的24.9.1 ds918
你试试仓库里最新的 25.1.0 然后选sa6400能不能行

Liuruicheng520, 2025-01-04:
从25.1.0到上面三个版本都测试过  虚拟机正在启动    tank后面 有没有别人提出过这个问题  ？

      
https://github.com/RROrg/rr/releases/tag/24.10.3

Liuruicheng520, 2025-01-04:
> > > > > > https://github.com/RROrg/rr/releases/tag/24.10.3
> > > > > 
> > > > > 
> > > > > ok 店家那边也反馈过这个问题了
> > > > 
> > > > 
> > > > 你最后解决这个虚拟的问题了吗？
> > > 
> > > 
> > > 用了切换了老版本并且用了ds918 老版本的sa6400也是失败的
> > 
> > 
> > 你现在用的是降级的 918？ 具体是什么型号 什么版本引导 我目前i9 9900t 也是虚拟机 一直准备中 能否截图 告知一下 或者输入 告诉我也行~~~这半个月不停开机 关机 折腾的要命
> 
> 我现在用的tank的24.9.1 ds918 你试试仓库里最新的 25.1.0 然后选sa6400能不能行

从25.1.0-24.x.x到上面三个版本都测试过  每次都是 虚拟机正在启动    tank后面 有没有别人提出过这个问题  ？

pumpkin-nbc, 2025-01-04:
> 从25.1.0-24.x.x到上面三个版本都测试过 每次都是 虚拟机正在启动 tank后面 有没有别人提出过这个问题 ？

暂时没有找到过类似的 你看上面写的24.10.3这个版本 也不可以吗

wjz304, 2025-01-08:
> > > > > > > > https://github.com/RROrg/rr/releases/tag/24.10.3
> > > > > > > 
> > > > > > > 
> > > > > > > ok 店家那边也反馈过这个问题了
> > > > > > 
> > > > > > 
> > > > > > 你最后解决这个虚拟的问题了吗？
> > > > > 
> > > > > 
> > > > > 用了切换了老版本并且用了ds918 老版本的sa6400也是失败的
> > > > 
> > > > 
> > > > 你现在用的是降级的 918？ 具体是什么型号 什么版本引导 我目前i9 9900t 也是虚拟机 一直准备中 能否截图 告知一下 或者输入 告诉我也行~~~这半个月不停开机 关机 折腾的要命
> > > 
> > > 
> > > 我现在用的tank的24.9.1 ds918 你试试仓库里最新的 25.1.0 然后选sa6400能不能行
> > 
> > 
> > 从25.1.0-24.x.x到上面三个版本都测试过 每次都是 虚拟机正在启动 tank后面 有没有别人提出过这个问题 ？
> 
> 暂时没有找到过类似的 你看上面写的24.10.3这个版本 也不可以吗
> 
> @wjz304 需要提供一些log给你吗

没有复现的环境，log也不知道看哪， 
```
lsmod | grep kvm
grep kvm /var/log/messages
```

pumpkin-nbc, 2025-01-08:
> 没有复现的环境，log也不知道看哪，
>
> ```
> lsmod | grep kvm
> grep kvm /var/log/messages
> ```

@Liuruicheng520  兄弟 方便你的log放一下吗 我这边得年底回去再试验了

pumpkin-nbc, 2025-01-29:
[image]
@wjz304 
这个是 tank RR-24.9.1 SA6400 i58600k 32G 显卡P4

wjz304, 2025-02-17:
Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./Z370M Pro4, BIOS P4.20G 04/01/2021

pumpkin-nbc, 2025-02-17:
> Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./Z370M Pro4, BIOS P4.20G 04/01/2021

华擎Z370M Pro4
现在sa6400引导切换到official都会正常

pumpkin-nbc, 2025-02-18:
> > > Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./Z370M Pro4, BIOS P4.20G 04/01/2021
> > 
> > 
> > 华擎Z370M Pro4 现在sa6400引导切换到official都会正常
> 
> 另一个issue 上那俩人聊的热火朝天的，啥有用信息都没有，也不搭理我，你上述日志也没有 kvm 的关键字

你的那命令 在启动的时候没有任何日志 所以我就贴了系统日志
我下次贴一下系统的error日志或者贴一下虚拟机服务的日志

closed 2025-09-07 by wjz304