RROrg/rr #3772: BUG

lichenxi1995, 2024-10-13:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  J3455
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.10.0
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS918+
 * version:  7.2.2

Issue:  编译结束后，选择直接启动，紧接着重启进入直接启动没有问题，进入dsm后，点击重启，或者关机后手动重启，就只能进入boot dsm选项，进入rr的系统了，也不会自动进行第二次引导，这时只能web或者ssh进rr用menu.sh再点击启动才能再重启进入直接引导dsm系统

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-10-17:
无法复现

wjz304, 2024-10-24:
无进一步反馈，该 issue关闭

closed 2024-10-24 by wjz304