RROrg/rr #4150: 静态IP设置BUG

Fyred, 2024-10-22:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  Winserver2025 Hyper-V虚拟化黑群
 * CPU:  i9700
 * NIC:  intel i211

RR version: (You can find it in the update menu.)  
 * RR:  24.10.1
 * addons:  default
 * modules:  default
 * lkms:  default

DSM: 
 * model:  SA6400 ( custom )
 * version:  7.2.2

Issue:  
1、内网环境有DHCP服务器，虚拟平台搭建完成，首次进入RR能正常获取到ip地址，此时设置好DHCP以外的静态IP地址子网掩码等，可见顶部黄色ip地址随之改变，编译完成后重启，用刚设置的静态ip，可进入DSM安装程序，安装完成后等10分钟也无反应，此时发现静态ip地址已无法ping通，反而此前自动获取的ip地址重新生效，通过此自动获取的ip才可完成之后的安装程序。过程中如果重新进入RR界面，会发现顶部黄色地址已变为自动获取的ip，但静态ip设置界面中仍有之前设置的ip信息，需再次确定后生效。

2、如果通过上述动态ip完成DSM安装，进入DSM后通过官方界面设置成静态ip，重启后会出现静态ip和动态ip均无法访问DSM的情况，只能到RR中重置网络。

logs:  
无，抱歉

ZMDLBX, 2024-10-23:
应该是没有BUG的，是你的思路和操作步骤不对。
如果你想在DHCP路由环境中使用静态IP地址，应该是在路由器中绑定MAC地址，让DHCP给你的网卡分配一个你想要的固定IP，然后继续后面的操作。
如果你不想DHCP了，就关闭路由器的DHCP。
RR引导--配置静态IP--重启--解析PAT--本地上传pat--编译--引导......（本人试过的，一路顺丰的）

wjz304, 2024-10-23:
> 应该是没有BUG的，是你的思路和操作步骤不对。 如果你想在DHCP路由环境中使用静态IP地址，应该是在路由器中绑定MAC地址，让DHCP给你的网卡分配一个你想要的固定IP，然后继续后面的操作。 如果你不想DHCP了，就关闭路由器的DHCP。 RR引导--配置静态IP--重启--解析PAT--本地上传pat--编译--引导......（本人试过的，一路顺丰的）

你和他说的不是一回事

wjz304, 2024-10-23:
> "可进入DSM安装程序，安装完成后等10分钟也无反应，此时发现静态ip地址已无法ping通，反而此前自动获取的ip地址重新生效，"

首次安装 静态ip设置不会生效，再次重启即可。

> "过程中如果重新进入RR界面，会发现顶部黄色地址已变为自动获取的ip，但静态ip设置界面中仍有之前设置的ip信息，需再次确定后生效。"

过程中如果重新进入RR界面，退出 menu 再次进入即可， 网卡设置延时问题。

> 2、如果通过上述动态ip完成DSM安装，进入DSM后通过官方界面设置成静态ip，重启后会出现静态ip和动态ip均无法访问DSM的情况，只能到RR中重置网络。

该问题无法解决。如果在DSM中设置静态IP，请在RR中删除相关设置。

Fyred, 2024-10-24:
> 应该是没有BUG的，是你的思路和操作步骤不对。 如果你想在DHCP路由环境中使用静态IP地址，应该是在路由器中绑定MAC地址，让DHCP给你的网卡分配一个你想要的固定IP，然后继续后面的操作。 如果你不想DHCP了，就关闭路由器的DHCP。 RR引导--配置静态IP--重启--解析PAT--本地上传pat--编译--引导......（本人试过的，一路顺丰的）

你这其实是一种变通的方式，而后来我确实是用这种方式实现了“伪静态”的IP，简单举个例子，早期一个C的局域网，都会设置dhcp地址池为192.168.0.10-192.168.0.254，而192.168.0.10之前的地址，就是保留用来设置静态ip的，且并不需要到路由上去分配静态地址的操作，多此一举

Fyred, 2024-10-24:
> 首次安装 静态ip设置不会生效，再次重启即可。
>
>
> 过程中如果重新进入RR界面，退出 menu 再次进入即可， 网卡设置延时问题。
>
>
> 该问题无法解决。如果在DSM中设置静态IP，请在RR中删除相关设置。

我是在路由器上给DSM分配了静态DHCP地址，同时在DSM里手动指定了同一ip地址，这样RR自动获取的总会和DSM里手动设置的地址一致。
其实我想说，RR里既然设置的静态ip没什么用，那留着这个选项，非常误导人，我甚至认为，既然能在RR里重置DSM里的网络设置，那RR里的静态地址设置，为何不能自动生效到DSM中呢？这样才更符合RR作为DSM引导程序的地位

wjz304, 2024-10-24:
> 我是在路由器上给DSM分配了静态DHCP地址，同时在DSM里手动指定了同一ip地址，这样RR自动获取的总会和DSM里手动设置的地址一致。 其实我想说，RR里既然设置的静态ip没什么用，那留着这个选项，非常误导人，我甚至认为，既然能在RR里重置DSM里的网络设置，那RR里的静态地址设置，为何不能自动生效到DSM中呢？这样才更符合RR作为DSM引导程序的地位

事实就是能生效，RR的设置会设置到 DSM，DSM 的设置 RR不能够获取到（这很麻烦），每次开机时 引导都会应用这个设置。
只是首次安装时，系统会初始化一次（目前不知道如何屏蔽），第二次（初始化以后的系统）重启后 是RR设置的。

wjz304, 2024-10-24:
关于意义，也只是解决部分人没有DHCP服务器的情况，
我从来不建议在下游设置静态ip，静态ip 只建议在 上游（路由测）设置，不只是DSM，任何系统都是。