RROrg/rr #4212: sata_dev_port_check.c:16 Failed to get slot informtion of sata1

Fyred, 2024-10-26:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  华擎z390m-itx/ac&Hyper-v&Winserver2025
 * CPU:  i7-9700
 * NIC:  i211-AT

RR version: (You can find it in the update menu.)  
 * RR:  24.10.2
 * addons:  default
 * modules:  default
 * lkms:  default

DSM: 
 * model:  SA6400(custom)
 * version:  7.2.2

Issue:  
Hyper-v虚拟DSM7.2.2，直通jmb585 SATA 5口扩展卡，接了4块硬盘使用一切正常，但是在DSM日志（messages和scemd.log）中每30s频繁记录3条"sata_dev_port_check.c:16 Failed to get slot informtion of sata1"错误如下：

messages截取：
2024-10-26T01:14:31+08:00 Fyred-NAS synostgd-disk[10275]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T01:14:31+08:00 Fyred-NAS synostgd-disk[10275]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T01:14:31+08:00 Fyred-NAS synostgd-disk[10275]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T01:15:00+08:00 Fyred-NAS synostgd-disk[5459]: SYSTEM:	Last message 'sata_dev_port_check.' repeated 3 times, suppressed by syslog-ng on Fyred-NAS
2024-10-26T01:15:31+08:00 Fyred-NAS synostgd-disk[10275]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T01:15:31+08:00 Fyred-NAS synostgd-disk[10275]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T01:15:31+08:00 Fyred-NAS synostgd-disk[10275]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T01:16:00+08:00 Fyred-NAS synostgd-disk[5459]: SYSTEM:	Last message 'sata_dev_port_check.' repeated 3 times, suppressed by syslog-ng on Fyred-NAS
2024-10-26T01:16:31+08:00 Fyred-NAS synostgd-disk[10275]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T01:16:31+08:00 Fyred-NAS synostgd-disk[10275]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T01:16:31+08:00 Fyred-NAS synostgd-disk[10275]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T01:17:00+08:00 Fyred-NAS synostgd-disk[5459]: SYSTEM:	Last message 'sata_dev_port_check.' repeated 3 times, suppressed by syslog-ng on Fyred-NAS

scemd.log截取：
2024-10-26T11:57:07+08:00 Fyred-NAS scemd[5459]: SYSTEM:	Last message 'sata_dev_port_check.' repeated 14 times, suppressed by syslog-ng on Fyred-NAS
2024-10-26T11:57:06+08:00 Fyred-NAS scemd[11963]: redundant_power_check.c:323 Fail to read power status.
2024-10-26T11:57:06+08:00 Fyred-NAS scemd[11963]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T11:57:06+08:00 Fyred-NAS scemd[11963]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T11:57:06+08:00 Fyred-NAS scemd[11963]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T11:57:13+08:00 Fyred-NAS scemd[5459]: SYSTEM:	Last message 'sata_dev_port_check.' repeated 11 times, suppressed by syslog-ng on Fyred-NAS
2024-10-26T11:57:12+08:00 Fyred-NAS scemd[11963]: redundant_power_check.c:323 Fail to read power status.
2024-10-26T11:57:12+08:00 Fyred-NAS scemd[11963]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T11:57:12+08:00 Fyred-NAS scemd[11963]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T11:57:12+08:00 Fyred-NAS scemd[11963]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T11:57:17+08:00 Fyred-NAS scemd[5459]: SYSTEM:	Last message 'sata_dev_port_check.' repeated 8 times, suppressed by syslog-ng on Fyred-NAS
2024-10-26T11:57:17+08:00 Fyred-NAS scemd[11963]: redundant_power_check.c:323 Fail to read power status.
2024-10-26T11:57:17+08:00 Fyred-NAS scemd[11963]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T11:57:17+08:00 Fyred-NAS scemd[11963]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1
2024-10-26T11:57:17+08:00 Fyred-NAS scemd[11963]: sata_dev_port_check.c:16 Failed to get slot informtion of sata1

可在存储管理器里未见硬盘错误
[image]

如何能消除这些错误？谢谢

logs:  

/etc.defaults/model.dtb
[model.zip](https://github.com/user-attachments/files/17528999/model.zip)

cat /sys/block/sata*/device/syno_block_info
driver=ahci
pciepath=a9f3:00:00.0
ata_port_no=0
driver=ahci
pciepath=a9f3:00:00.0
ata_port_no=1
driver=ahci
pciepath=a9f3:00:00.0
ata_port_no=2
driver=ahci
pciepath=a9f3:00:00.0
ata_port_no=3
driver=ahci

ls -l /sys/class/scsi_host
total 0
lrwxrwxrwx 1 root root 0 Oct 26 00:18 host0 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/d4f21e61-bf21-45ef-82da-8e246d69f1a5/host0/scsi_host/host0
lrwxrwxrwx 1 root root 0 Oct 26 00:18 host1 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/a9153c15-a9f3-46ed-82d6-0b77783a2e9a/pcia9f3:00/a9f3:00:00.0/ata1/host1/scsi_host/host1
lrwxrwxrwx 1 root root 0 Oct 26 00:18 host2 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/a9153c15-a9f3-46ed-82d6-0b77783a2e9a/pcia9f3:00/a9f3:00:00.0/ata2/host2/scsi_host/host2
lrwxrwxrwx 1 root root 0 Oct 26 00:18 host3 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/a9153c15-a9f3-46ed-82d6-0b77783a2e9a/pcia9f3:00/a9f3:00:00.0/ata3/host3/scsi_host/host3
lrwxrwxrwx 1 root root 0 Oct 26 00:18 host4 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/a9153c15-a9f3-46ed-82d6-0b77783a2e9a/pcia9f3:00/a9f3:00:00.0/ata4/host4/scsi_host/host4
lrwxrwxrwx 1 root root 0 Oct 26 00:18 host5 -> ../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0004:00/VMBUS:00/a9153c15-a9f3-46ed-82d6-0b77783a2e9a/pcia9f3:00/a9f3:00:00.0/ata5/host5/scsi_host/host5

Fyred, 2024-10-26:
莫非是我5口扩展卡空一个口不行？

wjz304, 2024-10-26:
sata1 应该是引导盘，

Fyred, 2024-10-26:
> sata1 应该是引导盘，sa6400 的 sata 引导 使用了特殊手段，是有个报错的 暂时不好处理，

懂了，期待今后有机会解决

closed 2024-10-26 by Fyred