RROrg/rr #4226: BUG SA6400 + 7.2.1 kernel 5.10.55 + intel 10-12代 會導致VM Windows 不斷當機

chicman, 2024-10-28:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  鐵威馬F2-424 Pro
 * CPU:  12th i3-N305
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  22.10.0
 * addons:  *太多...
 * modules:  全選了
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2.1 / 69057

Issue:  

VMM下裝Windows 10 /11 會不斷重複藍屏重啟。調試過VMM各種設置，（官方iso/第三方iso）都試過都無法解決。
Docker版本的windows guest也是一樣。

今天在國外論壇看到原來是 kernel 問題，配合intel 10代之後 CPU 就會。
原文：https://forums.virtualbox.org/viewtopic.php?t=101667&start=45
ticket: https://www.virtualbox.org/ticket/20180

長話短說：windows會認為[(STATUS_DATATYPE_MISALIGNMENT)]，因為
簡單的解法是：Linux kernel parameter：split_lock_detect=off
VirtualBox官方的解法是版本更新了，不確定他們怎麼做。

我重裝過loader config中的custom kernel, 版本號也是5.10.55+, 似乎沒有解決。
還請大神有因緣的話，可以看看。不知道開機是否可以加入kernel參數呢？

P.S: 群暉官方用AMD的U所以這版本kernel才被選用吧？

logs:  

log沒有什麼有用資訊

wjz304, 2024-10-28:
似乎有效，我测试一下吧，届时更新引导就好了

cjwkof, 2024-10-28:
希望可以解决，因为这个问题，我装了PVE。。。

wjz304, 2024-10-28:
> N305+最新引导遇到了相同问题，打开了direct boot选项，就没问题了，目前已经稳定运行四天。

他这个方法可以 兼容 非直接启动，看看吧，一般要凑一波修改再发布版本

cjwkof, 2024-10-29:
请教一下，直接启动是不是就不会使用rr的内核，所以就不会出现蓝屏？

> > N305+最新引导遇到了相同问题，打开了direct boot选项，就没问题了，目前已经稳定运行四天。
> 
> 他这个方法可以 兼容 非直接启动，看看吧，一般要凑一波修改再发布版本

wjz304, 2024-10-29:
> 请教一下，直接启动是不是就不会使用rr的内核，所以就不会出现蓝屏？
> 
> > > N305+最新引导遇到了相同问题，打开了direct boot选项，就没问题了，目前已经稳定运行四天。
> > 
> > 
> > 他这个方法可以 兼容 非直接启动，看看吧，一般要凑一波修改再发布版本

是

wjz304, 2024-10-29:
https://github.com/RROrg/rr/releases/tag/24.10.3

本来想等几天，但是突然有事要回老家，先发一个版本吧

closed 2024-10-29 by wjz304

chicman, 2024-10-29:
> N305+最新引导遇到了相同问题，打开了direct boot选项，就没问题了，目前已经稳定运行四天。

請問direct boot是在哪裡？rr編譯還是bios?

Lucas-lyh, 2024-10-29:
应该也可以在编译时设置，我是在rr manager套件中直接勾选的。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年10月29日 14:33 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [RROrg/rr] BUG SA6400 + 7.2.1 kernel 5.10.55 + intel 10-12代 會導致VM Windows 不斷當機 (Issue #4226) |

N305+最新引导遇到了相同问题，打开了direct boot选项，就没问题了，目前已经稳定运行四天。

請問direct boot是在哪裡？rr編譯還是bios?

chicman, 2024-10-29:
> 应该也可以在编译时设置，我是在rr manager套件中直接勾选的。
>
>
>
>
> ---- 回复的原邮件 ----
> | 发件人 | ***@***.***> |
> | 日期 | 2024年10月29日 14:33 |
> | 收件人 | ***@***.***> |
> | 抄送至 | ***@***.***>***@***.***> |
> | 主题 | Re: [RROrg/rr] BUG SA6400 + [...]

用rr manager輕鬆修改。目前看來vim w11很穩。不過rr manager無法直接更新啊。回報error no update file.但是確定新版的引導已經下載rr/tmp.

或許我要另外開一個issue ticket...?

wjz304, 2024-11-06:
> > 应该也可以在编译时设置，我是在rr manager套件中直接勾选的。
> 
> 用rr manager輕鬆修改。目前看來vmm w11很穩。不過rr manager無法直接更新啊。回報error no update file.但是確定新版的引導已經下載rr/tmp.
> 
> 或許我要另外開一個issue ticket...?

rr manager 的问题已有人反馈