RROrg/rr #4256: BUG PhotoFace-Patch

MartinKuhl, 2024-10-30:
**Install ENV:**

CPU: Intel Atom
NIC: SKY2
RR version: latest 24.10.3

**DSM:**

model: 3622xs+
version: 7.2.1 Patch 5

**Issue:**

The PhotoFace-Patch is not enabling the face recognition in DSM. All tries to recognize the faces ending in timeouts.
If I directly execute the patch from here: https://github.com/jinlife/Synology_Photos_Face_Patch/blob/main/bin/patch.sh
it is directly working. Is there a special order how to install the DSM Photo app and installing the patch from the loader after the DSM-app installation?

wjz304, 2024-11-01:
加载程序选则addon，启动DSM后安装 Photo ，重启DSM

MartinKuhl, 2024-11-02:
I alreasy installed the addon but it isn't working

wjz304, 2024-11-04:
I tested it and found that there is a bug in the addon.
Since Synology Photos is installed in the storage pool, the path where it is located will only exist after the application is started.
And the addon is run too early, resulting in the path where the photos library does not exist during runtime.

wjz304, 2024-11-04:
This problem is currently difficult to solve.

MartinKuhl, 2024-11-04:
Looks Like arc loader fixed this issue in one of the last updates. Just FYI maybe you can adopt this solution.

wjz304, 2024-11-04:
The changes of arc don't make sense, but I did find a solution.

wjz304, 2024-11-05:
https://github.com/RROrg/rr/releases/tag/24.11.0
test

AuxXxilium, 2024-11-06:
> The changes of arc don't make sense, but I did find a solution.

they make sense, because the load point of service was wrong: if we change it back to`After=multi-user.target` it is working like it should

wjz304, 2024-11-07:
> > The changes of arc don't make sense, but I did find a solution.
> 
> they make sense, because the load point of service was wrong: if we change it back to`After=multi-user.target` it is working like it should

Yes, but that's not strict, the photosfacepatch service should be run after the storage pool is loaded (theoretically, before the photo app is started).

closed 2024-11-15 by wjz304