RROrg/rr #4304: rr版本更新v24.5.1 => 24.10失败

eliliu911, 2024-11-06:
在线更新，完成解压后报错
/opt/rr/menu.sh: 行 2896: 2 + : 语法错误：需要操作数（错误记号是 "+ "）

wjz304, 2024-11-07:
https://rrorg.cn/archives/rrupdate

closed 2024-11-07 by wjz304

fairylyy, 2026-10-29:
2026年，我也遇到了一样的问题。使用AI解决了，供参考。

原因已经定位清楚：**这是 RR 24.3.6 自带的 `updateRR()` 脚本缺陷，并非 `updateall` 包损坏或上传失败。**

你贴出的 `menu.sh` 与 RR 24.3.6 的版本一致。问题在 `updateRR()` 的 2628、2629 两行：

```bash
SIZENEW=$((${SIZENEW} + $(du -sm "${TMP_PATH}/update/${VALUE}" ... | awk '{print $1}')))
SIZEOLD=$((${SIZEOLD} + $(du -sm "${VALUE}" ... | awk '{print $1}')))
```

新版 `updateall` 的 `update-list.yml` 中新增了类似下面的文件：

```text
/mnt/p1/boot/grub/logo.png
/mnt/p1/boot/grub/memtest
```

这些文件在 24.3.6 的引导盘里尚不存在。此时 `du -sm` 不输出数字，shell 算式就变成了：

```text
2 + 
```

所以报错：

```text
2 + : 语法错误：需要操作数
```

这里的 `2` 是前面已累计的文件大小。由于脚本在真正复制文件之前就退出了，**系统没有被改坏，升级包也没有问题**。

## 修改方法

你现在正在 `root@rr` 的 shell 里，直接执行：

```bash
cd /opt/rr
cp -a menu.sh menu.sh.bak

sed -i "2628s/awk '{print \$1}'/awk 'END{print \$1+0}'/" menu.sh
sed -i "2629s/awk '{print \$1}'/awk 'END{print \$1+0}'/" menu.sh

sed -n '2628,2629p' menu.sh
bash -n menu.sh
```

修改后这两行应显示为：

```bash
SIZENEW=$((${SIZENEW} + $(du -sm "${TMP_PATH}/update/${VALUE}" 2>/dev/null | awk 'END{print $1+0}')))
SIZEOLD=$((${SIZEOLD} + $(du -sm "${VALUE}" 2>/dev/null | awk 'END{print $1+0}')))
```

这样当目标文件不存在时，`du` 没有输出也会计为 `0 MB`。

验证语法没有问题后，**不要重启**，直接重新启动菜单：

```bash
cd /opt/rr
bash menu.sh
```

然后进入：

```text
Update menu -> Update All
```

`updateall` 包还在 `/tmp` 里的话不需要重新上传。成功后按提示重启，新的 initrd 才会生效。

> 注意：`/opt/rr` 在内存盘中，修改只在本次启动有效。若先重启，补丁和 `/tmp` 里的升级包都会丢失，需要重新上传补丁和升级包。