RROrg/rr #4336: rr引导编译成功， 但在局域网中找不到synology设置

JiangYeming, 2024-11-10:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS920+
 * version:  7.2

Issue:  VMbox虚拟机安装，rr引导编译成功， 但在局域网中找不到synology设置,是什么原因？

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-11-10:
（请先看一下https://github.com/RROrg/rr/issues/173、https://github.com/RROrg/rr/issues/175、https://github.com/RROrg/rr/issues/226 的内容）
（请先查看 https://github.com/RROrg/rr/issues/173、https://github.com/RROrg/rr/issues/175、https://github.com/RROrg/rr/issues/226 的内容）

closed 2024-11-10 by wjz304

JiangYeming, 2024-11-11:
已解，是操作系统的问题
发自我的 iPhone