RROrg/rr #4340: Unable to set Static IP for RR system

manleong, 2024-11-10:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  24.11.1
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS918+
 * version:  DSM 7.2.1-69057 Update 5

Issue:  
Unable to set static IP from RR menu, tried to manual amend the **/etc/network/interfaces** but seem like after restart everything will be go back to default state, not sure is it part on the buildroot default setting

Same goes to RR menu static IP, first configure is fine to get ping from new ip, but it will go back to dhcp assign ip after restart

logs:  
Let me know if any log is required to debug

wjz304, 2024-11-10:
[image]
[image]

manleong, 2024-11-10:
> ![image](https://private-user-images.githubusercontent.com/5615843/384672618-0860eceb-c526-412a-9bb2-23118f5b3d50.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEyMzIzODYsIm5iZiI6MTczMTIzMjA4NiwicGF0aCI6Ii81NjE1ODQzLzM4NDY3MjYxOC0wODYwZWNlYi1jNTI2LTQxMmEtOWJiMi0yMzExOGY1YjNkNTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTBUMDk0ODA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWRlYTczOWZkYjNkNmZiZTI2YWUwZmViYjA3MDBmYzc3NTQ5NjI2MWI3OTMwOTE3OTAyZjAyMjZkMTFlNDYzOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.YPOXY8UKKug3xTQkDg4_MJI4VO4IA0n9Ej-39-zqawE) ![image](https://private-user-images.githubusercontent.com/5615843/384672660-11986f6b-9246-45f8-b570-13f45d10342d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEyMzIzODYsIm5iZiI6MTczMTIzMjA4NiwicGF0aCI6Ii81NjE1ODQzLzM4NDY3MjY2MC0xMTk4NmY2Yi05MjQ2LTQ1ZjgtYjU3MC0xM2Y0NWQxMDM0MmQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTBUMDk0ODA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzU3Mjg3MzU4MmQ4Yzc4MmY5YmYwYjY1OWJhZjk1ODQwYmJjOTVmYmQxZjY4NmNkMDRjMjNmNDQ3NWY1YWE1ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.TQn4_akvnhJA9NDQfdc4GMofGxzMcSbnUpThj_K2PKM)

Yes, i tried to set the static ip here, it take effective until i restart and it will be back to the default dhcp assign ip address

wjz304, 2024-11-19:
test https://github.com/RROrg/rr/releases/tag/24.11.11

manleong, 2024-11-23:
> test https://github.com/RROrg/rr/releases/tag/24.11.11

Tried on 24.11.12, issue not fixed

wjz304, 2025-06-05:
https://github.com/RROrg/rr/releases/tag/25.5.9

closed 2025-06-05 by wjz304