RROrg/rr #4348: 问题反馈：关于"启动"后无法找到DSM的一些发现

q331723516, 2024-11-12:
物理机配置：主板z370，cpu i7 8700 &主板b360 ，cpu I5 8500
目前 RR：24.6.7 

RR24.8.0之后的版本（包括最新的）编译自带**核显**机型，启动后无法进入DSM；改为直接启动可正常进入DSM；但是编译**不带核显**机型，直接启动true或false，启动后都正常进入DSM。
同时还发现编译自带**核显**机型,主板BIOS关掉VT-D,启动后也是可以正常进入DSM，不需更改直接启动。

QQ群里有人也有一样的情况

q331723516, 2024-11-12:
希望对开发者 有所帮助

Gabriel-six, 2024-11-12:
同样情况。而且今晚还崩了一次引导，24.10版本

q331723516, 2024-11-16:
> 同样情况。而且今晚还崩了一次引导，24.10版本

崩？从来没崩过！我用的920
只是24.7.3之后的版本不改直接启动进不了dsm；
两个方法：要么改直接启动进dsm
要么关闭bios，进dsm

Gabriel-six, 2024-11-16:
> > 同样情况。而且今晚还崩了一次引导，24.10版本
> 
> 崩？从来没崩过！我用的920 只是24.7.3之后的版本不改直接启动进不了dsm； 两个方法：要么改直接启动进dsm 要么关闭bios里的 vt-d；也可以进dsm

啊哈哈我那个可能是u盘太垃圾了导致的

tualatina, 2024-11-18:
dell3060sff 8600t 一直就没起来过 关vt-d 换直接启动 换dev 都没用 boot一会儿能获取dhcp然后过会儿路由器里就看不到这家伙了。。

q331723516, 2024-11-18:
时有时无应该是网卡的问题，我说的情况是，以前可以正常，24.8.0版本之后不可以的

q331723516, 2024-11-19:
> https://github.com/RROrg/rr/releases/tag/24.11.11

解决了 👍

closed 2024-11-19 by wjz304