RROrg/rr #4368: 更新到24.11.1 无法找到群辉

eliliu911, 2024-11-14:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Intel Default string/ITX40D, BIOS 5.13 08/03/2022
 * CPU:  Intel(R) Celeron(R) J4125 CPU @ 2.00GHz
 * NIC:  (pid & vid)0x0001

RR version: (You can find it in the update menu.)  
 * RR:  24.11.1
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS920+(geminilake)
 * version:  7.2(72806)

Issue:  更新到24.11.1重启NAS，提示进入引导，但是ip地址无法访问，无法使用Synology Assistant查找到服务器，路由器中无法看到对应的NAS ip地址。 只有在重启NAS时可以通过ssh进入nas。

logs:  

型号:     DS920+(geminilake)
版本:     7.2(72806)
内核:     official
LKM:      prod
DMI:      Intel Default string/ITX40D, BIOS 5.13 08/03/2022
CPU:      Intel(R) Celeron(R) J4125 CPU @ 2.00GHz
MEM:      7534 MB
Cmdline:
withefi dom_szmax=30543 earlyprintk syno_hw_version=DS920+ console=ttyS0,115200n8 consoleblank=600 netif_num=2 pid=0x0001 earlycon=uart8250,io,0x3f8,115200n8 synoboot_satadom=2 mac2=********* mac1=********* sn=********* modprobe.blacklist=evbug,cdc_ether,mpt3sas pcie_aspm=off rootwait vid=0x46f4 elevator=elevator loglevel=15 HddHotplug=1 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 intel_iommu=igfx_off panic=5 vender_format_version=2 log_buf_len=32M root=/dev/md0 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8
检测到 2 个网卡.
eth0 已连接.                  
获取 IP.
eth0(r8168): 在浏览器中访问 http://192.168.50.135:5000 链接 DSM.
eth1(r8168): 未连接
                                                                          
加载 DSM 内核 ...
引导中 ...

[该界面已不可操作. 请在数分钟后
通过 http://find.synology.com/ 或者 Synology Assistant 查找 DSM 并访问.]

eliliu911, 2024-11-14:
已找到原因：
解决方法：大概率是kernel崩溃掉了，导致搜索不到ip。 切换直接启动：true  重启解决。

closed 2024-11-14 by eliliu911

shua1zh, 2024-11-15:
我应该是遇到和你一样的问题了，J4125软路由物理机安装黑裙，用的最新的rr引导，到最后一步加载DSM 内核就黑屏没反应，然后自动重启还是搜不到ip，请问你切换直接启动：true ，这个操作是在哪改的啊

eliliu911, 2024-11-15:
> 我应该是遇到和你一样的问题了，J4125软路由物理机安装黑裙，用的最新的rr引导，到最后一步加载DSM 内核就黑屏没反应，然后自动重启还是搜不到ip，请问你切换直接启动：true ，这个操作是在哪改的啊

可能类似，但也不太一样。我的是不重启的，就是直接启动了，但是找不到ip。你可以试试看：
你需要给nas接个显示器，重启nas后会有提示，让你使用ssh或者web打开ip地址以中断操作，我一般是ssh链接，进入终端后可以在opt/rr目录下找到menu.sh，运行后就是控制面板，里面第一页就有 `直接启动：false`，需要把这个改成true。如果找不到这个选项，可以看看`高级设置`里。