RROrg/rr #4391: N100核显直通，ffmpeg只要使用qsv转码就会报错

shikaiwei1, 2024-11-16:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  intel n100
 * NIC:  (pid & vid)  vid:0x46f4   pid: 0x0001

RR version: (You can find it in the update menu.)  
 * RR:  24.11.1
 * addons:  
 * modules:  
 * lkms:  prod

DSM: 
 * model:  SA6400
 * version:  DSM 7.2.2-72806

Issue:  
N100核显直通，ffmpeg只要使用qsv转码就会报错。（安装的是ffmpeg7）

下列日志是jellyfin中播放时的日志报错。

logs1:  

[2024-11-16 22:52:44.188 +08:00] [INF] [60] MediaBrowser.MediaEncoding.Transcoding.TranscodeManager: "/var/packages/ffmpeg7/target/bin/ffmpeg" "-analyzeduration 200M -probesize 1G -init_hw_device vaapi=va:/dev/dri/renderD128,driver=iHD -init_hw_device qsv=qs@va -filter_hw_device qs -i file:\"/volume2/video/2.动画/[欧美][动作][我是传奇][高清RMVB][1280×720][中英双字幕].RMVB\" -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:1 -map -0:s -codec:v:0 hevc_qsv -tag:v:0 hvc1 -preset veryfast -b:v 4184342 -maxrate 4184342 -bufsize 8368684 -profile:v:0 main -level 50 -g:v:0 72 -keyint_min:v:0 72 -vf \"setparams=color_primaries=bt709:color_trc=bt709:colorspace=bt709,scale=trunc(min(max(iw\,ih*a)\,3440)/2)*2:trunc(ow/a/2)*2,format=nv12\" -codec:a:0 libfdk_aac -ac 2 -ab 256000 -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -f hls -max_delay 5000000 -hls_time 3 -hls_segment_type fmp4 -hls_fmp4_init_filename \"6562cce61ede5aad9c91397ebea8c6e2-1.mp4\" -start_number 0 -hls_base_url \"hls/6562cce61ede5aad9c91397ebea8c6e2/\" -hls_segment_filename \"/volume2/@appdata/jellyfin/cache/transcodes/6562cce61ede5aad9c91397ebea8c6e2%d.mp4\" -hls_playlist_type event -hls_list_size 0 -y \"/volume2/@appdata/jellyfin/cache/transcodes/6562cce61ede5aad9c91397ebea8c6e2.m3u8\""
[2024-11-16 22:52:44.358 +08:00] [ERR] [60] MediaBrowser.MediaEncoding.Transcoding.TranscodeManager: FFmpeg exited with code 134
[2024-11-16 22:52:44.406 +08:00] [ERR] [60] Jellyfin.Api.Middleware.ExceptionMiddleware: Error processing request. URL "GET" "/videos/8443cdda-a900-8ba4-c4c2-646c7cd9b76c/live.m3u8".
MediaBrowser.Common.FfmpegException: FFmpeg exited with code 134
   at MediaBrowser.MediaEncoding.Transcoding.TranscodeManager.StartFfMpeg(StreamState state, String outputPath, String commandLineArguments, Guid userId, TranscodingJobType transcodingJobType, CancellationTokenSource cancellationTokenSource, String workingDirectory)
   at Jellyfin.Api.Controllers.DynamicHlsController.GetLiveHlsStream(Guid itemId, String container, Nullable`1 static, String params, String tag, String deviceProfileId, String playSessionId, String segmentContainer, Nullable`1 segmentLength, Nullable`1 minSegments, String mediaSourceId, String deviceId, String audioCodec, Nullable`1 enableAutoStreamCopy, Nullable`1 allowVideoStreamCopy, Nullable`1 allowAudioStreamCopy, Nullable`1 breakOnNonKeyFrames, Nullable`1 audioSampleRate, Nullable`1 maxAudioBitDepth, Nullable`1 audioBitRate, Nullable`1 audioChannels, Nullable`1 maxAudioChannels, String profile, String level, Nullable`1 framerate, Nullable`1 maxFramerate, Nullable`1 copyTimestamps, Nullable`1 startTimeTicks, Nullable`1 width, Nullable`1 height, Nullable`1 videoBitRate, Nullable`1 subtitleStreamIndex, Nullable`1 subtitleMethod, Nullable`1 maxRefFrames, Nullable`1 maxVideoBitDepth, Nullable`1 requireAvc, Nullable`1 deInterlace, Nullable`1 requireNonAnamorphic, Nullable`1 transcodingMaxAudioChannels, Nullable`1 cpuCoreLimit, String liveStreamId, Nullable`1 enableMpegtsM2TsMode, String videoCodec, String subtitleCodec, String transcodeReasons, Nullable`1 audioStreamIndex, Nullable`1 videoStreamIndex, Nullable`1 context, Dictionary`2 streamOptions, Nullable`1 maxWidth, Nullable`1 maxHeight, Nullable`1 enableSubtitlesInManifest, Boolean enableAudioVbrEncoding, Boolean alwaysBurnInSubtitleWhenTranscoding)
   at lambda_method1443(Closure, Object)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ActionMethodExecutor.TaskOfActionResultExecutor.Execute(ActionContext actionContext, IActionResultTypeMapper mapper, ObjectMethodExecutor executor, Object controller, Object[] arguments)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeActionMethodAsync>g__Awaited|12_0(ControllerActionInvoker invoker, ValueTask`1 actionResultValueTask)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeNextActionFilterAsync>g__Awaited|10_0(ControllerActionInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Rethrow(ActionExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeInnerFilterAsync>g__Awaited|13_0(ControllerActionInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeNextResourceFilter>g__Awaited|25_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Rethrow(ResourceExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeFilterPipelineAsync>g__Awaited|20_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
   at Jellyfin.Api.Middleware.ServerStartupMessageMiddleware.Invoke(HttpContext httpContext, IServerApplicationHost serverApplicationHost, ILocalizationManager localizationManager)
   at Jellyfin.Api.Middleware.WebSocketHandlerMiddleware.Invoke(HttpContext httpContext, IWebSocketManager webSocketManager)
   at Jellyfin.Api.Middleware.IPBasedAccessValidationMiddleware.Invoke(HttpContext httpContext, INetworkManager networkManager)
   at Jellyfin.Api.Middleware.LanFilteringMiddleware.Invoke(HttpContext httpContext, INetworkManager networkManager, IServerConfigurationManager serverConfigurationManager)
   at Microsoft.AspNetCore.Authorization.AuthorizationMiddleware.Invoke(HttpContext context)
   at Jellyfin.Api.Middleware.QueryStringDecodingMiddleware.Invoke(HttpContext httpContext)
   at Swashbuckle.AspNetCore.ReDoc.ReDocMiddleware.Invoke(HttpContext httpContext)
   at Swashbuckle.AspNetCore.SwaggerUI.SwaggerUIMiddleware.Invoke(HttpContext httpContext)
   at Swashbuckle.AspNetCore.Swagger.SwaggerMiddleware.Invoke(HttpContext httpContext, ISwaggerProvider swaggerProvider)
   at Microsoft.AspNetCore.Authentication.AuthenticationMiddleware.Invoke(HttpContext context)
   at Jellyfin.Api.Middleware.RobotsRedirectionMiddleware.Invoke(HttpContext httpContext)
   at Jellyfin.Api.Middleware.LegacyEmbyRouteRewriteMiddleware.Invoke(HttpContext httpContext)
   at Microsoft.AspNetCore.ResponseCompression.ResponseCompressionMiddleware.InvokeCore(HttpContext context)
   at Jellyfin.Api.Middleware.ResponseTimeMiddleware.Invoke(HttpContext context, IServerConfigurationManager serverConfigurationManager)
   at Jellyfin.Api.Middleware.ExceptionMiddleware.Invoke(HttpContext context)

# logs2:
> 下面是手动运行转码时的报错：
root@John-Nas-SA6400:/var/packages/ffmpeg7/target/bin# /var/packages/ffmpeg7/target/bin/ffmpeg  -hwaccel qsv  -i "/volume2/video/2.动画/Ice.Age.Continental.Drift.2012.1080p.BluRay.x264.DTS-HDC/Ice.Age.Continental.Drift.2012.1080p.BluRay.x264.DTS-HDC.mkv"  -c:v h264_qsv -y "/volume2/temp/Ice.Age.Continental.Drift.output.mp4"
ffmpeg version 7.0.2-5 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 8.5.0 (GCC)
  configuration: --target-os=linux --cross-prefix=/github/workspace/toolchain/syno-x64-7.1/work/x86_64-pc-linux-gnu/bin/x86_64-pc-linux-gnu- --prefix=/var/packages/ffmpeg7/target --extra-cflags=-I/github/workspace/spk/ffmpeg7/work-x64-7.1/install/var/packages/ffmpeg7/target/include --extra-ldflags=-L/github/workspace/spk/ffmpeg7/work-x64-7.1/install/var/packages/ffmpeg7/target/lib --extra-libs='-lxml2 -ldl -lm' --pkg-config=/usr/bin/pkg-config --ranlib=/github/workspace/toolch [... cut ...]
Device creation failed: -1313558101.
[vist#0:0/h264 @ 0x55cee34e78c0] [dec:h264_qsv @ 0x55cee34c8f40] No device available for decoder: device type qsv needed for codec h264_qsv.
[vist#0:0/h264 @ 0x55cee34e78c0] [dec:h264_qsv @ 0x55cee34c8f40] Hardware device setup failed for decoder: Unknown error occurred
[vost#0:0/h264_qsv @ 0x55cee34e9f00] Error initializing a simple filtergraph
Error opening output file /volume2/temp/Ice.Age.Continental.Drift.output.mp4.
Error opening output files: Unknown error occurred

langsisi, 2024-12-21:
解决了吗？我这边N100也是转码失败

shikaiwei1, 2024-12-21:
后来用docker跑jellfin，可以正常硬解了

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年12月21日 19:29 |
| 收件人 | ***@***.***> |
| 抄送至 | John ***@***.***>***@***.***> |
| 主题 | Re: [RROrg/rr] N100核显直通，ffmpeg只要使用qsv转码就会报错 (Issue #4391) |

解决了吗？我这边N100也是转码失败

langsisi, 2024-12-22:
version: '3' 
services:
  emby:
    image: jellyfin/jellyfin 
    container_name: jellyfin2               # 容器名称
    ports:
      - '8296:8096'                   # 将容器的 8096 端口映射到主机的 8296 端口
    volumes:
      - /volume2/docker/jellyfin2/config:/config         
      - /volume2/docker/jellyfin2/cache:/cache
      - /volume3/迅雷:/media
    privileged: true
    devices:
      - /dev/dri:/dev/dri             # 将主机的 /dev/dri 设备挂载到容器
我用docker还是不行，缺少了什么吗？

langsisi, 2024-12-22:
docker run -d --volume /volume2/docker/jellyfin2/config:/config --volume /volume2/docker/jellyfin2/cache:/cache --mount type=bind,source=/volume3/迅雷,target=/media --net=host --restart=always --device /dev/dri/renderD128:/dev/dri/renderD128 --device /dev/dri/card0:/dev/dri/card0 --name="jellyfin2" jellyfin/jellyfin
容器运行是正常的，有些视频不能解码，有的能硬解

langsisi, 2024-12-23:
问题终于解决了，使用nyanmisaka/jellyfin镜像，可以硬解了，记录一下，方便其他人看到：
version: '3'
services:
    jellyfin:
        volumes:
            - /volume2/docker/jellyfin/config:/config
            - /volume2/docker/jellyfin/cache:/cache
            - /volume3/迅雷:/media
        network_mode: host
        restart: always
        privileged: true
        ports:
            - 8096:8096
        devices:
            - /dev/dri/renderD128:/dev/dri/renderD128
            - /dev/dri/card0:/dev/dri/card0
        container_name: jellyfin
        image: nyanmisaka/jellyfin

zhouhuan18, 2025-01-01:
> 问题终于解决了，使用nyanmisaka/jellyfin镜像，可以硬解了，记录一下，方便其他人看到： version: '3' services: jellyfin: volumes: - /volume2/docker/jellyfin/config:/config - /volume2/docker/jellyfin/cache:/cache - /volume3/迅雷:/media [...]

我这边硬解会出现播放器致命错误，然后退出，12300T物理机安装的黑群，是按照你的配置构建的，不知道什么原因