RROrg/rr #4407: 关于使用阵列卡只能读取第一块盘的Smart数据的解决思路

Phachd, 2024-11-18:
https://github.com/[332505/syno_smartctl](https://github.com/332505/syno_smartctl/)/
我看到用户332505提供了一个读取阵列卡SMART数据的解决方案，并且尝试后成功了（sa6400 7.2.1 3008直通卡 修改后在dsm安卓app里面看到了正确的SAMET数据）

该用户原话是“HBA卡显示错误的SMART信息其实是因为群晖在调用smartctl命令的时候使用了"-d ata"这个参数指定设备类型，但是对于HBA卡上的sata硬盘得用"-d sat"才对。选错了设备类型当然显示不了正确的SMART信息。我写了一个脚本通过劫持smartctl修改参数，应该能让HBA卡上的sata硬盘正常显示，使用的时候记得改一下脚本里的blacklist就行。”

这个好像能够让群晖定时任务正确的smart检测阵列卡下的硬盘，从而提升数据的安全性。
感觉这个是思路是当作一个插件自动化的，有没有大佬跟进一下

wjz304, 2024-11-19:
https://github.com/RROrg/rr/releases/tag/24.11.11

Phachd, 2024-11-19:
> https://github.com/RROrg/rr/releases/tag/24.11.11

牛的大佬，进度真快。我是到12月月底之前是真的没啥时间写代码PR（工作+私事等）。感谢大佬抬举

closed 2024-11-19 by Phachd

wjz304, 2024-11-19:
你试试吧，我没测试