RROrg/rr #4408: BUG Request to adapt Intel(R) N150 graphics driver under DSM_SA6400

CWWK-TechnicalSupport, 2024-11-18:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Default string Default string/Default string, Blos 5.27 10/31/2024
 * CPU:  Intel(R)N150
 * NIC:  (pid & vid) eth0 igc

RR version: (You can find it in the update menu.)  
 * RR:  24.11.11
 * addons:  nvmesystem
 * modules:  
 * lkms:  prod

DSM: 
 * model:  SA6400(cpyc7002)
 * version:  7.2(72806)

Issue:  Request to adapt Intel(R) N150 graphics driver under DSM_SA6400

My current environment is Debian12 Linux CW-ADLN-NAS-B01 6.11.5+bpo-amd64 https://github.com/moetayuko/intel-gpu-i915-backports/pull/1 SMP PREEMPT_DYNAMIC Debian 6.11.5-1~bpo12+1 (2024-11-11) x86_64 GNU/Linux can drive Intel(R ) N150 graphics card driver, but not under Proxmox or DSM_SA6400, please adapt it.

logs:  

root@CW-ADLN-NAS-B01:/mnt/Smb_Mount/Video/Link# lspci -nn -s 00:02.0
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [Intel Graphics] [8086:46d4]
This is the corresponding hardware ID

00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [Intel Graphics] (prog-if 00 [VGA controller])
DeviceName: Onboard - Video
Subsystem: Intel Corporation Alder Lake-N [Intel Graphics]
Flags: bus master, fast devsel, latency 0, IRQ 179, IOMMU group 0
Memory at 6005000000 (64-bit, non-prefetchable) [size=16M]
Memory at 4000000000 (64-bit, prefetchable) [size=256M]
I/O ports at a000 [size=64]
Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
Capabilities: [40] Vendor Specific Information: Len=0c <?>
Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
Capabilities: [d0] Power Management version 2
Capabilities: [100] Process Address Space ID (PASID)
Capabilities: [200] Address Translation Service (ATS)
Capabilities: [300] Page Request Interface (PRI)
Capabilities: [320] Single Root I/O Virtualization (SR-IOV)
Kernel driver in use: i915
Kernel modules: i915

Architecture: x86_64
CPU op-mode(s): 32-bit, 64-bit
Address sizes: 39 bits physical, 48 bits virtual
Byte Order: Little Endian
CPU(s): 4
On-line CPU(s) list: 0-3
Vendor ID: GenuineIntel
BIOS Vendor ID: Intel(R) Corporation
Model name: Intel(R) N150
BIOS Model name: Intel(R) N150 To Be Filled By O.E.M. CPU @ 2.8GHz
BIOS CPU family: 1
CPU family: 6
Model: 190
Thread(s) per core: 1
Core(s) per socket: 4
Socket(s): 1
Stepping: 0
CPU(s) scaling MHz: 19%
CPU max MHz: 3600.0000
CPU min MHz: 700.0000
BogoMIPS: 1612.80
Flags: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology no
nstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm
3dnowprefetch cpuid_fault epb cat_l2 cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a rdseed adx smap clflushopt clwb int
el_pt sha_ni xsaveopt xsavec xgetbv1 xsaves split_lock_detect avx_vnni dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req vnmi umip pku ospke waitpkg gfni vaes vpclmulqdq rdpid movdiri movdir
64b fsrm md_clear serialize arch_lbr ibt flush_l1d arch_capabilities
Virtualization features:
Virtualization: VT-x
Caches (sum of all):
L1d: 128 KiB (4 instances)
L1i: 256 KiB (4 instances)
L2: 2 MiB (1 instance)
L3: 6 MiB (1 instance)
NUMA:
NUMA node(s): 1
NUMA node0 CPU(s): 0-3
Vulnerabilities:
Gather data sampling: Not affected
Itlb multihit: Not affected
L1tf: Not affected
Mds: Not affected
Meltdown: Not affected
Mmio stale data: Not affected
Reg file data sampling: Mitigation; Clear Register File
Retbleed: Not affected
Spec rstack overflow: Not affected
Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl
Spectre v1: Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Spectre v2: Mitigation; Enhanced / Automatic IBRS; IBPB conditional; RSB filling; PBRSB-eIBRS Not affected; BHI BHI_DIS_S
Srbds: Not affected
Tsx async abort: Not affected

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

closed 2024-11-18 by moetayuko

CWWK-TechnicalSupport, 2024-11-18:
@moetayuko Why is it closed? What happened?

moetayuko, 2024-11-18:
will be supported in the next release of rr

wjz304, 2024-11-19:
https://github.com/RROrg/rr/releases/tag/24.11.11