RROrg/rr #4427: SA6400 内核没有加载设备树

ShiYiKongKong, 2024-11-20:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  N100
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
"format":"img","language":"en_US","model":"SA6400","version":"7.2.2-72806-0","kernel":"official","addons":"acpid,mountloader,powersched,reboottoloader,codecpatch,cpuinfo,hdddb,nvmecache,nvmesystem,nvmevolume,photosfacepatch,sensors

DSM: 
 * model:  SA6400
 * version:  7.2.2-72806

Issue:  
 内核没有加载设备树（model.dtb），/proc 中没有device_tree，内核模块中读取of_root 为 NULL
[boot_params.tar.gz](https://github.com/user-attachments/files/17827109/boot_params.tar.gz)

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2024-11-20:
是的

wjz304, 2024-11-25:
无进一步反馈，该 issue 关闭